arendst/Tasmota #16746: increment a variable in a rule return not the expected value

Performance20, 2022-10-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I use a simple rule to increment a variable (var1) by 0.1 but the result is different like I would expected:
Rule1
ON System#Boot DO Backlog RuleTimer1 2; var1 9000 ENDON  
ON Rules#Timer=1 DO Backlog RuleTimer1 2; var1=var1+0.1  ENDON

Results:
9000.100
9000.199
9000.299
9000.398
  ...

If I use a start value like 500, it works well (500.100  500.200  500.300....)
Possible I do something wrong?

Helge

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [VSC] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Wemos board

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":120,"Free":391,"Rules":"ON System#Boot DO Backlog RuleTimer1 2; var1 9000 ENDON   ON Rules#Timer=1 DO Backlog RuleTimer1 2; var1=var1+0.1  ENDON"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
STATUS0 = {"Status":{"Module":18,"DeviceName":"Gaszähler","FriendlyName":["Gaszähler"],"Topic":"Gasmeter_87DB0D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:43","StartupUTC":"2022-10-07T11:42:22","Sleep":50,"CfgHolder":4617,"BootCount":71,"BCResetTime":"2022-09-29T17:06:44","SaveCount":247,"SaveAddress":"F5000"},"StatusFWR":{"Version":"12.1.1.4(tasmota)","BuildDateTime":"2022.10.06 13:57:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"365/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Ratingen2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":572,"Free":428,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F908787","04140001","000000CF","010013C0","40005881","00004000","00000000","54000020","00000008"],"Drivers":"1,2,3,4,7,8,9,10,12,20,21,24,29,62","Sensors":"1,2,3,4,5,6,33"},"StatusNET":{"Hostname":"Gasmeter-87DB0D-6925","IPAddress":"192.168.7.53","Gateway":"192.168.7.150","Subnetmask":"255.255.255.0","DNSServer1":"192.168.7.167","DNSServer2":"192.168.7.150","Mac":"E0:98:06:87:DB:0D","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.7.145","MqttPort":1883,"MqttClientMask":"Gasmeter_%06X","MqttClient":"Gasmeter_87DB0D","MqttUser":"Gasmeter","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-10-07T11:44:05","Local":"2022-10-07T13:44:05","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:58","Sunset":"19:16"},"StatusSNS":{"Time":"2022-10-07T13:44:05","QMC5883L":{"XaxisInduction":-4923,"YaxisInduction":0,"ZaxisInduction":1201,"MagneticInduction":5067,"Temperature":17.0},"TempUnit":"C"},"StatusSTS":{"Time":"2022-10-07T13:44:05","Uptime":"0T00:01:43","UptimeSec":103,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Ratingen2","BSSId":"FA:9F:C2:7D:DE:1C","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:50:18.070 RUL: SYSTEM#BOOT performs "Backlog RuleTimer1 2; var1 9000"
13:50:18.071 SRC: Rule
13:50:18.073 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 23, Pld -99, Data 'RuleTimer1 2; var1 9000'
13:50:18.107 SRC: Backlog
13:50:18.109 CMD: Grp 0, Cmd 'RULETIMER', Idx 1, Len 1, Pld 2, Data '2'
13:50:18.117 MQT: stat/Gasmeter_87DB0D/RESULT = {"T1":1,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
13:50:18.357 SRC: Backlog
13:50:18.359 CMD: Grp 0, Cmd 'VAR', Idx 1, Len 4, Pld 9000, Data '9000'
13:50:18.364 MQT: stat/Gasmeter_87DB0D/RESULT = {"Var1":"9000"}
13:50:20.433 RUL: RULES#TIMER=1 performs "Backlog RuleTimer1 2; var1=var1+0.1"
13:50:20.435 SRC: Rule
13:50:20.437 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 27, Pld -99, Data 'RuleTimer1 2; var1=var1+0.1'
13:50:20.461 SRC: Backlog
13:50:20.463 CMD: Grp 0, Cmd 'RULETIMER', Idx 1, Len 1, Pld 2, Data '2'
13:50:20.470 MQT: stat/Gasmeter_87DB0D/RESULT = {"T1":2,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
13:50:20.711 SRC: Backlog
13:50:20.713 CMD: Grp 0, Cmd 'VAR', Idx 1, Len 9, Pld -99, Data '=var1+0.1'
[... 63 lines cut ...]
13:50:38.729 SRC: Backlog
13:50:38.732 CMD: Grp 0, Cmd 'VAR', Idx 1, Len 9, Pld -99, Data '=var1+0.1'
13:50:38.739 MQT: stat/Gasmeter_87DB0D/RESULT = {"Var1":"9000.697"}
13:50:40.566 CMD: rule1 0
13:50:40.568 SRC: WebConsole from 192.168.7.37
13:50:40.571 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 1, Pld 0, Data '0'
13:50:40.577 MQT: stat/Gasmeter_87DB0D/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":120,"Free":391,"Rules":"ON System#Boot DO Backlog RuleTimer1 2; var1 9000 ENDON   ON Rules#Timer=1 DO Backlog RuleTimer1 2; var1=var1+0.1  ENDON"}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
use the rule

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
see problem description

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-10-07:
Intenal rounding. Better use integer values

sfromis, 2022-10-07:
Arithmetic in rules use single-precision floating point meaning that 7 digits or more start to be imprecise. While using integers is better to mitigate rounding issues, you will still get into trouble when wanting more than 6 digits.

Performance20, 2022-10-07:
As workaround I do it already with integer values, but I thought it makes sense to point out the issue once .