arendst/Tasmota #16751: SDM72 does not show right value for total

ChGeorgi, 2022-10-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
SDM72 does not show right value for total.
SDM 72 has 41 KWH tasmota says 39
also resattable export,import and total is missing

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:29:19.755 CMD: Status 0
21:29:19.757 SRC: WebConsole from 192.168.178.135
21:29:19.760 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
21:29:19.766 MQT: /PVZaehler/STATUS = {"Status":{"Module":18,"DeviceName":"PVZaehler","FriendlyName":["PVZaehler"],"Topic":"PVZaehler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:29:19.774 MQT: /PVZaehler/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"External System","Uptime":"0T00:31:33","StartupUTC":"2022-10-07T19:57:46","Sleep":50,"CfgHolder":4617,"BootCount":76,"BCResetTime":"2022-06-06T12:59:07","SaveCount":136,"SaveAddress":"F5000"}}
21:29:19.781 MQT: /PVZaehler/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-10-07T21:48:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"376/699"}}
21:29:19.787 MQT: /PVZaehler/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7530 ZI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
21:29:19.804 MQT: /PVZaehler/STATUS4 = {"StatusMEM":{"ProgramSize":658,"Free":344,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","879AC787","043E8001","000000CF","010013C0","C000F989","00004004","20001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6,53"}}
21:29:19.812 MQT: /PVZaehler/STATUS5 = {"StatusNET":{"Hostname":"PVZaehler-4207","IPAddress":"192.168.178.189","Gateway":"192.168.178.1","Subnetmask":"255.255.254.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"B4:E6:2D:23:10:6F","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:29:19.817 MQT: /PVZaehler/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.28","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_23106F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:29:19.825 MQT: /PVZaehler/STATUS7 = {"StatusTIM":{"UTC":"2022-10-07T20:29:19","Local":"2022-10-07T21:29:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:58","Sunset":"18:16"}}
21:29:19.830 MQT: /PVZaehler/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:29:19.839 MQT: /PVZaehler/STATUS10 = {"StatusSNS":{"Time":"2022-10-07T21:29:19","ENERGY":{"TotalStartTime":"2022-06-06T12:59:07","Total":39.333,"Yesterday":26.057,"Today":11.701,"ExportActive":0.015,"Power":297,"ExportPower": 0,"ImportPower":42,"ResetTotalActive":42,"Reset Import":42,"Reset Export":42,"TotalActive":42}}}
21:29:19.875 MQT: /PVZaehler/STATUS11 = {"StatusSTS":{"Time":"2022-10-07T21:29:19","Uptime":"0T00:31:33","UptimeSec":1893,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FRITZ!Box 7530 ZI","BSSId":"2C:91:AB:B8:61:53","Channel":11,"Mode":"11n","RSSI":100,"Signal":-24,"LinkCount":1,"Downtime":"0T00:00:05"}}}

21:29:20.003 NRG: EnergyTotal[0] 42.3210 kWh
21:29:20.256 DMP: 01 04 04 43 92 6C 5C 62 D4
21:29:20.502 DMP: 01 04 04 42 29 49 BA 88 17
21:29:20.751 DMP: 01 04 04 43 92 B5 0F 79 79
21:29:21.002 DMP: 01 04 04 80 00 00 00 D2 44
21:29:21.256 DMP: 01 04 04 42 29 3A 5E AD 6C
21:29:21.502 DMP: 01 04 04 3C 75 C2 8F F6 CA
21:29:21.752 DMP: 01 04 04 42 29 49 BA 88 17
21:29:22.001 DMP: 01 04 04 42 29 3A 5E AD 6C
21:29:22.256 DMP: 01 04 04 3C 75 C2 8F F6 CA

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ChGeorgi, 2022-10-07:
PVZaehler
Active Power		291	 	W
Energy Today		11.729	 	kWh
Energy Yesterday		26.057	 	kWh
Energy Total		39.361	 	kWh
Export Active		0.015	 	kWh
Export Power		0	 	W
Import Power		42	 	W
Configuration
Information
Firmware Upgrade
Consoles
Restart
[Tasmota 12.1.1 by Theo Arends](https://bit.ly/tasmota)

arendst, 2022-10-08:
> SDM72 does not show right value for total.

Enable hardware totals with command ``SetOption72 1``

> also resattable export,import and total is missing

SDM72 driver does not support these registers. Only SDM230 driver does support these registers.

closed 2022-10-17 by arendst