arendst/Tasmota #16763: compilation of DS3231 module not supported ?

lolo888, 2022-10-08:
### PROBLEM DESCRIPTION
Hello, I am trying to compile from the source code in order to include an RTC DS3231 module on a Sonoff 1CH, despite thirty attempts the binary never wants to take this module into account.
I was able to test both compilers online, either Tasmocompiler or on GitPod. (here is the Gitpod share link: https://arendst-tasmota-jnyb4f1q16j.ws-eu70.gitpod.io )

The Sonoff does not recognize the module when it is visible with the I2CScan command in the console

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
#ifndef USE_RTC_CHIPS
#define USE_RTC_CHIPS
#endif
#ifndef USE_DS3231
#define USE_DS3231
#endif
#ifndef USE_AM2320
#define USE_AM2320
#endif
#ifndef USE_DISPLAY_SSD1306
#define USE_DISPLAY_SSD1306
#endif

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8285
00:00:00.050 CFG: Chargé de la flash à F8, Compte 2555
00:00:00.055 QPC: Count 1
00:00:00.070 Projet tasmota - Sonoff-Water Version 12.1.1(tasmota)-2_7_4_9(2022-10-08T17:40:59)
00:00:00.083 CMD: 
00:00:00.530 WIF: Connexion à l'AP1 xxxxxxxxxx Channel 3 BSSId xxxxxxxxxxxxxxxxxx en mode 11n comme Sonoffxxxxxxxxx...
00:00:01.802 WIF: Connecté
00:00:02.011 HTP: Serveur web actif sur Sonoffxxxxxxxxxxxxxx avec l'adresse IP 192.168.xxxxxx
00:00:02.014 RSL: INFO1 = {"Info1":{"Module":"1 Channel","Version":"12.1.1(tasmota)","FallbackTopic":"cmnd/DVES_xxxxxxx_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:02.027 RSL: INFO2 = {"Info2":{"WebServerMode":"xxxxxxxxxxxx","Hostname":"Sonoffxxxxxxxxxxxxxx","IPAddress":"192.168.xxxxxxx"}}
00:00:02.037 RSL: INFO3 = {"Info3":{"RestartReason":"Power On","BootCount":251}}
00:00:02.044 RSL: RESULT = {"POWER":"OFF"}
00:00:02.046 RSL: POWER = OFF
21:26:33.524 QPC: Reset
21:26:33.529 RSL: STATE = {"Time":"2022-10-08T21:26:33","Uptime":"0T00:00:08","UptimeSec":8,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxxx","BSSId":"xxxxxxxxxxxxxxxxxx","Channel":3,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:26:41.258 CMD: I2Cscan
21:26:41.277 RSL: RESULT = {"I2CScan":"Device(s) found at 0x68"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-10-08:
Your logs doesn't seem to have Weblog 4 enabled
Please set as requested, restart tasmota and update your logs

lolo888, 2022-10-08:
there it's done

> Vos journaux ne semblent pas avoir Weblog 4 activé Veuillez définir comme demandé, redémarrez tasmota et mettez à jour vos journaux

barbudor, 2022-10-08:
> 22:00:00.917 RSL: STATUS4 = {"StatusMEM":{... "Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62", ...

You seem to have every drivers enabled which is not great as some drivers may claim the address 0x68 before the DS3231's driver.

In order to optimize Tasmota execution you should disable every driver that you are not intending to use with [`I2Cdriver<num> 0`](https://tasmota.github.io/docs/I2CDEVICES/#management)

At least you should disable driver 25 (MPU6050) and 58 (MPU_ACCEL) using `backlog i2cdriver25 0; i2cdriver58 0`
Let me know if this makes a change

barbudor, 2022-10-08:
Checking your Status output, your binary neither includes USE_DS3231 nor USE_SSD1306

Please check how you are building
And learn how to use `decode-status` to check what is inside your binary (once uploaded)
It is located here  https://github.com/arendst/Tasmota/tree/development/tools
and doc is there : https://github.com/tasmota/decode-status

lolo888, 2022-10-08:
When I send the command "i2cdriver58 0" the Sonoff reboot immediately, I have almost no time to copy the console
 CMD: backlog i2cdriver25 0
22:30:18.840 SRC: WebConsole from 192.168.x.x
22:30:18.843 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 13, Pld -99, Data 'i2cdriver25 0'

26:08.075 CMD: i2cdriver58 0
22:26:08.076 SRC: WebConsole from 192.168.x.x
22:26:08.078 CMD: Grp 0, Cmd 'I2CDRIVER', Idx 58, Len 1, Pld 0, Data '0'
22:26:08.081 RSL: RESULT = {"I2CDriver":"7,26,60"}

and the module is still not supported in the console when it reboots

barbudor, 2022-10-08:
reboot is normal after command i2cdriver

But as my just previous message stated, your build is faulty, no DS3132 in your binary

sfromis, 2022-10-08:
> At least you should disable driver 25 (MPU6050) and 58 (MPU_ACCEL) using

FTR, the `Status 0` output does not include which I2C drivers are enabled, the `"Drivers"` values are instead matching the driver file names, like driver number 56 for `xdrv_56_rtc_chips.ino`

lolo888, 2022-10-08:
> 

I tried to follow the instructions to the letter to compile in gitpod or tasmocompiler, I don't understand why it doesn't work

barbudor, 2022-10-08:
In TasmoCompiler, what boxes did you checked ?
Did you entered the #define in the last text section before launching compile ?
You may try more simply wihtout the #ifdef
```
#define USE_RTC_CHIPS
#define USE_DS3231
#define USE_AM2320
#define USE_DISPLAY_SSD1306
```

sfromis, 2022-10-08:
AFAICT, barbudor, the `Status 0` output indicates that both
```
#define USE_RTC_CHIPS
#define USE_DS3231
```
were in force, but not `USE_DISPLAY_SSD1306` as the display support requires further `#define`

When having a working DS3231 connected, my boot log indicate that the RTC chip was successfully detected, log starts with:
```
00:00:00.002 HDW: ESP8266EX
00:00:00.051 CFG: Loaded from flash at F6, Count 5126
00:00:00.060 I2C: DS3231 found at 0x68
00:00:00.065 RTC: UTC 2022-10-08T20:29:24, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
22:29:24.000 RTC: Synced by DS3231
22:29:24.005 ROT: Mode 1
22:29:24.007 Project tasmota
```

barbudor, 2022-10-08:
Must have made a mistake with my grep, indeed you right
For the SSD1306 it is missing USE_DISPLAY