arendst/Tasmota #16764: Node ESP6288 Led off state sporadic small flash in the dark

enricota66, 2022-10-08:
### PROBLEM DESCRIPTION
Node ESP6288 Led off state sporadic small flash in the dark

### REQUESTED INFORMATION

- [ ] Device used : Node ESP6288 Node ESP6288 
- [ ] Tasmota binary firmware version number used: 12.1.1.3

  - [ ] Self-compiled on Atom on  linux debian ( fully update) 
- [ ] Flashing tools used:  OTA Upgrade file upload

Configuration output here:
{"NAME":"NodePP2","GPIO":[1,1,1,1,1,1,0,0,1,1,1,1,0,0],"FLAG":0,"BASE":18}

Rules output here:
22:50:37.217 CMD: Backlog Rule1; Rule2; Rule3
22:50:37.242 MQT: stat/NodePP2/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":105,"Free":406,"Rules":"on switch4#state do publish stat/NodePP2/MOTION %value% endon on Switch4#State do LedPower1 %value% endon"}}
22:50:37.621 MQT: stat/NodePP2/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":42,"Free":469,"Rules":"on Power4#state do LedPower1 %value% endon"}}
22:50:38.019 MQT: stat/NodePP2/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

STATUS 0 output here:
22:51:47.657 MQT: stat/NodePP2/STATUS = {"Status":{"Module":0,"DeviceName":"NodePP2","FriendlyName":["Lampada","Ciabatta","Studio","Notturna"],"Topic":"NodePP2","ButtonTopic":"0","Power":4,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,3,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
22:53:46.908 CMD: weblog 4
22:53:46.915 MQT: stat/NodePP2/RESULT = {"WebLog":4}
22:53:47.581 CFG: Saved to flash at F9, Count 22461, Bytes 4096
22:53:48.316 DHT: Pin1 read 02CF00E1B2
22:53:52.316 DHT: Pin1 read 02CE00E0B0
22:53:56.320 DHT: Pin1 read 02CE00DFAF
22:53:59.915 WIF: Checking connection...
22:54:00.327 DHT: Pin1 read 02CF00E1B2
22:54:02.113 UDP: Packet (90)
22:54:02.115 UDP: HUE
22:54:02.123 UPP: Hue 3 response packets sent to 192.168.0.32:36309
22:54:02.125 UDP: Packet (90)
22:54:04.323 DHT: Pin1 read 02CF00E1B2
22:54:08.323 DHT: Pin1 read 02CF00E2B3
22:54:12.323 DHT: Pin1 read 02CF00E2B3
22:54:13.517 HTP: Hue API (/1476dd/lights/21458388) from 192.168.0.11
22:54:13.521 /lights path=/lights/21458388
22:54:13.529 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Notturna","modelid":"NodePP2","manufacturername":"Tasmota","uniqueid":"d8:bf:c0:14:76:dd:00:11-04"})
22:54:16.325 DHT: Pin1 read 02CF00E2B3
22:54:17.318 WIF: Sending Gratuitous ARP
22:54:19.919 WIF: Checking connection...
22:54:20.116 UDP: Packet (119)
22:54:20.118 UDP: HUE
22:54:20.125 UPP: Hue 3 response packets sent to 192.168.0.28:6224
22:54:20.127 UDP: Packet (119)
22:54:20.327 DHT: Pin1 read 02CF00E4B5
22:54:23.723 HTP: Hue API (/1476dd/lights/21458388) from 192.168.0.12
22:54:23.726 /lights path=/lights/21458388
22:54:23.734 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Notturna","modelid":"NodePP2","manufacturername":"Tasmota","uniqueid":"d8:bf:c0:14:76:dd:00:11-04"})
22:54:24.330 DHT: Pin1 read 02CD00DFAE
22:54:28.330 DHT: Pin1 read 02CE00E1B1
22:54:32.332 DHT: Pin1 read 02CE00E1B1
22:54:33.322 UDP: Packet (90)
22:54:33.324 UDP: HUE
22:54:33.332 UPP: Hue 3 response packets sent to 192.168.0.30:35815
22:54:33.333 UDP: Packet (90)
22:54:36.333 DHT: Pin1 read 02CF00E3B4
22:54:39.928 WIF: Checking connection...
22:54:40.335 DHT: Pin1 read 02CD00DFAE
22:54:40.528 HTP: Hue API (/1476dd/lights/21458387) from 192.168.0.12
22:54:40.531 /lights path=/lights/21458387
22:54:40.539 HTP: Hue Result ({"state":{"on":true,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Studio","modelid":"NodePP2","manufacturername":"Tasmota","uniqueid":"d8:bf:c0:14:76:dd:00:11-03"})
22:54:40.929 HTP: Hue API (/1476dd/lights/21458387) from 192.168.0.13
22:54:40.932 /lights path=/lights/21458387
22:54:40.940 HTP: Hue Result ({"state":{"on":true,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Studio","modelid":"NodePP2","manufacturername":"Tasmota","uniqueid":"d8:bf:c0:14:76:dd:00:11-03"})
22:54:44.338 DHT: Pin1 read 02CD00E0AF
22:54:48.339 DHT: Pin1 read 02CD00E0AF
22:54:52.342 DHT: Pin1 read 02CF00E3B4
22:54:56.343 DHT: Pin1 read 02CD00E0AF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-10-09:
You did not give any details about what you even mean, what pins, what led you see a "flash" on, or what board you use.

If what you are talking about is the ESP8266-based NodeMCU, those boards generally have hardwired active low leds on gpio 2 and 16. If you use those gpio pins for other purposes, expect that the leds may show activity or status.

barbudor, 2022-10-09:
Some Esp12 modules also have a led on serial TX. So the only way to stop serial activity is SerialLog 0

enricota66, 2022-10-09:
> You did not give any details about what you even mean, what pins, what led you see a "flash" on, what board you use, or circumstances when it happens.
> 
> If what you are talking about is the ESP8266-based NodeMCU, those boards generally have hardwired active low leds on gpio 2 and 16. If you use those gpio pins for other purposes, expect that the leds may show activity or status.
The card I am using is a nodemcu chips esp8266 the pin that powers the led as shown in the photo and the D3 / GPIO0. It has always been the same configuration since Tasmota 9.xxx.  Only with the 12xx version did I notice that there are small interference voltages that power the LED with small flashes visible only in deep darkness.  These interferences are not related to any activity or event

enricota66, 2022-10-09:
> Some Esp12 modules also have a led on serial TX. So the only way to stop serial activity is SerialLog 0

This is very interesting in fact there have always been log pulses for events.  But as I wrote above the disturbance that I noticed is not this. Anyway thanks for the info

sfromis, 2022-10-09:
If what you want is light, it is not appropriate to config the pin as "Led", as this indicates a status led, which by default can light up to indicate wifi activity, like when publishing at TelePeriod. To have full control over the led, it is better to either config it as a Relay and use Power commands for on/off, or as a PWM which additionally allows you to control brightness. Alternatively, you also have options to partially control how Tasmota handles the status led.

enricota66, 2022-10-09:
> If what you want is light, it is not appropriate to config the pin as "Led", as this indicates a status led, which by default can light up to indicate wifi activity, like when publishing at [...]

Thanks for the suggestions I will try, in fact now configured as LED I get on signals with a single flash when I switch the others relays configured on the same board.  But what I wanted to point out here.  It is the strangeness of these electrical 'interference' in the LED.  Which I had not encountered with all previous versions.  Now I am testing version 11.1.04 which seems to me to be the latest of the 11.xx series.  I will  try with the suggested configuration.

sfromis, 2022-10-09:
As I explained, having it as a status led will by default blink on wifi activity, which is normal during operation, not just when you push buttons. This situation is nowhere near "electrical inteference", just how it works depending on your config.

Of course, all depends on what you want to achieve, I was just telling you that blinks on a status led are normal even when *you* are not yourself doing anything. What I mentioned about reconfiguring the pin was because it looked like you were wanting to use it to provide light, but if that's not the case, forget it.

enricota66, 2022-10-10:
I have changed the configuration as advised. Now I turn on the LED as if it were a relay.
 I use this as a night light combined with an infrared sensor, no disturbances are seen.
For this I think we can close here

closed 2022-10-10 by enricota66