arendst/Tasmota #16771: Version 12.1.1 NOT sending MQTT autoconfig.

dss65MI, 2022-10-09:
Hello,
I upgraded from 10.1.0 to 12.1.0 but I noticed that verstion 12.1.1 does not send autoconfig messages.
Therefore autodiscovery on Home Assistant doesn't work.
At startup 10.1.0 sends:
14:16:03.504 MQT: homeassistant/switch/xxxxxx_RL_1/config = {"name":" etc. etc.
While 12.1.0 does not.
I am going back to 10.1.0 but I think you should know.
Thanks for the INCREDIBLE work you did.

closed 2022-10-09 by close-issue-app[bot]

barbudor, 2022-10-09:
Home Assistant discovery has been deprecated in favor of Tasmota Discovery
You need to install HaTasmota plugin on Home Assistant

nikito7, 2022-10-18:
SetOption19 0

barbudor, 2022-10-18:
The old home assistant intégration is now deprecated.
You should use HATASMOTA plug-in in your HA with Setoption19 0

nikito7, 2022-10-18:
"HaTasmota" is the python library :-)

[image]