arendst/Tasmota #16775: eQ-3 TRV driver not compatible with newer TRV firmware versions (e.g. 1.46)

AleXSR700, 2022-10-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The Tasmota eQ-3 TRV driver actively polls the status of the BLE TRVs at regular intervals  or sends commands when issued by the user.
For this to work a bluetoot connection needs to be established with the TRV.

This works as expected for older firmware versions such as 1.10 or 1.20.

With newer firmware (e.g. 1.46), the manufacturer implemented a mandatory 6 digit pairing key. Pairing is no longer possible without the key.
This causes the integration to stop working properly and results in error messages
```:57:22.679 EQ3 001A2214A73A: trv operation failed - retrying -12
09:57:24.688 EQ3 001A2214A73A: trv operation failed - retrying -12
09:57:26.696 EQ3 001A2214A73A: trv operation failed - retrying -12
09:57:28.660 EQ3 001A2214A73A: trv operation failed - no more retries -12
09:57:42.677 EQ3 001A2214ADBF: trv operation failed - retrying -12
09:57:45.666 EQ3 001A2214ADBF: trv operation failed - retrying -12
09:57:51.673 EQ3 001A2214ADBF: trv operation failed - retrying -12
09:57:53.679 EQ3 001A2214ADBF: trv operation failed - no more retries -12
09:58:08.690 EQ3 001A2214A6DF: trv operation failed - retrying -12
09:58:11.658 EQ3 001A2214A6DF: trv operation failed - retrying -12
09:58:16.674 EQ3 001A2214A6DF: trv operation failed - retrying -12
09:58:19.689 EQ3 001A2214A6DF: trv operation failed - no more retries -12
09:58:32.696 EQ3 001A2214B86E: trv operation failed - retrying -12
09:58:35.658 EQ3 001A2214B86E: trv operation failed - retrying -12
09:58:37.667 EQ3 001A2214B86E: trv operation failed - retrying -12
09:58:39.677 EQ3 001A2214B86E: trv operation failed - no more retries -12
09:58:43.691 EQ3 poll overrun, deferred - last loop only got to 8, not 16
10:00:11.967 BLE: failed to connect to device
10:00:12.670 EQ3 001A2214ADC8: trv operation failed - retrying -11
10:00:14.675 EQ3 001A2214ADC8: trv operation failed - retrying -12
10:00:16.678 EQ3 001A2214ADC8: trv operation failed - retrying -12
10:00:18.684 EQ3 001A2214ADC8: trv operation failed - no more retries -12
10:00:43.677 EQ3 poll overrun, deferred - last loop only got to 5, not 16```

The pairing key seems to be a constant (dooes not seem to change when connecting a new device with the same key).

So maybe it would be possible to introduce a new command that assigns a pairing key per device MAC.
e.g.
```TRVkey 001A5678=123456```
For older firmware versions no key is needed, but since it is a constant 4 digit key it could just as easily be provided.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): eQ-3 (CC-RT-BLE)
- [ x] Tasmota binary firmware version number used: 12.1.4 (but also 12.1.1 and 10.0.0)
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try to connect an eQ-3 TRV with firmware version 1.46 or newer.

Pairing key obtainable by holding the dial key for a few seconds.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Successful BLE onnection.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-10-09:
Do you know what we should do with that key ?

AleXSR700, 2022-10-09:
Ideally use it to pair with the TRV
https://github.com/rytilahti/python-eq3bt#pairing

But how this can be implemented in this driver, I do not know :(
My coding skills are too poor.

barbudor, 2022-10-09:
Is that the key you have to enter there ?
```
[agent] Enter passkey (number in 0-999999): <enter pin>
```

AleXSR700, 2022-10-09:
The 6 digit key, yes.

It is unique to each device but seems to be a constant.

barbudor, 2022-10-09:
The difficulty is now finding where in the code this should be provided for the pairing....

AleXSR700, 2022-10-09:
If there is anything you would like me to test, just let me know. I will try to supply whatever information I can.

barbudor, 2022-10-09:
@btsimonh Are you available to have a look at that or give directions how to implement it in the EQ3 driver ?
thanks

btsimonh, 2022-10-10:
It actually does not look to hard.

[Nimble Docs](https://h2zero.github.io/NimBLE-Arduino/class_nim_b_l_e_client_callbacks.html#a9850764aec546747537c6baa2a4622f5)

Two things are needed:

1/ to store a 32 bit number per device (the key, in decimal).
This should probably be a BLE config against MAC address, rather than an EQ3 thing, so that it would work for any device with static key.

2/ to add:

```
class BLESensorCallback : public NimBLEClientCallbacks {
  ... existing callbacks

  uint32_t onPassKeyRequest	( ) {
   // find if a key has been associated with the MAC
   // return the key for the device we are trying to connect to
   return foundKey;
  }	
}
```

It's a pity we don't get a pClient in the call to onPassKeyRequest	( ) - so we'll need to know what connection is currently in progress, and hope we don;t have two in progress at the same time (maybe we should be double sure we can't - although the ESP claims that it can have up to 3 (?), probably not a good idea to have two connection attempts ongoing at the same time...

If done at the BLE driver, then should be testable against any BLE device with a static pin.

I don't see any decent way deal with a dynamic pin - since the callback is synchronous, I'm not sure we can, for example, have someone enter it in the web UI dynamically. @h2zero - thoughts?

@sfromis - any additional ideas?

btsimonh, 2022-10-10:
A really hacky first test could be done by prefixing an the alias with a pin, and setting a 'currentPinCode' variable at the time of connection request by alias?

h2zero, 2022-10-10:
> It's a pity we don't get a pClient in the call to onPassKeyRequest ( )

Good call, this should be added :)

> I don't see any decent way deal with a dynamic pin - since the callback is synchronous, I'm not sure we can, for example, have someone enter it in the web UI dynamically. @h2zero - thoughts?

Unfortunately you are correct, the security handling needs some work to make it async capable. This has been brought up before and has been on my mind to do but I have been too busy lately.

AleXSR700, 2022-10-16:
May I ask: will the update/fix be postponed until dynamic PINs are supported by Tasmota or will there be an intermediate fix? The TRVs do not require handling of dynamic PIN as the PIN is static but I of course understand if you would not want to have to touch the code twice :)

barbudor, 2022-10-16:
I'm trying to add `BLEkey` command following @btsimonh advice
Crossing finger to have something to test soon

@AleXSR700 will a standard `tasmota32-bluetooth.bin` be enough or do you need to self-compile ?

barbudor, 2022-10-16:
> > It's a pity we don't get a pClient in the call to onPassKeyRequest ( )
> Good call, this should be added :)

@btsimonh @h2zero 
So how do I get the MAC within `onPassKeyRequest()` if client is not provided ?
Should we first raise a PR to NimBLE and then upgrdae Tasmota ?

Or should I for now consider a single key, not assigned to a MAC
Would it work if we go device per device and type a `BLEkey <thekey>` command just before pairing a device ?

AleXSR700, 2022-10-16:
> @AleXSR700 will a standard `tasmota32-bluetooth.bin` be enough or do you need to self-compile ?
> 
Standard `tasmota32-bluetooth.bin` will do just fine, thank you :)

> Or should I for now consider a single key, not assigned to a MAC
> Would it work if we go device per device and type a `BLEkey <thekey>` command just before pairing a device ?
> 
I am less familiar with this of course than the others, but I think each device needs to have a key assigned because you need to connect and disconnect regularly and I am not sure the TRV will remember the client. So MAC assignment would probably be best, if somehow possible.
But I can of course test if client remembers if assignment is not possible.

barbudor, 2022-10-16:
So here is a binary of tasmota32-bluetooth with a new command `BleKey <key>` like `BleKey 1234`
It's not associated to a MAC so I would assume you have to enter it when trying to pair the TRV and it will be provided to any BLE device asking for pairing at that time. Once pairing is done, I don't expect the key to be requested anymore for any further reconnection. Unless the pairing is lost by pairing with another device like a mobile phone.
With WebLog 3, you should see a log similar to `BLE: onPassKeyRequest 1234` when the key is passed to a device
Let me know if this is doing anything good
[tasmota32-bluetooth-blekey.zip](https://github.com/arendst/Tasmota/files/9795548/tasmota32-bluetooth-blekey.zip)

We will see later how we can improve and attach the key to a MAC if needed

AleXSR700, 2022-10-17:
Thank you @barbudor ,
I upgraded to your blekey version and set blekey to a known pairing key. But I never got a `BLE: onPassKeyRequest 1234` message.
Would it be possible to implement a print showing me that a key was requested and that the key was sent so I can check that for you?

barbudor, 2022-10-17:
the message I listed is exactly the one you are looking for. It would be printed in request for a key and confitm the exact key sent in return

Correct no Mac and the value is on a Uint32 so between 0 and 2^32-a

AleXSR700, 2022-10-21:
Any news on a debug version or would more detailed information not be useful to you?

barbudor, 2022-10-21:
I've done what I could do.
The call back is not fired and I don't know why
This needs someone with more knowledge on the working of NimBLE to take over

AleXSR700, 2022-10-22:
I just reverted to a file I found on my NAS and with that version the TRVs work even with new firmware.

Is it maybe possible to "reverse engineer" why it worked back then?
```
Program Version	10.0.0.4(bluetooth)
Build Date & Time	2021-12-06T20:54:01
Core/SDK Version	2_0_1_1/v4.4-dev-3728-g7cac8278e
```
[tasmota32-bluetooth_eq3.zip](https://github.com/arendst/Tasmota/files/9844057/tasmota32-bluetooth_eq3.zip)

I think @btsimonh shared this as a testfile back in the day, but I am not sure. But it works with the TRVs.

Einstein67, 2022-10-22:
Tasmota 12.0.2.4 .... the latest version with nimBLE 1.3.6 works fine with all EQ3-Firmware-Versions.

With nimBLE 1..4.0 the problem begins ...

barbudor, 2022-10-22:
@Jason2866 Could there be something wrong with 1.4.0 ?

barbudor, 2022-10-22:
@h2zero Can you help please ?

h2zero, 2022-10-22:
Looks to me that your code changes left out the security mode setting required for passkey authentication.
Try adding these lines after initializing NimBLE:
```
NimBLEDevice::setSecurityAuth(false, true, true); // make these all true if you want to enable bonding
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY); // use passkey authentication
```

If you want to force security on connection you can call `pClient->secureConnection();` in the `onConnect` callback.

barbudor, 2022-10-22:
Thanks @h2zero , I will try
Any reason why it seems to work without any change using NimBLE 1.3.6 and breaks with NimBLE 1.4.0 ?
Does NimBLE 1.4.0 enforce something that need the above calls to work ?

h2zero, 2022-10-22:
I honestly don't know, nothing I can think of was changed that would affect this. The NimBLE core was updated and could have something in it that caused this.

barbudor, 2022-10-22:
@AleXSR700 Please find an update which includes the changes recommended by h2zero
Still works fine with my Mija sensor running PVVX firmware

Hopefully it will help
If not, I have no idea what to do next
[tasmota32-bluetooth-blekey2.zip](https://github.com/arendst/Tasmota/files/9845233/tasmota32-bluetooth-blekey2.zip)

Question: Regarding your tests, you mean that 12.1.1 and previous are working fine with latest TRV firmware even without providing a key ?
So we may look into the wrong direction

AleXSR700, 2022-10-23:
> @AleXSR700 Please find an update which includes the changes recommended by h2zero
> Still works fine with my Mija sensor running PVVX firmware
>
> Hopefully it will help
> If not, I have no idea what to do [...]

Thank you, but it still is not working.
By the way, for pairing keys with leading zero, the zero is cut from the variable when it is stored.

> Question: Regarding your tests, you mean that 12.1.1 and previous are working fine with latest TRV firmware even without providing a key ?
> So we may look into the wrong direction

For 12.1.1 it works without any pairing key. So whatever was different there made TRVs work without having to enter any keys. 
So it looks like the options are
a) get keys to work with current version's NimBLE (or whatever else is the cause)
or 
b) change the code (or settings) of NimBLE (or whatever else is the cause)

h2zero, 2022-10-23:
I'm not familiar with the code here but to understand this problem it would be very helpful to see the log output from NimBLE. If such logs can be provided I would be happy to look into them and resolve any related issues with the library.

AleXSR700, 2022-10-23:
> I'm not familiar with the code here but to understand this problem it would be very helpful to see the log output from NimBLE. If such logs can be provided I would be happy to look into them and resolve any related issues with the library. 

How would I get those logs?

h2zero, 2022-10-23:
You should be able to add a compile flag `-DCONFIG_NIMBLE_CPP_LOG_LEVEL=5` to get the log output.

AleXSR700, 2022-10-23:
Maybe I built it incorrectly.
```build_flags                 = ${env:tasmota32_base.build_flags}
                              -D FIRMWARE_TASMOTA32
                              -D USE_WIFI_RANGE_EXTENDER
                              -D USE_WIFI_RANGE_EXTENDER_NAPT
                              -D CONFIG_NIMBLE_CPP_LOG_LEVEL=5```
in platformio_tasmota_cenv.ini.
Or do I need to then enable it somehow? Neither with weblog 2 nor 4 did I get a different console output.
Or maybe only working via syslog?

sfromis, 2022-10-23:
You should not blindly replicate the example I posted, which was for another context. It was **only** meant to illustrate a technique for setting build flags. Details absolutely has to match what you're trying to build.

AleXSR700, 2022-10-23:
> You should not blindly replicate the example I posted, which was for another context. It was **only** meant to illustrate a technique for setting build flags. Details absolutely has to match what you're trying to build.

That's the default. I only added the last line, which I understood was a working approach, sorry.

AleXSR700, 2022-10-23:
> You should not blindly replicate the example I posted, which was for another context. It was **only** meant to illustrate a technique for setting build flags. Details absolutely has to match what you're trying to build.
I simply added one line to the default. But probably at the wrong place, yes.

AleXSR700, 2022-10-23:
I don't know if I built the firmware correctly and this now includes the information you are looking for. If not, would you maybe upload me a test firmware?

This is the log I got from seriallog with log level 4.
I let it run until both TRVs were found (one connected successfully and the other did not).

001A2214ADC8: Firmware 1.46, not working in latest firmware anymore (new 6 digit pseudo-mandatory key)

001A2208C9FA: Firmware 1.10, working in all firmwares (old 4 digit key)

[putty.log](https://github.com/arendst/Tasmota/files/9846224/putty.log)

h2zero, 2022-10-23:
Thanks, those logs were perfect. I can see the authentication being triggered when subscribing to the characteristic notifications and the event occurs but no callback. I'll investigate further.

h2zero, 2022-10-23:
I'm attaching some modified NimBLE files, please copy and paste these into the NimBLE folder and build/log again. This will provide more information to diagnose the issue.

[nimble-enhanced-logs.zip](https://github.com/arendst/Tasmota/files/9846929/nimble-enhanced-logs.zip)

AleXSR700, 2022-10-23:
Thank you.

Attached the new log.

[putty_ftdi_12.2.0.1_nimble_v2.log](https://github.com/arendst/Tasmota/files/9846988/putty_ftdi_12.2.0.1_nimble_v2.log)

AleXSR700, 2022-10-23:
So do you need me to build from barbudor's repo with your numble files?

h2zero, 2022-10-23:
That would give some more clues for sure if you could.

AleXSR700, 2022-10-23:
Log with @barbudor build

[putty_ftdi_12.2.0.1_barbudor.log](https://github.com/arendst/Tasmota/files/9847046/putty_ftdi_12.2.0.1_barbudor.log)

h2zero, 2022-10-23:
I think you missed adding the log flags for the build.

AleXSR700, 2022-10-23:
Yes, that's why I deleted my post. Will post again in 5 min :)

AleXSR700, 2022-10-23:
Here we go :)

[tasmota32-bluetooth_12.2.0.1_barbudor_nimble.log](https://github.com/arendst/Tasmota/files/9847063/tasmota32-bluetooth_12.2.0.1_barbudor_nimble.log)

h2zero, 2022-10-23:
Interesting, thanks! Still the same error.. so I need to know more about the device. Does it have a display? If you connect to it with a app like nRFConnect and subscribe to the characteristic what happens?

AleXSR700, 2022-10-23:
Device: https://www.eq-3.de/Downloads/eq3/downloads_produktkatalog/eqiva/bda/CC-RT-BLE-EQ_UM_GE_eQ-3_160428.pdf

Interestingly the pairing key has now change on the device I am testing.

h2zero, 2022-10-23:
Are you sure you compiled with barbudor's latest changes? I'm not seeing the call to secureconnection from the onconnect event in the log that should be happening with that.

AleXSR700, 2022-10-23:
Sorry, no, I was in dev branch and thought it was in there. But there is a dedicated BleKey branch.

AleXSR700, 2022-10-23:
Here the new log

[putty_ftdi.log](https://github.com/arendst/Tasmota/files/9847148/putty_ftdi.log)

I let it run for a while and then also tried entering the BleKey.

h2zero, 2022-10-23:
Thanks, now we're getting somewhere. Give this file a try.
[NimBLEClient.cpp.txt](https://github.com/arendst/Tasmota/files/9847172/NimBLEClient.cpp.txt)

h2zero, 2022-10-23:
@AleXSR700 I was able to reproduce this locally as well, the full fix will require some minor changes to NimBLE.

barbudor, 2022-10-23:
Great news. Many thanks
Will my change with BLE Key still required as it looks v1.3.6 it works with TRV without the key?

h2zero, 2022-10-23:
Yes, the changes made will likely be necessary. The problem was the device requests pairing as soon as it's connected to, then when trying to subscribe the security fails because it's already in progress. I'll fix the handling for this part in NimBLE, but the other changes that set the IOCAPS and MITM setting + passkey callback will be required. I would suggest enabling bonding as well so the pairing process doesn't need to be repeated each time since it looks like they key does change.

barbudor, 2022-10-23:
What about the other change to get pClient as an argument to the onPassKeyRequest callback so we can save one key per device ?
Does it actually makes sense to save the key and match it to a MAC if it is not required later ?

h2zero, 2022-10-23:
I don't think it makes much sense to save the key since it can change at any time. If bonded the key will only need to be needed once anyway, but I see you have NVS disabled here so that could be problematic.

h2zero, 2022-10-23:
Here is the full fix, if you could please test this and let me know if it works. I will push this to the library once confirmed.

[NimBLE-pairing-fix.zip](https://github.com/arendst/Tasmota/files/9847370/NimBLE-pairing-fix.zip)

AleXSR700, 2022-10-23:
Does it matter whether I tried it in @barbudor's repo or should I try directly in tasmota master?

h2zero, 2022-10-23:
You'll need to use barbudor's changes.

h2zero, 2022-10-23:
@barbudor I will also add the client reference to the callback separately to this in another commit as it should be done anyway.

[33 more replies]

closed 2022-10-24 by arendst