arendst/Tasmota #16782: SetOption86 PWM Dimmer toggle with hard buttons

heffneil, 2022-10-10:
### PROBLEM DESCRIPTION
There is a hard button sequence which will re-enable the timeout of the brightness LED's. We have found in daily usage that this combination is very easy for end-users to re-enable the timeout inadvertently.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
This is on the Martin Jerry Dimming Switch.  I am running the latest version but I believe it applies to 10+ versions

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:12:29.194 CMD: STATUS 0
14:12:29.196 SRC: WebConsole from 172.16.1.64
14:12:29.199 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
14:12:29.205 MQT: stat/Kitchen_Island/STATUS = {"Status":{"Module":73,"DeviceName":"Kitchen Island Remote","FriendlyName":["Kitchen Island Remote"],"Topic":"Kitchen_Island","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:12:29.212 MQT: stat/Kitchen_Island/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T18:50:03","StartupUTC":"2022-10-09T18:22:26","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-02-15T03:40:06","SaveCount":2437,"SaveAddress":"F7000"}}
14:12:29.217 MQT: stat/Kitchen_Island/STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022-04-13T06:40:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"418/699"}}
14:12:29.222 MQT: stat/Kitchen_Island/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C80001000600003C5A0A190000000000","00000080","00006078","00004000"]}}
14:12:29.234 MQT: stat/Kitchen_Island/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,56,62","Sensors":"1,2,3,4,5,6"}}
14:12:29.240 MQT: stat/Kitchen_Island/STATUS5 = {"StatusNET":{"Hostname":"Kitchen-Island-6233","IPAddress":"172.16.2.126","Gateway":"172.16.0.1","Subnetmask":"255.255.0.0","DNSServer1":"172.16.0.253","DNSServer2":"192.168.10.251","Mac":"DC:4F:22:7A:F8:59","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:12:29.245 MQT: stat/Kitchen_Island/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.0.253","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7AF859","MqttUser":"XXXXXX","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:12:29.254 MQT: stat/Kitchen_Island/STATUS7 = {"StatusTIM":{"UTC":"2022-10-10T13:12:29","Local":"2022-10-10T14:12:29","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:03","Sunset":"18:10"}}
14:12:29.260 MQT: stat/Kitchen_Island/STATUS10 = {"StatusSNS":{"Time":"2022-10-10T14:12:29"}}
14:12:29.266 MQT: stat/Kitchen_Island/STATUS11 = {"StatusSTS":{"Time":"2022-10-10T14:12:29","Uptime":"0T18:50:03","UptimeSec":67803,"Heap":21,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":3,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"XXXXXXXX","Channel":11,"Mode":"11n","RSSI":100,"Signal":-35,"LinkCount":3,"Downtime":"0T00:00:44"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
No output to provide for this behavior

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
"Tediore — 10/08/2022
Okay so funny enough, I just did it on another dimmer. It's just really easy to hit both the bottom and middle buttons at the same time"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shouldn't toggle a set option so easily from the user interface (buttons on actual hardware)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

heffneil, 2022-10-10:
While I believe this is, in fact, designed I guess I am asking for a feature request to disable the option since it seems that a couple of users have inadvertently changed the option with the hardbuttons.  I would like an option to disable that or make that fat-fingering the option change to be more challenging so daily use isn't the culprit.

heffneil, 2022-10-20:
Any update here with an option to disable the physical buttons from changing this setoption? Thanks!

heffneil, 2022-11-06:
Any thoughts on this?

Jason2866, 2022-11-24:
Moving to discussions as an idea, since it is an feature request.