arendst/Tasmota #16785: Nous A5T wrong daily/total kWh

bdkacz, 2022-10-10:
### PROBLEM DESCRIPTION

I have Smart Power Strip Nous A5T with Tasmota. Latest verison 12.1.1.0 and I have problem with wrong  daily/total kWh. The current,power and voltage is  correct (calibrated),but increasing of kWh i incorrect

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Nous A5T
- [ x] Tasmota binary firmware version number used: 12.1.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: preinstalled
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
 
stat/tasmota_DC3757/RESULT = {"NAME":"Nous_A5T","GPIO":[0,3072,544,3104,0,259,0,0,225,226,224,0,35,4704],"FLAG":0,"BASE":18}
 stat/tasmota_DC3757/RESULT = {"Module":{"0":"Nous_A5T"}}

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

20:18:33.433 MQT: stat/tasmota_DC3757/STATUS = {"Status":{"Module":0,"DeviceName":"NousA5T","FriendlyName":["NousA5T","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_DC3757","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:18:33.439 MQT: stat/tasmota_DC3757/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:51","StartupUTC":"2022-10-10T19:10:42","Sleep":50,"CfgHolder":4617,"BootCount":64,"BCResetTime":"2022-09-30T15:41:58","SaveCount":234,"SaveAddress":"FA000"}}
20:18:33.446 MQT: stat/tasmota_DC3757/STATUS2 = {"StatusFWR":{"Version":"12.1.1.4(tasmota)","BuildDateTime":"2022-10-10T14:41:10","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"396/699"}}
20:18:33.451 MQT: stat/tasmota_DC3757/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HUAWEI","repeater"],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","2805C80001000680003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
20:18:33.463 MQT: stat/tasmota_DC3757/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
20:18:33.470 MQT: stat/tasmota_DC3757/STATUS5 = {"StatusNET":{"Hostname":"tasmota-DC3757-5975","IPAddress":"192.168.2.177","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.191","DNSServer2":"192.168.2.220","Mac":"E0:98:06:DC:37:57","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:18:33.476 MQT: stat/tasmota_DC3757/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.160","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DC3757","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:18:33.483 MQT: stat/tasmota_DC3757/STATUS7 = {"StatusTIM":{"UTC":"2022-10-10T19:18:33","Local":"2022-10-10T20:18:33","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:03","Sunset":"18:10"}}
20:18:33.486 MQT: stat/tasmota_DC3757/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:18:33.536 MQT: stat/tasmota_DC3757/STATUS10 = {"StatusSNS":{"Time":"2022-10-10T20:18:33","ANALOG":{"A0":1024},"ENERGY":{"TotalStartTime":"2022-09-30T15:41:58","Total":44.116,"Yesterday":19.043,"Today":25.042,"Power":259,"ApparentPower":259,"ReactivePower":0,"Factor":1.00,"Voltage":232,"Current":1.117}}}
20:18:33.545 MQT: stat/tasmota_DC3757/STATUS11 = {"StatusSTS":{"Time":"2022-10-10T20:18:33","Uptime":"0T00:07:51","UptimeSec":471,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"HUAWEI","BSSId":"F0:C4:2F:1B:9F:07","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

20:19:38.664 DMP: 55 5A 02 F1 E8 00 07 9D 00 3F 3E 00 35 F8 50 DB 38 00 B2 A2 F1 D6 93 3A
20:19:38.753 CFG: Saved to flash at F9, Count 235, Bytes 4096
20:19:38.796 DMP: 55 5A 02 F1 E8 00 07 9D 00 3F 3E 00 35 F8 50 DB 38 00 B2 A2 F1 D6 95 3C
20:19:38.801 DMP: 55 5A 02 F1 E8 00 07 9D 00 3F 3E 00 35 F8 50 DB 38 00 B2 A2 F1 D6 97 3E
20:19:38.840 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 A2 F1 D6 9A 3B
20:19:38.943 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 A2 F1 D6 9C 3D
20:19:39.047 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 A2 F1 D6 9E 3F
20:19:39.150 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 A0 80
20:19:39.254 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 A3 83
20:19:39.380 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 A5 85
20:19:39.461 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 A7 87
20:19:39.565 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 AA 8A
20:19:39.668 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 AC 8C
20:19:39.772 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 E1 F1 D6 AE 8E
20:19:39.877 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 FA 50 DB 38 00 B2 E1 F1 D6 B1 93
20:19:39.980 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 FA 50 DB 38 00 B2 E1 F1 D6 B3 95
20:19:40.084 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 FA 50 DB 38 00 B2 CC F1 D6 B5 82
20:19:40.188 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 FA 50 DB 38 00 B2 CC F1 D6 B7 84
20:19:40.291 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 FA 50 DB 38 00 B2 CC F1 D6 BA 87
[... 22 lines cut ...]
20:19:42.574 DMP: 55 5A 02 F1 E8 00 07 9E 00 3F 3E 00 35 F8 50 DB 38 00 B2 C2 F1 D6 EC B4
20:19:42.677 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 C2 F1 D6 EE AF
20:19:42.781 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 C2 F1 D6 F1 B2
20:19:42.885 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 C2 F1 D6 F3 B4
20:19:42.989 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 84 F1 D6 F5 78
20:19:43.126 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 84 F1 D6 F8 7B
20:19:43.196 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 84 F1 D6 FA 7D
20:19:43.300 DMP: 55 5A 02 F1 E8 00 07 97 00 3F 3E 00 35 F8 50 DB 38 00 B2 84 F1 D6 FC 7F
```

### TO REPRODUCE

Connect any device and turn on the switch :)

### EXPECTED BEHAVIOUR

Load of 250W should generate after one hour 0,25 kWh,but generate  2,5 kWh 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-10-11:
What happens if you remove power from the Nous, wait a few seconds and then power on again. How does the logging look like then?

Watch for byte 20 which is 0xF1 in your case signalling an overflow from the CF pulses. Do a test for ten minutes and see if you receive a usage of 250W / 6 = 0, 042 kWh.

domasofan, 2022-10-11:
Seems i have a similar issue with my Nous A1T single plug.

I sent the following also to the Tasmota mailing list:

I compiled Tasmota 12.1.1 some days ago using TasmoCompiler 9.1.0. 
I compiled it for my Nous A1T smart plugs. 
I used the generic board with power monitoring functions and added scripting and sendmail to my config. 

I have the feeling i have some issues with power monitoring. 
And mostly it seems to happen on a tuesday or at least from monday to tuesday. 
At least thats the feeling i have. 
I receive yesterdays energy data via email every day using this script: 

>D 25 
ey=0 
to="<to@example.com>" 

>T 
ey=ENERGY#yesterday 
if chg[ey]>0 
then 
=>sendmail [mail.example.com:465:from@example.com:password:<from@example.com>:%to%:Energy %frnm% %tstamp%]* 
endif 

>m 
Time: %tstamp% 
Device: %frnm% 
Energy yesterday (kWh): %ey% 
# 

That works pretty well most of the time. 
Though i seem to have regular issues with the energy measurements. 

Yesterday I got a reading of 0.190 KWh on my computer power monitor. 
Though it wasn't turned on the whole day and the plug had no power. 
Today reading seems to be normal. 
Currently showing 0.014 KWh after 14 mins of computer usage which is pretty accurate. 

Today i have on my living room monitor 0.229 kwH which also seems to be pretty wrong. 
There was just one smart speaker running which uses 3-5 W for the whole day. 
This morning i was listening a little bit radio just for around 45 minutes with around 15-17 W because the stereo was running.

I also want to say this data also shows on the web interface. Not just in the email. 
So that seems not to be a problem with the script itself. 

Though what i have monitored the uptime of the livingroom device is 0D18:06:23 which points that the device seems to have rebooted at 00:00. 
The computer plug has restarted at the exact same time. 

I have another device in the office. 
A Shelly Plug S running a Tasmota 12.1.1 from the official OTA server with nothing special enabled. 
This runs now i guess for over 30 days. 

Logs are not present at the moment unfortunately. 

Restart reasons this morning according to the Info Screen of both devices:
Computer: Software Watchdog 
Living room: Software/System restart 

Template in both devices used is the one from the template repo: 
{"NAME":"NOUS A1T","GPIO":[32,0,0,0,2720,2656,0,0,2624,320,224,0,0,1],"FLAG":0,"BASE":49} 

Greetings and thanks a lot, 
Simon

bdkacz, 2022-10-12:
> What happens if you remove power from the Nous, wait a few seconds and then power on again. How does the logging look like then?
> 
> Watch for byte 20 which is 0xF1 in your case signalling an overflow from the CF pulses. Do a test for ten minutes and see if you receive a usage of 250W / 6 = 0, 042 kWh.
> 
> Make sure to perform the power calibration. It looks like your `PowerSet` is a factor 10 too large.

Hello. I remove the power from nous,wait 30 seconds and plug it again. Aftern turn on switch  the same - daily usage increasing too fast. Logs   bellow

```
18:21:37.776 DMP: 55 5A 02 F1 E8 00 08 16 00 3F 3E 00 39 36 50 DB 38 00 C9 D0 71 0B 6C C9
18:21:37.820 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 D0 71 0B 6E CC
18:21:37.837 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 D0 71 0B 70 CE
18:21:37.969 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 D0 71 0B 72 D0
18:21:38.046 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 D0 71 0B 74 D2
18:21:38.149 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 D0 71 0B 76 D4
18:21:38.253 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 E3 71 0B 78 E9
18:21:38.357 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 E3 71 0B 7A EB
18:21:38.474 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 E3 71 0B 7C ED
18:21:38.565 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 E3 71 0B 7E EF
18:21:38.668 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 E3 71 0B 80 F1
18:21:38.772 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 36 50 DB 38 00 C9 E3 71 0B 82 F3
18:21:38.876 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 E3 71 0B 85 CD
18:21:38.980 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 E3 71 0B 87 CF
18:21:39.083 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 E3 71 0B 89 D1
18:21:39.222 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 E3 71 0B 8B D3
18:21:39.292 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 E3 71 0B 8D D5
18:21:39.395 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 8F 5F
18:21:39.499 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 91 61
18:21:39.603 DMP: 55 5A 02 F1 E8 00 08 17 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 93 63
18:21:39.721 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 95 60
18:21:39.811 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 97 62
18:21:39.915 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 99 64
18:21:40.019 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 9B 66
18:21:40.122 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 9D 68
18:21:40.226 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B 9F 6A
18:21:40.329 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 6B 71 0B A1 6C
18:21:40.762 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 B6 71 0B A3 B9
18:21:40.811 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 B6 71 0B A5 BB
18:21:40.814 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 B6 71 0B A7 BD
18:21:40.818 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 0D 50 DB 38 00 C9 B6 71 0B A9 BF
18:21:40.849 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 B6 71 0B AB ED
18:21:40.970 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 B6 71 0B AD EF
18:21:41.056 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 B6 71 0B AF F1
18:21:41.160 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 B6 71 0B B1 F3
18:21:41.264 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 B6 71 0B B3 F5
18:21:41.368 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 B6 71 0B B5 F7
18:21:41.472 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B B7 D9
18:21:41.575 DMP: 55 5A 02 F1 E8 00 08 12 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B B9 DB
18:21:41.680 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B BB DF
18:21:41.783 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B BD E1
18:21:41.887 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B BF E3
18:21:41.991 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B C1 E5
18:21:42.095 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B C3 E7
18:21:42.222 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B C5 E9
18:21:42.302 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B C8 EC
18:21:42.406 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B CA EE
18:21:42.509 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 96 71 0B CC F0
18:21:42.614 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 D9 71 0B CE 35
18:21:42.725 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 39 50 DB 38 00 C9 D9 71 0B D0 37
18:21:42.821 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 30 50 DB 38 00 C9 D9 71 0B D2 30
18:21:42.926 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 30 50 DB 38 00 C9 D9 71 0B D4 32
18:21:43.029 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 30 50 DB 38 00 C9 D9 71 0B D6 34
18:21:43.133 DMP: 55 5A 02 F1 E8 00 08 14 00 3F 3E 00 39 30 50 DB 38 00 C9 D9 71 0B D8 36
[... cut ...]
18:21:48.970 DMP: 55 5A 02 F1 E8 00 08 15 00 3F 3E 00 39 1E 50 DB 38 00 C9 B8 71 0C 49 76
18:21:49.052 DMP: 55 5A 02 F1 E8 00 08 15 00 3F 3E 00 39 1E 50 DB 38 00 C9 9A 71 0C 4B 5A
18:21:49.155 DMP: 55 5A 02 F1 E8 00 08 15 00 3F 3E 00 39 1E 50 DB 38 00 C9 9A 71 0C 4E 5D
18:21:49.259 DMP: 55 5A 02 F1 E8 00 08 15 00 3F 3E 00 39 1E 50 DB 38 00 C9 9A 71 0C 50 5F
18:21:49.363 DMP: 55 5A 02 F1 E8 00 08 15 00 3F 3E 00 39 1E 50 DB 38 00 C9 9A 71 0C 52 61
18:21:49.472 DMP: 55 5A 02 F1 E8 00 08 15 00 3F 3E 00 39 1E 50 DB 38 00 C9 9A 71 0C 54 63

```

bdkacz, 2022-10-12:
After  remove power from the Nous, wait  30 seconds and then power on again - the same BUT after  again  calibration seems working ok - 0,042 kWh after 10 minutes of working. But why it display ok - 2xxW but calculating as 2xxx W ?

closed 2022-11-11 by github-actions[bot]