arendst/Tasmota #16791: Random power today value after power loss

e9hack, 2022-10-11:
### PROBLEM DESCRIPTION
A power monitoring plug shows random values for energy today after a power loss. The device is a Avatar AWP16L (similar to the [Elehot](https://sylvainzimmer.com/blog/2021/09/smart-plugs-tasmota/)).
Before power loss:
  Energy Today: 0.02424kWh
  Energy Yesterday: 1.13630kWh
  Energy Total: 16.20227kWh

After power loss:
Energy Today: 0.60965kWh
Energy Yesterday: 1.13630kWh
Energy Total: 16.78769kWh

Energy Today looks like a random value. The value for Energy Total is increased, but not by the exact value of Energy Today. If I introduce a second power loss a few minutes later, the issue doesn't occur. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Avatar Mini Smart Wifi Plug
- [x] Tasmota binary firmware version number used: 12.1.1.4 (with a few modifications)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: tasmoadmin or firmware upgrade via web interface
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:03:16.405 CMD: Backlog Template; Module; GPIO 255
15:03:16.454 MQT: stat/PlugAvatar_2/RESULT = {"NAME":"Generic","GPIO":[0,1,320,1,0,2720,0,0,2624,32,2656,224,0,0],"FLAG":0,"BASE":77}
15:03:16.663 MQT: stat/PlugAvatar_2/RESULT = {"Module":{"77":"Avatar AWP16L"}}
15:03:16.917 MQT: stat/PlugAvatar_2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:01:39.218 CMD: Status 0
15:01:39.224 MQT: stat/PlugAvatar_2/STATUS = {"Status":{"Module":77,"DeviceName":"VDR","FriendlyName":["PlugAvatar 2"],"Topic":"PlugAvatar_2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:01:39.255 MQT: stat/PlugAvatar_2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://tasmoadmin.osexpert.lan:80/data/firmwares/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:21:07","StartupUTC":"2022-10-11T12:40:32","Sleep":50,"CfgHolder":4618,"BootCount":51,"BCResetTime":"2022-08-27T18:21:16","SaveCount":1891,"SaveAddress":"F4000"}}
15:01:39.290 MQT: stat/PlugAvatar_2/STATUS2 = {"StatusFWR":{"Version":"12.1.1.4(sensors)","BuildDateTime":"2022-10-09T21:29:20","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"426/699"}}
15:01:39.310 MQT: stat/PlugAvatar_2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["The Internet of Shitty Things","AndroidAP"],"TelePeriod":15,"Resolution":"568B80C0","SetOption":["00008029","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
15:01:39.347 MQT: stat/PlugAvatar_2/STATUS4 = {"StatusMEM":{"ProgramSize":676,"Free":324,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"xxxxxx","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F8A8787","0405A005","B7F7BFCF","05DA9FC4","24367CC7","00084052","20000000","54000020","00000001"],"Drivers":"1,2,3,4,5,6,8,9,10,12,14,16,17,20,24,29,34,56,62","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98"}}
15:01:39.387 MQT: stat/PlugAvatar_2/STATUS5 = {"StatusNET":{"Hostname":"PlugAvatar-2-xxxx","IPAddress":"192.168.255.40","Gateway":"192.168.255.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.255.1","DNSServer2":"0.0.0.0","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:01:39.415 MQT: stat/PlugAvatar_2/STATUS6 = {"StatusMQT":{"MqttHost":"mosquitto.osexpert.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_xxxxxx","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:01:39.438 MQT: stat/PlugAvatar_2/STATUS7 = {"StatusTIM":{"UTC":"2022-10-11T13:01:39","Local":"2022-10-11T15:01:39","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:35","Sunset":"18:40"}}
15:01:39.456 MQT: stat/PlugAvatar_2/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:01:39.475 MQT: stat/PlugAvatar_2/STATUS10 = {"StatusSNS":{"Time":"2022-10-11T15:01:39","ENERGY":{"TotalStartTime":"2022-09-23T18:32:51","Total":16.78823,"Yesterday":1.13630,"Today":0.61020,"Power":1.6,"ApparentPower":4.0,"ReactivePower":3.6,"Factor":0.40,"Voltage":233.7,"Current":0.017}}}
15:01:39.500 MQT: stat/PlugAvatar_2/STATUS11 = {"StatusSTS":{"Time":"2022-10-11T15:01:39","Uptime":"0T00:21:07","UptimeSec":1267,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"The Internet of Shitty Things","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Let run the device under power for ~1day and introduce a power loss by simply disconnect the plug from the wall socket.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-10-12:
Closing since answered

closed 2022-10-12 by Jason2866

e9hack, 2022-10-12:
> Nothing random about that, what happens is simply that consumption data since last save is, well, not saved. Settings data in flash is not saved continuously, as this would wear down the flash chip.

This answer looks wrong. If last saved data are restored, all values must change. Energy Today and Energy Yesterday can be any value but not the same value. Energy Total must be a lower than before power loss.

danutchereches, 2022-11-14:
This also happens to me and it's definitely a bug.
Why would Energy Total be higher after the power loss than before? That should not happen.
Maybe it's trying to guess the lost data and fails?
@Jason2866 please reopen.

danutchereches, 2022-11-14:
Tracking the energy consumption in the backend would be fine, but if the values are higher after the power loss, the consumption will be inflated.
That's how I noticed the problem. Home assistant tracks the power consumption and I had a big spike after I took down the power for a small maintenance job.

[image]

Jason2866, 2022-11-14:
Power loss is a not defined state. It can happen that garbage is written to flash.
The solution i see, is to check (in backend) if the values after a off time seems plausible.
Or you do not use the power today value at all. Monitoring the values getting via telelperiod in backend. Summing up the values yourself.

danutchereches, 2022-11-14:
I just noticed that both counters, in both cases, increased with 0.775.
Also the first value after the power was restored was correct.

I'm assuming that 0.775 is maybe a daily value saved from a previous day and it didn't update because I kept restarting the device.
But why is it being added to the total?

[image]
[image]

barbudor, 2022-11-14:
> I didn't even know tasmota had a shutdown command. How exactly? I couldn't find anything about shutdown in the documentation.
As per the documentation: [restart 2](https://tasmota.github.io/docs/Commands/#restart)

> When is the data being saved? At a fixed time or every 24 hours after boot?
Everyday at midnight

You can force more recurrent saving using a rule, but at the expense of wearing the flash memory faster. For example, every hour:
`Rule1 on time#minute|60 do savedata endon`

> If the total energy is stored without today's energy, it might explain what I'm seeing, at least partially.
So an option to disable today's energy save to flash would help.

All three EnergyToday, EnergyYesterday and EnergyTotal are saved in the flash
There is no option for not saving EnergyTotal but you are free to ignore it in your MQTT backend.

danutchereches, 2022-11-14:
Ok, thank you for your answers, but they still do not address the issue.

How is this not a bug?
All I did was to disconnect the power, just now.
I understand that it's not a guarantee, it can lose data when the power is lost.
But this is something else.

[image]
[image]

Jason2866, 2022-11-14:
Have you done `restart 2` before disconnecting power?
If not, as explained before, there is no defined state.

danutchereches, 2022-11-14:
No, I did not. I was trying to replicate a power outage.
That was the entire point. When I manually reboot it, it's fine.

What do you mean by "no defined state"?

It's not nothing, it's clearly something.
It looks like it's previous data counted twice.

You said that garbage might be written to flash, but I really doubt that's true.
What do you need from me in order to verify?
Assuming it's garbage or just dismissing me doesn't help anyone.

danutchereches, 2022-11-14:
```16:29:33.748 CMD: status 0
16:29:33.760 MQT: stat/dualr3_office_1024F8/STATUS = {"Status":{"Module":0,"DeviceName":"dualr3_office","FriendlyName":["Sonoff Dual R3 Office",""],"Topic":"dualr3_office_1024F8","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:29:33.817 MQT: stat/dualr3_office_1024F8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T02:48:33","StartupUTC":"2022-11-14T12:41:00","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2022-11-01T17:26:45","SaveCount":87}}
16:29:33.862 MQT: stat/dualr3_office_1024F8/STATUS2 = {"StatusFWR":{"Version":"12.2.0(bluetooth)","BuildDateTime":"2022-10-17T08:39:34","Core":"2_0_5","SDK":"v4.4.3.4","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"415/699"}}
16:29:33.884 MQT: stat/dualr3_office_1024F8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WNET-C",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:29:33.927 MQT: stat/dualr3_office_1024F8/STATUS4 = {"StatusMEM":{"ProgramSize":1584,"Free":1295,"Heap":97,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7C7","00148001","000000CF","010013C0","C0000981","00004080","00200000","5400082C","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,85","Sensors":"1,2,3,5,6,52,62,127"}}
16:29:33.987 MQT: stat/dualr3_office_1024F8/STATUS5 = {"StatusNET":{"Hostname":"dualr3-office-1024F8-1272","IPAddress":"10.10.105.247","Gateway":"10.10.105.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"0.0.0.0","Mac":"C4:DE:E2:10:24:F8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:29:34.019 MQT: stat/dualr3_office_1024F8/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.105.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1024F8","MqttUser":"allsensors","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:29:34.058 MQT: stat/dualr3_office_1024F8/STATUS7 = {"StatusTIM":{"UTC":"2022-11-14T15:29:34","Local":"2022-11-14T16:29:34","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:57","Sunset":"17:10"}}
16:29:34.091 MQT: stat/dualr3_office_1024F8/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:29:34.123 MQT: stat/dualr3_office_1024F8/STATUS10 = {"StatusSNS":{"Time":"2022-11-14T16:29:34","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-11-01T17:26:45","Total":25.363,"Yesterday":1.601,"Today":2.753,"Power":[83,0],"ApparentPower":[150,0],"ReactivePower":[125,0],"Factor":[0.55,0.00],"Voltage":238,"Current":[0.630,0.000],"BL09XX":{"Temperature":26.1}},"TempUnit":"C"}}
16:29:34.170 MQT: stat/dualr3_office_1024F8/STATUS11 = {"StatusSTS":{"Time":"2022-11-14T16:29:34","Uptime":"0T02:48:34","UptimeSec":10114,"Heap":96,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"WNET-C","BSSId":"C2:6E:1F:81:2A:4E","Channel":11,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:04"}}}```

Jason2866, 2022-11-14:
I would not use a Bluetooth build on a device where i want long term stability.

arendst, 2022-11-14:
How do you simulate power outage? How long do you keep the device powered off before restoring power?

My local tests on both ESP8266 and ESP32 do not show your (reproducable) errors (but then I never use bluetooth either).

danutchereches, 2022-11-14:
Just a few seconds.
When I first noticed, it wasn't a test and it was down for ~30 minutes.

It happened with 2 different devices. (but both Sonoff Dual R3 v2 with bluetooth)
I will do some tests without bluetooth.

arendst, 2022-11-14:
You might want to try latest dev. release. Pre-compiled are ready after one hour.

With that enable ``weblog 3`` and watch the console after power on. Expect to see someting like this:
```
00:00:00.292-209/49 NRG: Rtc valid 1, kWhtoday_ph Rtc 1767/2/0, Set 1767/2/0
```
shortly after power on. It shows if the RTC data is valid (re-initialized or survived power outage) and the three phase kwhtoday values (multiplied by 100). This can give a clue about the situation you observe.

danutchereches, 2022-11-15:
```
00:00:00.230 NRG: Rtc valid 1, kWhtoday_ph Rtc 286805/932/0, Set 286805/932/0
```
I flashed today a dev build of tasmota32, but I couldn't reproduce the bug.
The counters just reverted to a smaller value, which is expected.

Before power off:
[image]
After:
[image]

I will re-test (maybe the saved data needs to be older) but I hope that the firmware change has fixed the issue I had.

danutchereches, 2022-11-16:
It happened again.
```
00:00:00.282 NRG: Rtc valid 1, kWhtoday_ph Rtc 286805/932/0, Set 286805/932/0
```

Before:
[image]
After:
[image]

Does this sound right?

According to the log output, the saved data is valid, but it's the same as yesterday. (2868.05Wh)
So why did the total increase?
My assumption is that the total was saved (at 23:00 UTC) but the daily was not.

This is what I have stored in an external db:
```
2022-11-15T22:58:26.978825Z sensor dualr3_office_energy_total 29.178
2022-11-15T22:59:26.984816Z sensor dualr3_office_energy_total 29.179
2022-11-15T23:00:26.98304Z  sensor dualr3_office_energy_total 29.181
2022-11-15T23:01:26.984839Z sensor dualr3_office_energy_total 29.182
```
```
2022-11-15T22:58:26.979187Z sensor dualr3_office_energy_today 3.192
2022-11-15T22:59:26.985166Z sensor dualr3_office_energy_today 3.194
2022-11-15T23:00:26.983397Z sensor dualr3_office_energy_today 0.001
2022-11-15T23:01:26.985168Z sensor dualr3_office_energy_today 0.002
```

This is the first logged values after booting:
```
16:32:03.216 MQT: stat/dualr3_office_1024F8/STATUS10 = {"StatusSNS":{"Time":"2022-11-16T16:32:03","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-11-01T17:26:45","Total":32.057,"Yesterday":3.194,"Today":2.878,"Power":[255,53],"ApparentPower":[377,75],"ReactivePower":[277,54],"Factor":[0.68,0.70],"Voltage":237,"Current":[1.591,0.319],"BL09XX":{"Temperature":24.8}},"TempUnit":"C"}}
```

32057 (total) - 2878 (today) = 29179

Maybe the daily should be set/saved to 0 when the day changes.
So you lose the current day when reboot without saving, which is not perfect, but it doesn't mess up apps that monitor the change.
```
For sensors with state_class STATE_CLASS_TOTAL_INCREASING, a decreasing value is interpreted as the start of a new meter cycle or the replacement of the meter. 
```

arendst, 2022-11-17:
I suppose your external db time is off by one hour from your tasmota device. At 23:00 (db time) you see today value being reset to 0 (and may expect being saved as 3194 as yesterday. Total value is saved as 29179). Total is only saved once at midnight and incremented with the current today value in all displays from midnight.

The code responsible for saving at midnight is in file ``xdrv_03_energy.ino`` starting at line 355:
```
      if (LocalTime() == Midnight()) {
        for (uint32_t i = 0; i < 3; i++) {
          Settings->energy_kWhyesterday_ph[i] = RtcSettings.energy_kWhtoday_ph[i];

          RtcSettings.energy_kWhtotal_ph[i] += RtcSettings.energy_kWhtoday_ph[i];
          Settings->energy_kWhtotal_ph[i] = RtcSettings.energy_kWhtotal_ph[i];

          Settings->energy_kWhexport_ph[i] = RtcSettings.energy_kWhexport_ph[i];

          Energy.period[i] -= RtcSettings.energy_kWhtoday_ph[i];     // this becomes a large unsigned, effectively a negative for EnergyShow calculation
          Energy.kWhtoday[i] = 0;
          Energy.kWhtoday_offset[i] = 0;
          RtcSettings.energy_kWhtoday_ph[i] = 0;
          Energy.start_energy[i] = 0;
//        Energy.kWhtoday_delta = 0;                                 // dont zero this, we need to carry the remainder over to tomorrow
          Energy.daily_sum_import_balanced = 0.0;
          Energy.daily_sum_export_balanced = 0.0;
        }
        EnergyUpdateToday();
#if defined(USE_ENERGY_MARGIN_DETECTION) && defined(USE_ENERGY_POWER_LIMIT)
        Energy.max_energy_state  = 3;
#endif  // USE_ENERGY_POWER_LIMIT
      }
```

It may go wrong the next day when power is lost and ``Settings->energy_kWhtoday_ph[]`` is never updated.

I'll add it to the midnight run to reset it to 0 and only set it on controlled restart (``restart 1``) or powerdown (``restart 2``).

arendst, 2022-11-17:
Try latest dev release.

danutchereches, 2022-11-19:
Yes, it seems fine now.
Energy Total reverted to what it was at midnight, and Energy Today was 0.
Thank you!