arendst/Tasmota #16831: Power calibration does not work

Elmi77, 2022-10-15:
### PROBLEM DESCRIPTION
When calibrating the power of a Delock 11827 according to the Tasmota-manual, the resulting measured power is wrong

### REQUESTED INFORMATION

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Delock 11827
- [X ] Tasmota binary firmware version number used: 12.1.1.6
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: OTA-update
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:33:39.867 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:33:40.095 RSL: RESULT = {"Module":{"0":"Generic"}}
16:33:40.352 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2688":"HLW8012 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:34:30.155 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ShineBaby","FriendlyName":["ShineBaby"],"Topic":"tasmota_D74227","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:34:30.159 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T01:04:08","StartupUTC":"2022-10-15T14:30:22","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-10-14T11:27:10","SaveCount":82,"SaveAddress":"F4000"}}
16:34:30.164 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.1.6(tasmota)","BuildDateTime":"2022-10-13T15:59:58","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"345/699"}}
16:34:30.168 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Meham_Guest",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A002800000000","00000081","00006000","00000000","00000000"]}}
16:34:30.178 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
16:34:30.182 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D74227-0551","IPAddress":"192.168.3.31","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.3.1","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:D7:42:27","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
16:34:30.188 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-10-15T15:34:30","Local":"2022-10-15T16:34:30","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:10","Sunset":"18:00"}}
16:34:30.191 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:34:30.198 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-10-15T16:34:30","ENERGY":{"TotalStartTime":"2022-10-14T11:27:10","Total":0.151,"Yesterday":0.000,"Today":0.151,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":232,"Current":0.000}}}
16:34:30.203 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-10-15T16:34:30","Uptime":"0T01:04:08","UptimeSec":3848,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Meham_Guest","BSSId":"70:8B:CD:D4:13:3A","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- connect a simple 400 W heater to the device together with a measurement tool for voltage and power
- set command VoltageSet 233 via console
- set command PowerSet 377 via console

### EXPECTED BEHAVIOUR
The shown, active power now should be close to 377 W.

Instead of this, Tasmota displays about 325 W (Power factor 1.0). Next I tried to set a value of 450 in order to check if the result then comes closer to the real value of 377 W - but this trashed the output completely, displayed power and current have been 0 then. This could be fixed only by restoring the configuration completely.

arendst, 2022-10-15:
Final step: currentset

Elmi77, 2022-10-16:
Does not change anything. No matter what I set, it does not influence the result.

Elmi77, 2022-10-16:
Btw: why can't Tasmota do this calculation do automatically after power and voltage have been set?

sfromis, 2022-10-16:
Because it is ideally not meant to be a calculation, that is just a bit of a shortcut if you are not able to do a "real calibration", measuring the actual current. When I calibrate, I use a "known good meter" able to provide all 3 numbers separately.