arendst/Tasmota #16879: Ikea Fyrtur Blind CurrentPositionLiftPercentage and BatteryPercentage

orichienal, 2022-10-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hi everyone,
I have a total of 4 of the Fyrtur blinds.
Two of them regularly report their position and battery when opening or closing.
But I can't do that with the other two.
I have re-registered them several times.
I connected them to the Ikea bridge and again to the Tasmota zigbee bridge (Sonoff).
But all without success.

It would be great if someone could give me a hint.

Best wishes
orichienal

Apart from the fact that the buttons supplied only last for a few days and then the battery is empty.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer, OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
`14:33:53.091 CMD: Backlog Template; Module; GPIO 255
14:33:53.092 SRC: WebConsole from 192.168.XXX.XXX
14:33:53.094 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
14:33:53.168 SRC: Backlog
14:33:53.170 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
14:33:53.174 MQT: stat/tasmota_XXXXXX/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:33:53.376 SRC: Backlog
14:33:53.378 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
14:33:53.381 MQT: stat/tasmota_XXXXXX/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
14:33:53.626 SRC: Backlog
14:33:53.627 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
14:33:53.634 MQT: stat/tasmota_XXXXXX/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
`14:34:35.655 CMD: Backlog Rule1; Rule2; Rule3
14:34:35.656 SRC: WebConsole from 192.168.XXX.XXX
14:34:35.658 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 19, Pld -99, Data 'Rule1; Rule2; Rule3'
14:34:35.715 SRC: Backlog
14:34:35.717 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
14:34:35.721 MQT: stat/tasmota_XXXXXX/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:34:35.924 SRC: Backlog
14:34:35.926 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
14:34:35.930 MQT: stat/tasmota_XXXXXX/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:34:36.174 SRC: Backlog
14:34:36.175 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
14:34:36.180 MQT: stat/tasmota_XXXXXX/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}`

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
`14:35:34.783 CMD: Status 0
14:35:34.784 SRC: WebConsole from 192.168.XXX.XXX
[... 23 lines cut ...]

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

This is from the working Blinds
14:40:56.611 MQT: tele/tasmota_AA4B28/SENSOR = {"ZbReceived":{"0xGGGG":{"Device":"0xA152","Name":"blind02","CurrentPositionLiftPercentage":35,"BatteryPercentage":33,"Endpoint":1,"LinkQuality":118}}}
14:40:50.781 MQT: tele/tasmota_AA4B28/SENSOR = {"ZbReceived":{"0xYYYY":{"Device":"0x1245","Name":"blind03","CurrentPositionLiftPercentage":6,"BatteryPercentage":29,"Endpoint":1,"LinkQuality":102}}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

working 
14:48:16.728 MQT: stat/tasmota_XXXXXX/RESULT = {"ZbStatus3":[{"Device":"0xGGGG","Name":"blind02","IEEEAddr":"0xGGGG","ModelId":"FYRTUR block-out roller blind","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":true,"BatteryPercentage":33,"BatteryLastSeenEpoch":1666444902,"LastSeen":437,"LastSeenEpoch":1666446059,"LinkQuality":118}]}

14:48:38.740 MQT: stat/tasmota_XXXXXX/RESULT = {"ZbStatus3":[{"Device":"0xYYYY","Name":"blind03","IEEEAddr":"0xYYYY","ModelId":"FYRTUR block-out roller blind","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":true,"BatteryPercentage":29,"BatteryLastSeenEpoch":1666418408,"LastSeen":459,"LastSeenEpoch":1666446059,"LinkQuality":102}]}

dosn't work
14:47:45.213 MQT: stat/tasmota_XXXXXX/RESULT = {"ZbStatus3":[{"Device":"0xDDDD","Name":"blind01","IEEEAddr":"0xDDDD","ModelId":"FYRTUR block-out roller blind","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":false,"BatteryPercentage":49,"BatteryLastSeenEpoch":1665775258,"LastSeen":5924,"LastSeenEpoch":1666440541,"LinkQuality":110}]}

14:48:53.519 MQT: stat/tasmota_XXXXXX/RESULT = {"ZbStatus3":[{"Device":"0xKKKK","Name":"blind04","IEEEAddr":"0xKKKK","ModelId":"FYRTUR block-out roller blind","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":false,"BatteryPercentage":50,"BatteryLastSeenEpoch":1665775227,"LastSeen":20646,"LastSeenEpoch":1666425887,"LinkQuality":100}]}

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-10-22:
Please try thé latest Tasmota version. It now includes auto probing of battery

orichienal, 2022-10-22:
Ok I'll try (but not until tonight).
Is it enough to update the bridge or do I have to re-register the blinds again?

and maybe the batteries in the remotes will last longer?

s-hadinger, 2022-10-22:
You don't need to re-pair devices, although it will not hurt to do so.

orichienal, 2022-10-22:
Yeah, now it works, all my blinds sends CurrentPositionLiftPercentage and BatteryPercentage 

but my Open/Close Remotes no longer send BatteryPercentage, but this is another issue, i think
because i want to track the drain.

orichienal, 2022-10-22:
AH i need again ZbSend {"Device":"Ikea-Device","Read":{"BatteryPercentage":true}} to activate it for the Remotes

closed 2022-10-22 by orichienal