arendst/Tasmota #16880: OpenTherm ot_save_setpoints doesn't save status

Upgreydd, 2022-10-22:
### PROBLEM DESCRIPTION
I've installed Wemos D1 mini with diyless OpenTherm master board and have an issue with some functions in Tasmota.
1. I can't save settings - every time I restart device I need run below commands manually:
```
ot_flags CHOD,DHW,CH
ot_ch 1
ot_tboiler 23
ot_twater 44
ot_save_setpoints
```
After restart status is:
```
ot_flags = CHOD,DHW
ot_ch = 0
ot_tboiler = 85
ot_twater = 36
```

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini (probably clone)
- [x] Tasmota binary firmware version number used: 12.2.0 self compiled with only OpenTherm enabled
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: platformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:27:41.886 MQT: stat/OpenThermGW/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:27:42.107 MQT: stat/OpenThermGW/RESULT = {"Module":{"0":"Generic"}}
15:27:42.363 MQT: stat/OpenThermGW/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4928":"OpenTherm RX"},"GPIO5":{"4960":"OpenTherm TX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
15:28:40.225 MQT: stat/OpenThermGW/STATUS = {"Status":{"Module":0,"DeviceName":"OpenThermGW","FriendlyName":["OpenThermGW"],"Topic":"OpenThermGW","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:28:40.260 MQT: stat/OpenThermGW/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:15:29","StartupUTC":"2022-10-22T14:13:11","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2022-10-22T11:24:25","SaveCount":55,"SaveAddress":"F5000"}}
15:28:40.294 MQT: stat/OpenThermGW/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-22T12:53:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"426/699"}}
15:28:40.313 MQT: stat/OpenThermGW/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WeeFee",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
15:28:40.348 MQT: stat/OpenThermGW/STATUS4 = {"StatusMEM":{"ProgramSize":642,"Free":360,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004204","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6,69"}}
15:28:40.385 MQT: stat/OpenThermGW/STATUS5 = {"StatusNET":{"Hostname":"OpenThermGW-3084","IPAddress":"172.16.3.102","Gateway":"172.16.0.1","Subnetmask":"255.255.252.0","DNSServer1":"172.16.0.2","DNSServer2":"172.16.0.3","Mac":"C8:C9:A3:5D:EC:0C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:28:40.410 MQT: stat/OpenThermGW/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.0.41","MqttPort":1883,"MqttClientMask":"OpenThermGW","MqttClient":"OpenThermGW","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:28:40.436 MQT: stat/OpenThermGW/STATUS7 = {"StatusTIM":{"UTC":"2022-10-22T14:28:40","Local":"2022-10-22T15:28:40","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:21","Sunset":"17:47"}}
15:28:40.459 MQT: stat/OpenThermGW/STATUS10 = {"StatusSNS":{"Time":"2022-10-22T15:28:40","OPENTHERM":{"conn":"BUSY","settings":7,"SLAVE":{"FAULT":0,"CH":0,"DHW":1,"FL":1,"COOL":0,"CH2":0,"DIAG":0,"RAW":3221226252},"BTMP":{"FAULT":0,"REQ":23.0,"ACT": 23.0},"HWTMP":{"REQ":44.0,"ACT": 44.0},"ASFF":{"FC":0,"OFC":0},"FLM":43.0,"TB":59.0,"TDHW":43.0,"TRET":0.0,"DHWS":44.0,"TMAX":45.0,"OT10":{"lo":0,"hi":0},"OT12":{"lo":0,"hi":0},"OT15":{"lo":25,"hi":0},"OT18":8.9,"OT19":0.0,"OT33":127,"OT116":46207,"OT120":3057,"OT121":6183}}}
15:28:40.503 MQT: stat/OpenThermGW/STATUS11 = {"StatusSTS":{"Time":"2022-10-22T15:28:40","Uptime":"0T00:15:29","UptimeSec":929,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WeeFee","BSSId":"68:D7:9A:1A:A7:B9","Channel":11,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Install latest compiled tasmota with OpenTherm enabled
2. Configure RX,TX for Opentherm

### EXPECTED BEHAVIOUR
1. Settings persists between reboots

barbudor, 2022-10-22:
You have not provided the logs in `WebLog 3` mode while applying your commands, especially ot_savepoints
Please edit and update your issue
Also provide the output of command `Savedata`

NO SCREENSHOTS PLEASE - Use text copy/paste and include your logs into triple backticks
[image]

Upgreydd, 2022-10-22:
thanks @barbudor . I've added WebLog 3 with all commands.
Problematic part is it's not saving settings and not detecting CW state

barbudor, 2022-10-22:
Hum,
Right after ot_savepoints there is a "save to flash"
I will check the code

barbudor, 2022-10-22:
The values are reset to defaults because they are out of bounds
Default bounds are:
>  40 <= tboiler <= 85
>  23 <= twater <= 55

If either is out of bounds, both values are reset to defaults:
>   tboiler = 85
>   twater = 36

If you need different values, you must define them in your user_config_overide.h and recompile:

```
#define OT_HOT_WATER_MIN 23
#define OT_HOT_WATER_MAX 55
#define OT_BOILER_MIN 40
#define OT_BOILER_MAX 85

#define OT_HOT_WATER_DEFAULT 36;
#define OT_BOILER_DEFAULT 85;
```

Upgreydd, 2022-10-22:
@barbudor I think they should be changed in the main code too in case you use floor heating. For the floor, heating temperatures are mainly set between 20*C to 28*C based on my experience to have 21*C at home.

barbudor, 2022-10-22:
> in case you use floor heating

That's your use case, one among others
In order to adapt to your use case, you have to override those values in your `user_config_override.h` file while compiling.
There is no reason to replace one use-case by another one in the main code. Both are valid and the main code can support your use case if you follow the instructions.

Upgreydd, 2022-10-22:
@barbudor Thanks for help. Have a nice day

closed 2022-10-22 by Upgreydd

poseidogithub, 2022-12-17:
> That's your use case, one among others In order to adapt to your use case, you have to override those values in your `user_config_override.h` file while compiling. There is no reason to replace one [...]

just for more info: water heated floors are very common, and as @Upgreydd says, setpoint temperatures below 30 degrees are very very common.

bkbartk, 2022-12-28:
I see this one is closed, but I have a similar issue and just changing the default values didn't help.

I added this in the user_config_override:
**``**
#ifndef USE_OPENTHERM
#define USE_OPENTHERM
#endif

#ifdef USE_DOMOTICZ
#undef USE_DOMOTICZ                              
#endif 

#define OT_HOT_WATER_MIN 10
#define OT_HOT_WATER_MAX 80
#define OT_BOILER_MIN 10
#define OT_BOILER_MAX 85
```
where min and max are beyond the boundaries, but on my boiler there are only hardware buttons to adjust this anyway.

and then the strange thing is.

```
22:03:03.997 CMD: weblog 3
22:03:04.006 MQT: stat/tasmota_Thermostat/RESULT = {"WebLog":3}
22:03:04.642 CFG: Saved, Count 746, Bytes 4096
22:03:10.949 HTP: Console
22:03:22.913 SRC: MQTT
22:03:22.916 CMD: Grp 0, Cmd 'OT_CH', Idx 1, Len 1, Pld 1, Data '1'
22:03:22.927 MQT: stat/tasmota_Thermostat/RESULT = {"ot_ch":1}
22:03:23.667 [OTH]: Central Heating transitioning from off to on
22:03:23.869 [OTH]: Setting Boiler Temp. Old: 50, New: 50, Force: Y

Restart

00:00:00.003 HDW: ESP32-D0WDQ6-V3 
00:00:00.060 UFS: FlashFS mounted with 304 kB free
00:00:00.116 CFG: Loaded from File, Count 746
00:00:00.126 QPC: Count 1
00:00:00.127 CFG: CR 431/699, Busy 0
00:00:00.134 ROT: Mode 1
00:00:00.268 CFG: No '*.autoconf' file found
00:00:00.272 BRY: Berry initialized, RAM used=4041 bytes
00:00:00.287 BRY: No 'preinit.be'
00:00:00.290 SRC: Restart
00:00:00.295 Project tasmota - Thermostat Version 12.3.1(tasmota)-2_0_5_3(2022-12-28T21:45:04)
00:00:00.295 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.312 DSB: Sensors found 1
00:00:01.288 BRY: No 'autoexec.be'
00:00:01.305 [OTH]: perform handshake
00:00:01.419 [OTH]: getLastResponseStatus SUCCESS. Slave Cfg: 40030083
00:00:01.559 WIF: Attempting connection...
00:00:02.013 WIF: Connecting to AP1 <SSID> Channel 10 BSSId <BSSID> in mode 11n as tasmota-Thermostat-1496...
00:00:02.447 [OTH]: Setting Boiler Temp. Old: 0, New: 50, Force: N
00:00:02.699 [OTH]: Setting Hot Water Temp. Old: 0, New: 36
00:00:03.048 WIF: Attempting connection...
00:00:04.061 WIF: Attempting connection...
00:00:05.073 WIF: Attempting connection...
00:00:06.048 WIF: IPv4 192.168.180.182, mask 255.255.255.0, gateway 192.168.180.1
00:00:06.051 WIF: Connected
00:00:06.347 HTP: Web server active on tasmota-Thermostat-1496 with IP address 192.168.180.182
00:00:06.946 WIF: Resolving 'homeassistant' (192.168.180.13)
00:00:06.961 RTC: UTC 2022-12-28T21:04:04, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
22:04:04.000 RTC: Synced by NTP
22:04:04.360 MQT: Attempting connection...
22:04:04.388 WIF: Resolving 'homeassistant' (192.168.180.13)
22:04:04.418 MQT: Connected
22:04:04.422 MQT: tele/tasmota_Thermostat/LWT = Online (retained)
22:04:04.424 MQT: cmnd/tasmota_Thermostat/POWER = 
22:04:04.425 MQT: Subscribe to cmnd/tasmota_Thermostat/#
22:04:04.427 MQT: Subscribe to cmnd/tasmotas/#
[... 22 lines cut ...]
22:04:05.769 SRC: MQTT
22:04:05.771 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 10, Data '10'
22:04:05.797 MQT: stat/tasmota_Thermostat/STATUS10 = {"StatusSNS":{"Time":"2022-12-28T22:04:05","DS18B20":{"Id":"072211556BBA","Temperature":22.9},"OPENTHERM":{"conn":"READY","settings":1,"SLAVE":{"FAULT":0,"CH":1,"DHW":0,"FL":1,"COOL":0,"CH2":0,"DIAG":0,"RAW":3221225482},"BTMP":{"FAULT":0,"REQ":50.0,"ACT": 50.0},"OEMD":0,"FLM":0.0,"TB":36.0,"TDHW":55.0,"TRET":29.0,"TMAX":56.0,"OT14":0.0,"OT18":25.5,"OT116":0,"OT120":0},"TempUnit":"C"}}
22:04:06.888 APP: Boot Count 86
22:04:07.975 CFG: Saved, Count 747, Bytes 4096
22:04:08.919 MQT: tele/tasmota_Thermostat/STATE = {"Time":"2022-12-28T22:04:08","Uptime":"0T00:00:11","UptimeSec":11,"Heap":148,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":58,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"<ssid>","BSSId":"<mac>","Channel":10,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:04:08.983 MQT: tele/tasmota_Thermostat/SENSOR = {"Time":"2022-12-28T22:04:08","DS18B20":{"Id":"072211556BBA","Temperature":22.9},"OPENTHERM":{"conn":"BUSY","settings":1,"SLAVE":{"FAULT":0,"CH":1,"DHW":0,"FL":1,"COOL":0,"CH2":0,"DIAG":0,"RAW":3221225482},"BTMP":{"FAULT":0,"REQ":50.0,"ACT": 50.0},"OEMD":0,"FLM":0.0,"TB":36.0,"TDHW":55.0,"TRET":29.0,"TMAX":56.0,"OT14":0.0,"OT18":25.5,"OT116":0,"OT120":0},"TempUnit":"C"}
22:04:10.501 [OTH]: command ASFF is not supported by the boiler. Last status: INVALID
22:04:11.513 [OTH]: command TOUT is not supported by the boiler. Last status: INVALID
22:04:11.975 [OTH]: command DHWS is not supported by the boiler. Last status: INVALID
22:04:12.479 [OTH]: command OT10 is not supported by the boiler. Last status: INVALID
22:04:12.679 [OTH]: command OT12 is not supported by the boiler. Last status: INVALID
22:04:13.083 [OTH]: command OT15 is not supported by the boiler. Last status: INVALID
22:04:13.486 [OTH]: command OT19 is not supported by the boiler. Last status: INVALID
22:04:13.686 [OTH]: command OT33 is not supported by the boiler. Last status: INVALID
22:04:14.089 [OTH]: command OT117 is not supported by the boiler. Last status: INVALID
22:04:14.491 [OTH]: command OT121 is not supported by the boiler. Last status: INVALID
22:05:02.450 CMD: ot_ch
22:05:02.453 SRC: WebConsole from 192.168.180.148
22:05:02.455 CMD: Grp 0, Cmd 'OT_CH', Idx 1, Len 0, Pld -99, Data ''
22:05:02.466 MQT: stat/tasmota_Thermostat/RESULT = {"ot_ch":0}
```

while booting, on line 56 it now tells met ot_ch =1, where first it didn't.
but when I try to retrieve the actual status, it suddenly returns 0