arendst/Tasmota #16881: ESP32 temperature is not displayed with Relase 12.2.0

GieGit, 2022-10-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff POWR320D
- [X] Tasmota binary firmware version number used: 12.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
The temperature of the ESP32 is no longer displayed in the WebUi after the update to 12.2.0.

### EXPECTED BEHAVIOUR
_"ESP32": {
    "Temperature": 28.3
  },
  "TempUnit": "C" ""._

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
I was able to fix the problem with a downgrade to version 12.1.1.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-10-23:
You did not do much searching. That "ESP32 Temperature" is mostly a crap number, without relation to reality, causing user confusion. Therefore it was by default disabled, as a very useful improvement. You can enable it with `SetOption146`.

GieGit, 2022-10-23:
> SetOption146

Thanks it works

closed 2022-10-23 by GieGit