arendst/Tasmota #16882: I2c sensors work on some devices with extension cable and not on others

Rush56, 2022-10-23:
Hi everyone.

I'm pretty new to all this.

I have set up some sensors on modified Sonoff basic R2 and Sonoff 4CH R3 PRO. I have Tasmota flashes on them. 

Now i'm making a housing for my controller and connected those sensors via 4-pol aux connectors and extension aux cables. Total lenght from sensor to device is around 3.5 meters. 

On one Sonoff basic R2 and 4CH R3 PRO it works, but on another basic R2 it does not. It does however work without extension. 

So i used original 2x 3m extension AUX cables on devices that work and made another cable, as the resistence was pretty high over 3m AUX cable (1.1 Ohm). As i only used connectors and swapped for 0.25mm cable wires i have gotten resistance down to 0.3-0.4 OHm. 

That did not help. 

I have another two Sonoffs basics R2 on reserve. Havent been in use, only flashed and setup. 

I tryed on those two and it doesn't work either. I reduced cable to 2,5 meters. Still nothing. 

I reduced again to around 2 meters. Still nothing. 

Each time before reducing the cable i also trying same cable on devices that did work with original extension cable lenght and it worked. Just to make sure i didn't mess up something when wiring.

Are some devices just too weak or is there any setting i could change? Quick google threw out to reduce clock speed/frequency on i2c. Is that possible on tasmota? 

What else can i do, without using some kind of i2c extenders?

If it weren't working on all devices it would be more logical..

closed 2022-10-23 by close-issue-app[bot]

joba-1, 2022-10-23:
what you experience is pretty normal.

To understand what’s going on:
https://www.i2c-bus.org/

To see what’s going on: 
any osci will do, e.g.
Quimat 2.4 Inches DSO Shell TFT Digital Oscilloscope Kit with BNC Clip Cable Soldered Probe Open Source SMD and Housing (Ready-Assembled) https://amzn.eu/d/icVXKyN
I use one like this, works, but you can easily spend more money (for more value).

Try and error might work as well and getting down resistance was a good start.
Next could be shield lines, modify termination resistors, …

barbudor, 2022-10-23:
4k7 pull ups if not already there may help
ESP internal pull ups are too weak

sfromis, 2022-10-23:
3.5 or 2 meter is excessively long for I2C, not surprising with reliability issues. The bus is designed for a board bus within one device, something like up to 20-30 cm.

Rush56, 2022-10-24:
Thank you guys.

So before seeing your replays, i have already tryed putting up 4.7k resistor. Now it works up to around 2 meters.

Now i see two options:
-shielded wires or
-smaller resistor, maybe, 2.2k or something?

barbudor, 2022-10-24:
Shielded cable may help if the problem is from electric parasites
But it may be only line length
Smaller pull up may help, as well as very small resistor in series but experiments with oscilloscope may be needed

Rush56, 2022-10-25:
Thanks guys.

Probably i'll be okay with these 2 meters for one devices. Yesterday i was testing it, then i soldered aux connectors back to cable which increased lenght for 30-40cm.

That was enough for the signal to get lost again.

Then it came to my mind to power up i2c display i have for these sensors, as someone somewhere mentioned that with more devices you need smaller pull-up.

As soon as i connected display the sensor started working again. I have them in parallel. Why is that?

sfromis, 2022-10-25:
The display may have pullup resistors too, leading to a total stronger pullup (lower resistance). By badly exceeding I2C specs, you are walking around in a mine field of unreliability.

Rush56, 2022-10-26:
Well, I've been testing these sensors for 2 days now, 2 with 3 meter cables and one with a bit over 2 meter. They work normally, checking temp and humidity history in HA i can see they are working constantly, not loosing data/connection.

The thing is I did not know about lenght problem of i2c as i have already made a controller for mushroom tent and it all worked with shorter cables. Then i decided to make a box and make everything nice, so it was one of last parts of controller box building, everything else have i already set up.

Now i could go with 3 wire sensors, but none i have tryed can compete with this sensors i use now. They are so fast, accurate. These are SHT30 in housing.

Even other i2c sensors i tryed can't compete with this one. So i really dont wanna change that :)

sfromis, 2022-10-26:
Simplest solution may be more ESP chips, and shorter I2C cables, not at all expensive. Your setup is basically why there are I2C extenders on the market, but you did not want such. You could also use good deal of time on fiddling with pullup resistors and voltages to see if you can hit a sweet spot with "tolerable reliability". Looking deeper with an oscilloscope may be helpful to understand how digital signals are badly malformed by analog effects, like when wires get long, and see how mitigation strategies change things.

Rush56, 2022-10-26:
Well i looked into i2c extenders. But didn't research enough.

Will look into it, this would be useful if these can be close to ESP or directly at i2c ports, so i could hide them in the box. Need to research it.

However i think that the current setup will work fine as it seems. The way i have it set now it works flowlessly.

Thanks for all help!

Rush56, 2022-10-26:
€ 1,36  21%OFF | diymore CJMCU-9515 I2C Module PCA9515A 2 Channel 2Bit I2C Repeater SMBus 400KHz Dual Bidirectional Repeater Module for Arduino
https://a.aliexpress.com/_mqN12qc

Is this of any use? Not sure what to look into exactly.