arendst/Tasmota #16940: Athom Power Switches Randomly Restarting After 12.2.0 Upgrade

trichromic-ab, 2022-10-28:
### PROBLEM DESCRIPTION
Athom power switch model PG04V2 with Tasmota 12.2.0 restarts with reason "Software Watchdog".
The restarts appear to be random and can be minutes or hours between.
The relay is heard to click andf there can be a very quick power outage.
I have reported to Athom Support via their forum and they suggested a downgrade to v10.1 (which is what they were supplied with).
So far the device downgraded has held without restarting for 6 hours.

One other possible relevent information is that I monitor the devices with TasmoAdmin, in case that querying Tasmota is what is causing the watchdog timeout.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Athom Plug V2
- [x] Tasmota binary firmware version number used: 12.2.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
20:43:02.192 CMD: Backlog Template; Module; GPIO 255
20:43:02.469 MQT: stat/tasmota-sw04/RESULT = {"Module":{"0":"Athom Plug V2"}}
20:43:02.676 MQT: stat/tasmota-sw04/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
20:54:02.821 MQT: stat/tasmota-sw04/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:54:03.023 MQT: stat/tasmota-sw04/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
20:54:03.224 MQT: stat/tasmota-sw04/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:54:46.001 CMD: status 0
20:54:46.008 MQT: stat/tasmota-sw04/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota-sw04","FriendlyName":["tasmota-sw04"],"Topic":"tasmota-sw04","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:54:46.014 MQT: stat/tasmota-sw04/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T01:07:53","StartupUTC":"2022-10-28T18:46:53","Sleep":50,"CfgHolder":4617,"BootCount":49,"BCResetTime":"2022-10-22T19:56:50","SaveCount":113,"SaveAddress":"F8000"}}
20:54:46.020 MQT: stat/tasmota-sw04/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-17T08:39:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"435/699"}}
20:54:46.025 MQT: stat/tasmota-sw04/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.1.180","LogPort":514,"SSId":["SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","0405C80001000680003C5A0A002800000000","08000080","00806000","00004000","00000000"]}}
20:54:46.038 MQT: stat/tasmota-sw04/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
20:54:46.044 MQT: stat/tasmota-sw04/STATUS5 = {"StatusNET":{"Hostname":"tasmota-sw04","IPAddress":"192.168.1.74","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"208.67.220.220","DNSServer2":"208.67.222.222","Mac":"4C:EB:D6:8F:60:05","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:54:46.048 MQT: stat/tasmota-sw04/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.19","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8F6005","MqttUser":"homeassistant","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:54:46.054 MQT: stat/tasmota-sw04/STATUS7 = {"StatusTIM":{"UTC":"2022-10-28T19:54:46","Local":"2022-10-28T20:54:46","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:30","Sunset":"17:36"}}
20:54:46.057 MQT: stat/tasmota-sw04/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:54:46.067 MQT: stat/tasmota-sw04/STATUS10 = {"StatusSNS":{"Time":"2022-10-28T20:54:46","ENERGY":{"TotalStartTime":"2022-10-22T19:56:50","Total":0.409,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":249,"Current":0.000}}}
20:54:46.076 MQT: stat/tasmota-sw04/STATUS11 = {"StatusSTS":{"Time":"2022-10-28T20:54:46","Uptime":"0T01:07:53","UptimeSec":4073,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"50:C7:BF:14:A0:42","Channel":1,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}
20:54:46.081 MQT: stat/tasmota-sw04/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["40003b53","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Issue is random. I will try to capture and update here later.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Random restarts experienced. No obvious trigger.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-10-29:
Pls report output of command ``ButtonDebounce``

trichromic-ab, 2022-10-29:
Hi Theo,

16:28:55.981 CMD: ButtonDebounce
16:28:55.983 SRC: WebConsole from 192.168.1.14
16:28:55.986 CMD: Grp 0, Cmd 'BUTTONDEBOUNCE', Idx 1, Len 0, Pld -99, Data ''
16:28:55.992 MQT: stat/tasmota-sw01/RESULT = {"ButtonDebounce":50}

Also I managed to capture one of the restarts in my syslog yesterday.
Text file attached

Regards,
Alex

________________________________
Trichromic LLP is a limited liability partnership registered in England and Wales under the Partnership No. OC317745
Registered office:12 Station Court, Station Approach, Wickford, Essex, SS11 7AT
VAT Reg. No. GB 882 3845 90

tasmota-sw04 ESP-HTP: Command
tasmota-sw04 ESP-SRC: WebCommand from 192.168.1.19
tasmota-sw04 ESP-CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota-sw04","FriendlyName":["tasmota-sw04"],"Topic":"tasmota-sw04","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T01:43:14","StartupUTC":"2022-10-28T18:46:53","Sleep":50,"CfgHolder":4617,"BootCount":49,"BCResetTime":"2022-10-22T19:56:50","SaveCount":115,"SaveAddress":"F6000"}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-17T08:39:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"435/699"}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"192.168.1.180","LogPort":514,"SSId":["SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","0405C80001000680003C5A0A002800000000","08000080","00806000","00004000","00000000"]}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS5 = {"StatusNET":{"Hostname":"tasmota-sw04","IPAddress":"192.168.1.74","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"208.67.220.220","DNSServer2":"208.67.222.222","Mac":"4C:EB:D6:8F:60:05","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.19","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8F6005","MqttUser":"homeassistant","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS7 = {"StatusTIM":{"UTC":"2022-10-28T20:30:07","Local":"2022-10-28T21:30:07","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:30","Sunset":"17:36"}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS10 = {"StatusSNS":{"Time":"2022-10-28T21:30:07","ENERGY":{"TotalStartTime":"2022-10-22T19:56:50","Total":0.409,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":249,"Current":0.000}}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS11 = {"StatusSTS":{"Time":"2022-10-28T21:30:07","Uptime":"0T01:43:14","UptimeSec":6194,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"50:C7:BF:14:A0:42","Channel":1,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:03"}}}
tasmota-sw04 ESP-MQT: stat/tasmota-sw04/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["40003b53","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 04 00 3F 11 07 3D BA 4F BA 28 3C 68 5B E1 63 69 22
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 04 00 3F 11 07 3D BA 4F BA 28 3E 00 C1 E1 63 69 22
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 3F 9B 25 E1 63 69 20
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 41 33 8B E1 63 69 20
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 42 CC F0 E1 63 69 1F
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 44 67 54 E1 63 69 20
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 46 00 B9 E1 63 69 20
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 47 99 1F E1 63 69 20
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 49 33 83 E1 63 69 20
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 4A CB E9 E1 63 69 1F
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 02 00 3F 11 07 3D BA 4F BA 28 4C 66 4D E1 63 69 20
[... cut ...]
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 03 00 3F 11 04 15 8A 4F BA 28 2F 1E B2 E1 63 6C C9
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 03 00 3F 11 04 15 8A 4F BA 28 30 B9 16 E1 63 6C C9
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 03 00 3F 11 04 15 8A 4F BA 28 32 51 7C E1 63 6C C9
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 03 00 3F 11 04 15 8A 4F BA 28 33 EB E0 E1 63 6C C8
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 03 00 3F 11 04 15 8A 4F BA 28 3D 84 3E E1 63 6C C9
tasmota-sw04 ESP-DMP: 55 5A 02 EE 00 00 03 03 00 3F 11 04 15 8A 4F BA 28 3F 1D A3 E1 63 6C C9

arendst, 2022-10-30:
Pls try latest dev build ``otaurl http://ota.tasmota.com/tasmota/tasmota.bin.gz``, ``upload 1``

arendst, 2022-10-30:
This might also help https://github.com/arendst/Tasmota/issues/7866#issuecomment-596260491

As it is an old solution pls try this instead:

1- Type ``reset 5`` on the console to erase all flash (but leaving wifi credentials)
2- Upload by OTA latest Tasmota with latest core ( http://ota.tasmota.com/tasmota/tasmota.bin.gz )
3- Then type ``reset 3`` (to reset wifi calibration)
4- Do a power cycle in order to let your device recalibrate the wifi. (you need to de-energize your device for some seconds and energize it again)
5- And finally configure again your device and try again.

Jason2866, 2022-10-30:
Updated my Athom Plug to latest development. If there is an issue (i doubt) will report here.

trichromic-ab, 2022-10-30:
No large loads at all. Its been happening on 8 different switches with various loads, some with none.

I'm currently following your 2 suggestions.
I have upgraded one switch to v12.2.0.2 and on another I've followed the reset 5, u/g, reset 3, power cycle procedure.
I will report back after a bit of soak-testing time.

trichromic-ab, 2022-10-30:
BTW, the one I’ve reset is now showing with a Sonoff Basic module and generic template, not Athom Plug v2 anymore.
But it’s still functioning ok with HA

________________________________
Trichromic LLP is a limited liability partnership registered in England and Wales under the Partnership No. OC317745
Registered office:12 Station Court, Station Approach, Wickford, Essex, SS11 7AT
VAT Reg. No. GB 882 3845 90

This might also help #7866 (comment)<https://github.com/arendst/Tasmota/issues/7866#issuecomment-596260491>

Jason2866, 2022-10-31:
Running TasmoAdmin all the time is not a good idea. It will result what you have seen.
TasmoAdmin is doing non stop web queries, after a while this results in a restart.
Stop TasmoAdmin. Case closed.

closed 2022-10-31 by Jason2866

trichromic-ab, 2022-10-31:
Good advice, thanks.
Is it not possible to make Tasmota more resilient and not react like that though?

________________________________
Trichromic LLP is a limited liability partnership registered in England and Wales under the Partnership No. OC317745
Registered office:12 Station Court, Station Approach, Wickford, Essex, SS11 7AT
VAT Reg. No. GB 882 3845 90

Jason2866, 2022-10-31:
Nope, the esp8266 has its limits. Tasmota is NOT designed for using with http
The "m" in the Tasmota stands for mqtt. (T)heo (A)rend(S)t  (M)qtt (OTA) Over The Air
Our recommended Management tool is [TDM](https://github.com/jziolkowski/tdm)

trichromic-ab, 2022-10-31:
AH! Thanks for explaining that.
It’s a great acronym.

And thanks for the pointer to TDM, which I’d not discovered yet.

All the best,
Alex

________________________________
Trichromic LLP is a limited liability partnership registered in England and Wales under the Partnership No. OC317745
Registered office:12 Station Court, Station Approach, Wickford, Essex, SS11 7AT
VAT Reg. No. GB 882 3845 90