arendst/Tasmota #16970: SerialConfig is saved but not applied after reboot (only HW serial)

TheChatty, 2022-10-31:
### PROBLEM DESCRIPTION
If a different serial config than 8N1 is set it will be saved but not applied if module is rebooted.
Even though "SerialConfig" shows valid config data looks like it's processed with 8N1.
To receive valid data again it is necessary to set some wrong config and again back to preferred one.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-12F in development board
- [x] Tasmota binary firmware version number used: 12.2.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool 4.3
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- configure GPIO3 as HW serial RX
- `SerialConfig 7E1` (must be different than before)
- observe valid data being received
- `restart 1`
- `SerialConfig` --> shows 7E1 was saved
- observe data being received with 8N1
- `SerialConfig 7E1` --> data is still being received with 8N1
- `SerialConfig 8N1`
- `SerialConfig 7E1`
- observe valid data being received again

### EXPECTED BEHAVIOUR
A reboot preserves configured serial setting.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-10-31:
@arendst I may have overlooked it, but I don't seem to find anything setting the config at boot in case Serial bridge is configured.

TheChatty, 2022-10-31:
As I wrote above this setting is resilent. But even though you see the correct setting being saved it is NOT applied to the GPIO.

arendst, 2022-11-01:
AFAIK ``SetSSerialBegin()`` does the trick executed at init time and every time a user changes baudrate and/or config....

barbudor, 2022-11-01:
@arendst, that seems to be for the software serial bridge as it use `Settings->sbaudrate` and `Settings->sserial_config`

The topic here is Serial bridge which uses `Baudrate` => `Settings->baudrate` and `SerialConfig` => `Settings->serial_config`
Shouldn't this line:
https://github.com/arendst/Tasmota/blob/da65c8798bb192465f4fabd8a95e4c4c6b54f87b/tasmota/tasmota.ino#L519
be something like:
```C++
SetSerial(Settings->baudrate * 300, ConvertSerialConfig(Settings->serial_config);
```
But I don't really understand when the above test regarding "changing module" would occur

TheChatty, 2022-11-02:
I did not change configuration but reboot.

arendst, 2022-11-02:
I'll have a look

arendst, 2022-11-02:
> But I don't really understand when the above test regarding "changing module" would occur

If a user changes module, either with command ``module`` or in the GUI, the serial (debugging) needs to be reset to it's default state as a new module is not supposed the change the baudrate and serialconfig from 115200-8N1 where the previous module could have used totally different serial parameters.

At restart serial is initiated with default settings 115200-8N1 to keep logging visible. Once settings have been found and read the baudrate is changed to settings parameters by function ``SetSerialBaudrate()``. Changing serial config is tricky as there is no means to read the current config other than setting the config once to a known state and store that locally (as Tasmota does).

There seems to be a need to change serial config at restart from 8N1 to Settings->serial_config. Using function ``SetSerial()`` won't do as it is supposed to be called by a driver taking over the serial interface from debugging.

I'll add a new init function setting both baudrate and serial config at restart if they differ from 115200 and 8N1.

arendst, 2022-11-02:
I'll add this function
```
void SetSerialInitBegin(void) {
  TasmotaGlobal.baudrate = Settings->baudrate * 300;
  if ((GetSerialBaudrate() != TasmotaGlobal.baudrate) || (TS_SERIAL_8N1 != Settings->serial_config)) {
    SetSerialBegin();
  }
}
```
replacing the initial call to ``SetSerialBaudrate()`` to be merged soon.