arendst/Tasmota #16985: Power switches on after temperature sensor is not alive

blacknell, 2022-11-03:
### PROBLEM DESCRIPTION
After the reported temperature is detected as not alive the power switch is turned on, resulting in huge energy costs.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Elite POW320D
- [X] Tasmota binary firmware version number used: 12.2.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
07:31:36.841 SRC: Backlog
07:31:36.845 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
07:31:36.856 MQT: stat/tasmota_CC7FF4/RESULT = {"Module":{"0":"Sonoff POWR320D"}}
07:31:37.092 SRC: Backlog
07:31:37.097 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
07:31:37.129 MQT: stat/tasmota_CC7FF4/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"9313":"Relay_b2"},"GPIO3":{"0":"None"},"GPIO4":{"9312":"Relay_b1"},"GPIO5":{"576":"LedLink_i"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"9280":"TM1621 DAT"},"GPIO15":{"0":"None"},"GPIO16":{"3104":"CSE7766 Rx"},"GPIO17":{"0":"None"},"GPIO18":{"320":"Led_i1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"9184":"TM1621 CS"},"GPIO26":{"9248":"TM1621 RD"},"GPIO27":{"9216":"TM1621 WR"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
07:33:23.481 SRC: Backlog
07:33:23.485 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
07:33:23.495 MQT: stat/tasmota_CC7FF4/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":139,"Free":372,"Rules":"ON mqtt#connected DO Subscribe DnTemp, stat/other-device-topic/SENSOR, Temperature ENDON   ON Event#DnTemp DO TempMeasuredSet %value% ENDON"}}
07:33:23.691 SRC: Backlog
07:33:23.695 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
07:33:23.708 MQT: stat/tasmota_CC7FF4/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":301,"Free":210,"Rules":"ON Power1#boot DO Backlog sensorinputset 1;controllermodeset 2;thermostatmodeset 1;temptargetset %mem1% ENDON ON mqtt#connected DO Publish2 stat/TestTopic/targetTempValue {\"Temp\":%mem1%} ENDON ON mem1#state DO Backlog temptargetset %value%;Publish2 stat/TestTopic/targetTempValue {\"Temp\":%mem1%} ENDON"}}
07:33:23.941 SRC: Backlog
07:33:23.944 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
```
- [X] Provide the output of this command: `Status 0`:
```lua
07:34:19.407 MQT: stat/tasmota_CC7FF4/STATUS0 = {"Status":{"Module":0,"DeviceName":"POWR320D","FriendlyName":["POWR320D"],"Topic":"tasmota_CC7FF4","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T10:56:01","StartupUTC":"2022-11-02T19:38:18","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2022-11-01T13:37:02","SaveCount":159},"StatusFWR":{"Version":"12.2.0(wight-therm)","BuildDateTime":"2022-11-02T06:57:36","Core":"2_0_5","SDK":"v4.4.3.4","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"402/699"},"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sandalwood",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["80008009","2805C80001000680003C5A0A192800000000","00000088","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":1448,"Free":1431,"Heap":140,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400844D2","20200000","D434482D","00000001"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,39,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"},"StatusNET":{"Hostname":"tasmota-CC7FF4-8180","IPAddress":"192.168.100.86","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"C0:49:EF:CC:7F:F4","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"hatest.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CC7FF4","MqttUser":"mqtt","MqttCount":6,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-11-03T06:34:19","Local":"2022-11-03T07:34:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:40","Sunset":"17:26"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2022-11-03T07:34:19","ENERGY":{"TotalStartTime":"2022-11-01T13:37:02","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":245,"Current":0.000},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":18.00,"CtrDutyCycleRead":100}},"StatusSTS":{"Time":"2022-11-03T07:34:19","Uptime":"0T10:56:01","UptimeSec":39361,"Heap":139,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":6,"Berry":{"HeapUsed":3,"Objects":45},"POWER":"ON","Wifi":{"AP":1,"SSId":"sandalwood","BSSId":"BC:A5:11:23:4B:90","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- `THERMOSTATSETMODE 1`
- `TEMPTARGETSET 18.0`
- Send payload `{temperature:22.5}` to topic `stat/other-device-topic/SENSOR`
- After a short while power turns off
- 1 hour later approx, temperature sensor is marked not alive. `07:42:36.775 Thermostat[ctr_output].status.sensor_alive: 0`
- Main menu shows measured temperature is 0

### EXPECTED BEHAVIOUR
If thermostat loses measured temperature readings it should failsafe to off. It seems to be hunting 18 degrees because measured temperature is being considered as 0 despite knowing the sensor is not alive

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Th0maz, 2022-11-03:
Not sure if this helps as a workaround, but the thermostat rule example has some kind of watchdog for a missing temperature sensor:
[https://tasmota.github.io/docs/Rules/#simple-thermostat-example](url)

blacknell, 2022-11-03:
> Not sure if this helps as a workaround, but the thermostat rule example has some kind of watchdog for a missing temperature sensor: [https://tasmota.github.io/docs/Rules/#simple-thermostat-example](url)

Don't think it is using the Thermostat module, rather the example implements a simple on/off thermostat.

stefanbode, 2022-11-21:
My idea would be to define a PULSETIME on the relay of double of the expected update time (TELEPERIOD) and in RULES I would set the relay to ON if required. If the TEMP is below it switches OFF and if the TEMP is missing after some time the PULSETIME will kick in and power it OFF. I did the same with my pool where sometimes the DS18B20 just disappear until it reboots..

Jason2866, 2022-11-24:
Closing since Tasmota has no routines to check sensors.

closed 2022-11-24 by Jason2866

blacknell, 2022-11-24:
> Closing since Tasmota has no routines to check sensors.

I don't understand the logic here.

The thermostat loop does compare the measured temperature and then acts accordingly. With no sensor present, and it does know this because it sets Alive to false, it aims to reach target temperature thinking measured is 0