arendst/Tasmota #16986: Enabling web_password leads to http response failure

isaacncz, 2022-11-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I enable #define WEB_PASSWORD   "xxxx" at my_user_config.h. After entering the wifi credentials, suppose it redirect showing a new IP address but the web page not available instead. 
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-11-03:
WebPassword is not really meant for the AP mode, which is for initial config, only after having the Wifi configured for having a connection to your network.

isaacncz, 2022-11-04:
hi thanks for replying, is it because of the web password is not stored after the authentication. so leading to the password to be missing during the redirect ?

sfromis, 2022-11-04:
I've not explored details, just observe that setting web password at build time without also setting wifi credentials looks like too much of an edge case to have been included in design/testing.

barbudor, 2022-11-07:
Maybe @ascillato is more aware of what's happenning there

ascillato, 2022-11-08:
Hi,

This is a bug since Webpassword was not designed for AP mode. Sorry.

isaacncz, 2022-11-08:
hi all, thanks for the reply. Just want to confirm that this is expected.

closed 2022-11-08 by isaacncz