arendst/Tasmota #16993: Crash on invalid SML script

TheChatty, 2022-11-04:
### PROBLEM DESCRIPTION
When meter description contains "." instead of "," Tasmota crashes when trying to save with script activated.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-12F
- [ ] Tasmota binary firmware version number used: 12.2.02
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Try to save the following (activated) script:
```
>D
>B
=>sensor53 r
>M 1
+1,14.o,0,9600,SML
1,1-0:1.7.0*255(@1,P_in,W,P_in,18
1,1-0:21.7.0*255(@1,L1,W,L1,18
1,1-0:41.7.0*255(@1,L2,W,L2,18
1,1-0:61.7.0*255(@1,L3,W,L3,18
1,1-0:1.8.0*255(@1,E_in,kWh,E_in,19
1,1-0:2.8.0*255(@1,E_out,kWh,E_out,19
1,1-0:0.0.0*255(@1,Netzbetreiber-ID,,NetID,0
1,0-0:96.1.255*255(@#),Seriennummer,,serial,0
#
```

### EXPECTED BEHAVIOUR
No crash just an error message.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

hagenbuch, 2022-11-24:
May not be related to your problem but I discovered that I had to add an empty line or # in the >D section!

If I read your >M area well, you are trying to have 18 or 19 decimals (see P_in, L1, L2, L3, E_in, E_out - is that reasonable?

The rest looks good. I assume you know that you can turn on SML logging on te console by issueing

    Sensor53 D1

Turn of with 

    Sensor53 D0

TheChatty, 2022-11-24:
@hagenbuch Unfortunately it is unrelated. The script was taken from [here](https://tasmota.github.io/docs/Smart-Meter-Interface/#easymeter-q3d-q3da1024-obis). It is working as is. The problem here is missing syntax verification for meter description (line starting with +).

@gemu2015 I mistakenly posted the problem here instead of your repo. Is this something worth fixing?

gemu2015, 2022-11-24:
No normally you can also post here

there are lots of possible syntax errors in scripts of all kinds.
it would blow code size on such small cpus to catch all these errors and therefore syntax has to be correct by design.
if you find an error in a documented script, please edit and correct the doku

TheChatty, 2022-11-25:
The script is fine. I just observed a crash on a typo I introduced accidentally.

Resolution: won't fix

closed 2022-11-25 by TheChatty