arendst/Tasmota #17002: Sensor reading missing after upgrade

AnanasPfirsichSaft, 2022-11-05:
### PROBLEM DESCRIPTION
I upgraded from 12.1.1 to 12.2.0 and the sensor data disappeared from the index page. Only "ON" or "OFF" are now displayed. If I downgrade, the sensor data is back again. The device is a [Delock 11827 "Power Socket Switch"](https://templates.blakadder.com/delock_11827.html).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Delock 11827
- [ ] Tasmota binary firmware version number used: 12.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
Version 12.1.1
{"NAME":"Delock 11827","GPIO":[0,0,0,32,2688,2656,0,0,2624,576,224,0,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Delock 11827"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2688":"HLW8012 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

Version 12.2.0
{"NAME":"Delock 11827","GPIO":[0,0,0,32,2688,2656,0,0,2624,576,224,0,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Delock 11827"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2688":"HLW8012 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
Version 12.1.1 
{"Status":{"Module":0,"DeviceName":"MagicWand","FriendlyName":["MagicWand"],"Topic":"tasmota_D77257","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"1T04:05:48","StartupUTC":"2022-11-04T09:27:40","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-12-27T21:24:43","SaveCount":912,"SaveAddress":"F6000"}}
{"StatusFWR":{"Version":"12.1.1(lite)","BuildDateTime":"2022-08-25T11:37:09","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"351/699"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MagicNet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
{"StatusMEM":{"ProgramSize":512,"Free":488,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000","04000020","00000000"],"Drivers":"1,2,3,4,9,10,12,16,20,21,37,62","Sensors":"3"}}
{"StatusNET":{"Hostname":"tasmota-D77257-4695","IPAddress":"192.168.100.192","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.100.1","DNSServer2":"0.0.0.0","Mac":"FF:FF:FF:FF:FF:FF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
{"StatusTIM":{"UTC":"2022-11-05T13:33:28","Local":"2022-11-05T14:33:28","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:43","Sunset":"17:23"}}
{"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
{"StatusSNS":{"Time":"2022-11-05T14:33:28","ENERGY":{"TotalStartTime":"2021-12-27T21:24:43","Total":0.959,"Yesterday":0.003,"Today":0.002,"Power": 2,"ApparentPower":17,"ReactivePower":17,"Factor":0.09,"Voltage":261,"Current":0.065}}}
{"StatusSTS":{"Time":"2022-11-05T14:33:28","Uptime":"1T04:05:48","UptimeSec":101148,"Vcc":3.499,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"MagicNet","BSSId":"EE:EE:EE:EE:EE:EE","Channel":6,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":5,"Downtime":"0T00:00:13"}}}

Version 12.2.0
{"Status":{"Module":0,"DeviceName":"MagicWand","FriendlyName":["MagicWand"],"Topic":"tasmota_D77257","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:06","StartupUTC":"2022-11-05T13:36:38","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-12-27T21:24:43","SaveCount":916,"SaveAddress":"FA000"}}
{"StatusFWR":{"Version":"12.2.0(lite)","BuildDateTime":"2022-10-17T08:41:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"351/699"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MagicNet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
{"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F000782","04108001","00000002","00000000","00000000","00000000","00000000","04000000","00000000"],"Drivers":"1,2,4,9,10,12,16,20,21,62","Sensors":""}}
{"StatusNET":{"Hostname":"tasmota-D77257-4695","IPAddress":"192.168.100.192","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.100.1","DNSServer2":"0.0.0.0","Mac":"FF:FF:FF:FF:FF:FF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
{"StatusTIM":{"UTC":"2022-11-05T13:37:44","Local":"2022-11-05T14:37:44","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:43","Sunset":"17:23"}}
{"StatusSNS":{"Time":"2022-11-05T14:37:44"}}
{"StatusSTS":{"Time":"2022-11-05T14:37:44","Uptime":"0T00:01:06","UptimeSec":66,"Vcc":3.500,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"MagicNet","BSSId":"EE:EE:EE:EE:EE:EE","Channel":6,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Well, make the sensor data available again :)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I bought them a while ago. They have Tasmota pre-installed and I followed the update path without any issues so far. And applied the (above linked) device template from the database.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-11-05:
To keep size of the `tasmota-lite.bin.gz` binary low enough, energy monitoring is no longer included. Should work if you upgrade to full `tasmota.bin.gz`.

eziologico, 2022-11-06:
Hi,
I'm experiencing the same issue with tasmota_sensors.bin.gz v.12.2.0 and a BMP280 T/H/B sensor, so I had to go back and downgrade to the previous version 12.1.1.
Trying to compile the tasmota_sensors v.12.2.0 by myself you see that all the features that use the I2C Bus are not enabled.
There seems to be no #define USE_I2C defined or there is #undef USE_I2C somewhere ...
Compiling my v.12.2.0 with just the drivers and features I need it works!

Thanks,
Ezio

sfromis, 2022-11-06:
Cannot be the "same issue", whatever happens with your sensors is unrelated to energy monitoring. Better not jump into unrelated threads.

And using tasmota-sensors 12.2 with BMP280 (T/B, not H) or BME280 (T/H/B) I see no problems, the sensors show up just as expected, both precompiled and self-built. If you are doing a build not getting `USE_I2C`, you may have some issue with how you are setting that up.

closed 2022-11-06 by Jason2866