arendst/Tasmota #17017: BME680 + INA219 = fail

tasmotari, 2022-11-07:
### PROBLEM DESCRIPTION
Sensor **BME680** together with **INA219** on i2c bus fail shortly after any activity from last one. If use BME680 alone it works fine and stable. I also tried to use INA219 on separate second i2c bus (modified driver xsns_13_ina219.ino and support.ino), it works, but the behavior is the same - BME680 just shortly disappear from the i2c bus, INA219 stay alone.

The power is stable. ESP32 use external 5V adpter via microUSB, sensors powered from 1A regulator LD1117V33.

Program Version: 12.2.0(tasmota)
Build &Time/Date: 2022-11-06T18:49:39
Core/SDK Version: 2_0_5/v4.4.3.4

`04:14:43.409 CMD: i2cscan
04:14:43.442 MQT: stat/MySensor/RESULT = {"I2CScan":"Device(s) found at 0x77"}
04:15:10.282 MQT: tele/MySensor/STATE = {"Time":"2022-11-06T04:15:10","Uptime":"0T00:01:10","UptimeSec":70,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":570,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"SMRTHM12","BSSId":"10:01:10:01:10:01","Channel":5,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}
04:15:10.313 MQT: tele/MySensor/SENSOR = {"Time":"2022-11-06T04:15:10","BME680":{"Temperature":22.2,"Humidity":51.6,"DewPoint":11.8,"Pressure":984.2,"Gas":82.75},"INA219":{"Id":40,"Voltage":1.34,"Current":0.055,"Power":0},"PressureUnit":"hPa","TempUnit":"C"}
04:15:26.217 RUL: INA219#CURRENT>0.01 performs "publish stat/MySensor/FIRE {"POWER":"ON"}"
04:15:26.224 MQT: stat/MySensor/FIRE = {"POWER":"ON"}
04:16:13.775 MQT: tele/MySensor/STATE = {"Time":"2022-11-06T04:16:13","Uptime":"0T00:02:13","UptimeSec":133,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":2064,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"wifi","BSSId":"10:01:10:01:10:01","Channel":5,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:04"}}
04:16:13.810 MQT: tele/MySensor/SENSOR = {"Time":"2022-11-06T04:16:13","BME680":{"Temperature":22.2,"Humidity":51.6,"DewPoint":11.8,"Pressure":984.2,"Gas":82.75},"INA219":{"Id":40,"Voltage":1.35,"Current":-0.000,"Power":-0},"PressureUnit":"hPa","TempUnit":"C"}
04:16:15.941 RUL: INA219#CURRENT<0.01 performs "publish stat/MySensor/FIRE {"POWER":"OFF"}"
04:16:15.950 MQT: stat/MySensor/FIRE = {"POWER":"OFF"}
04:16:20.101 CMD: I2CScan
04:16:20.135 MQT: stat/MySensor/RESULT = {"I2CScan":"No devices found"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-D0WDQ6 rev.1
- [x] Tasmota binary firmware version number used: 12.2.0(tasmota)
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmota Web Installer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
`18:24:32.019 MQT: stat/MySensor/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:24:32.253 MQT: stat/MySensor/RESULT = {"Module":{"1":"ESP32-DevKit"}}
18:24:32.515 MQT: stat/MySensor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"480":"Buzzer"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"609":"I2C SCL2"},"GPIO27":{"641":"I2C SDA2"},"GPIO32":{"640":"I2C SDA1"},"GPIO33":{"608":"I2C SCL1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}`
  

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
`18:29:43.137 CMD: Rule1
18:29:43.151 MQT: stat/MySensor/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":146,"Free":365,"Rules":"ON INA219#Current>0.01 DO publish stat/%topic%/FIRE {\"POWER\":\"ON\"} ENDON ON INA219#Current<0.01 DO publish stat/%topic%/FIRE {\"POWER\":\"OFF\"} ENDON"}}`

```
- [x] Provide the output of this command: `Status 0`:
`18:31:24.855 MQT: stat/MySensor/STATUS = {"Status":{"Module":1,"DeviceName":"MySensor","FriendlyName":["MySensor"],"Topic":"MySensor","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:31:24.878 MQT: stat/MySensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T14:17:25","StartupUTC":"2022-11-06T03:13:59","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2025-06-15T16:06:40","SaveCount":193}}
18:31:24.895 MQT: stat/MySensor/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-11-06T18:49:39","Core":"2_0_5","SDK":"v4.4.3.4","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"363/699"}}
18:31:24.910 MQT: stat/MySensor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SMRTHM12",""],"TelePeriod":60,"Resolution":"548580C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00000000","00000000"]}}
18:31:24.944 MQT: stat/MySensor/STATUS4 = {"StatusMEM":{"ProgramSize":1442,"Free":1437,"Heap":138,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","85DA9BC0","E0360DD7","400840D2","20200000","D434482D","00000001"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,50,51,52,54,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
18:31:24.970 MQT: stat/MySensor/STATUS5 = {"StatusNET":{"Hostname":"MySensor-3520","IPAddress":"10.0.1.11","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.1.1","DNSServer2":"0.0.0.0","Mac":"01:10:15:C1:ED:10","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
18:31:24.988 MQT: stat/MySensor/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.10","MqttPort":1883,"MqttClientMask":"MySensor","MqttClient":"MySensor","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:31:25.008 MQT: stat/MySensor/STATUS7 = {"StatusTIM":{"UTC":"2022-11-06T17:31:25","Local":"2022-11-06T18:31:25","StartDST":"2022-11-06T02:00:00","EndDST":"2022-11-06T03:00:00","Timezone":"+01:00","Sunrise":"08:35","Sunset":"17:26"}}
18:31:25.049 MQT: stat/MySensor/STATUS10 = {"StatusSNS":{"Time":"2022-11-06T18:31:25","BME680":{"Temperature":22.0,"Humidity":50.4,"DewPoint":11.2,"Pressure":982.9,"Gas":156.10},"PressureUnit":"hPa","TempUnit":"C"}}
18:31:25.071 MQT: stat/MySensor/STATUS11 = {"StatusSTS":{"Time":"2022-11-06T18:31:25","Uptime":"0T14:17:26","UptimeSec":51446,"Heap":140,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"wifi","BSSId":"10:01:10:01:10:01","Channel":5,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}}`

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-11-07:
That's not a conflict with addresses
Is the BME stable when INA disabled by i2cdriver?
(keep connected, same build)

sfromis, 2022-11-07:
You likely have a hardware issue, maybe with wiring. Just hooked BME680 and INA219 up to an ESP32. Getting values from both at the same time.
`{"I2CScan":"Device(s) found at 0x29 0x3c 0x40 0x77"}`

sfromis, 2022-11-07:
Since your first I2CScan says `{"I2CScan":"Device(s) found at 0x77"}`, something was already screwed up at that time, even if both sensors appeared in the TelePeriod JSON payload.

tasmotari, 2022-11-07:
> Since your first I2CScan says `{"I2CScan":"Device(s) found at 0x77"}`, something was already screwed up at that time, even if both sensors appeared in the TelePeriod JSON payload.

@sfromis It's LOG from Self-compiled firmware, where INA driver modified for second i2c bus. I think, that I2CScan don't show second bus devices?

sfromis, 2022-11-07:
I saw that, but output from that failed experiment was not useful to use as documentation. Expect the documentation you post to be carefully read for any clues. But never mind.....

Your hardware setup is still suspect.

tasmotari, 2022-11-07:
I just changed the board for new ESP32-D0WD-V3 rev.3, but issue is the same. **As soon as Current is passed via INA - BME hung up.**

sfromis, 2022-11-07:
By "hardware", I did not mean the dev board as such, but the rest of the build.

tasmotari, 2022-11-07:
I read in a nearby topic the similar problem with BMP280+AHT10, and the main idea was "AHT10 is a pig on the I2C bus". Is there a possibility that INA219 sensor have the same effect?

sfromis, 2022-11-07:
Not in my experience. I've seen no trouble in a combined setup, and I've seen users describing solar powered builds having INA219 along with I2C other sensors. AHT10 is a known outlier case of a sensor with inappropriate bus behavior, but INA219 is commonly used, and seems to work fine.

tasmotari, 2022-11-07:
@barbudor INA is measuring a curent in external very small power circuit. 

`INA219":{"Id":40,"Voltage":1.34,"Current":0.055,"Power":0}`

Moreover, power reserve for both sensors is huge 1A compared to consumption each of them.

barbudor, 2022-11-07:
Capacity of the power supply doesn't take into account wires which can act as resistance or inductance.
Current peak can create voltage peaks which can destabilize some chips. May be time to undust the scope.

Nevertheless you seem to have pointed that the problem occurs only when the INA start seeing current.
For me it seems to be a clue of something that start to happen badly at that time.
Could be a wiring problem, could be a power issue, ...
Not knowing your circuit and how you wired it, I can't suggest anything specific but that's probabbly an area where to look at in details.