arendst/Tasmota #17026: Sonoff POW Origin with Tasmota Version 12.2.0 in continuous reset loop after startup

ot3, 2022-11-08:
I've installed Tasmota Version 12.2.0 on a brand new Sonoff POW Origin using the web installer, including erasing the flash memory. I'm using an FTDI programmer that I've used for years for my ESP8266 projects and has always worked. This is my first ESP32.
The web installer completes normally but never makes it to the point where I can select a wifi SID.

While researching this issue, I've learned that sometimes the USB power supply through the FTDI programmer can't deliver enough current. I therefore also tried operating the device while connected to a professional lab PSU, set for 3.3V and a maximum current of 2A. The result is always the same. Below is a section of the Termite logfile:
```
00:00:00.003-209/49 HDW: ESP32-D0WD-V3 
00:00:00.023-207/49 UFS: FlashFS mounted with 308 kB free
00:00:00.052 CFG: Loaded from File, Count 2
00:00:00.068 QPC: Count 5
00:00:00.502 BRY: Berry initialized, RAM used=4711 bytes
00:00:00.524 Project tasmota - Tasmota Version 12.2.0(tasmota)-2_0_5(2022-10-17T08:39:59)
00:00:00.857 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.003-209/49 HDW: ESP32-D0WD-V3 
00:00:00.023-207/49 UFS: FlashFS mounted with 308 kB free
00:00:00.052 CFG: Loaded from File, Count 2
00:00:00.068 QPC: Count 6
00:00:00.502 BRY: Berry initialized, RAM used=4711 bytes
00:00:00.524 Project tasmota - Tasmota Version 12.2.0(tasmota)-2_0_5(2022-10-17T08:39:59)
00:00:00.857 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.003-209/49 HDW: ESP32-D0WD-V3 
00:00:00.023-207/49 UFS: FlashFS mounted with 308 kB free
00:00:00.052 CFG: Loaded from File, Count 2
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WD-V3 
00:00:00.022 UFS: FlashFS mounted with 312 kB free
00:00:00.027 CFG: Loaded from NVS, Count 2
00:00:00.031 FRC: Some settings have been reset (2)
00:00:00.464 BRY: Berry initialized, RAM used=4711 bytes
00:00:00.487 Project tasmota - Tasmota Version 12.2.0(tasmota)-2_0_5(2022-10-17T08:39:59)
00:00:00.795 WIF: WifiManager active for 3 minutes
ets Jul 29 2019 12:21:46
```

It seems to me (but I'm not certain) that the hardware is OK. Does anyone know what's happening here and suggest next steps?

closed 2022-11-08 by close-issue-app[bot]

sfromis, 2022-11-08:
What you posted looks normal, waiting for you to connect to the AP, to complete the onboarding process for connecting to your wifi network. And it happens that devices need mains power to operate on wifi, this is not just about amps available on 3.3V.

ot3, 2022-11-08:
Thanks for responding. The problem is there's nothing to connect to (using the web installer which just keeps spinning). I tried both the lab PSU and mains, no difference. I then tried to use ESP-flasher which is a bit more chatty; no luck though:
```
Using 'COM3' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ5-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: C0:49:EF:D1:BD:D4
Uploading stub...
Unexpected error: Error putting ESP in stub flash mode: Invalid head of packet (0x00): Possible serial noise or corruption.
```

From some posts on the web there are hints (but no solutions) that pressing the device's button before applying power to bring it into flash mode, may not be enough. The problem is that there's only one button. I hope this triggers some new insights since I'm completely stuck.

arendst, 2022-11-08:
You had it installed in the first entry 

Do that again. Remove serial cables. Connect AC and configure using webinterface over tasmota access point.

Simple

ot3, 2022-11-08:
I know it's simple as I've done it many times with ESP8266's. However I always got the AP SID but in this case nothing happens as the AP does not show up; not on AC, not via serial and not with a separate PSU. I've used 2 different programmers and 2 different tools. What am I missing?

ot3, 2022-11-09:
I just reread my posts and I see I mentioned running on AC/mains twice. Unless you mean something else than "connected to the grid" with the term "mains power" I just don't get it. Connecting to mains does not bring up the AP, no matter how long I wait and therefore there's nothing to connect to.

sfromis, 2022-11-09:
With `WIF: WifiManager active for 3 minutes`, the one log you posted shows Tasmota working as it should to provide an AP. After that, it is up to the hardware, including how it is powered. Nothing Tasmota can do about that, and not up to details of the tools you used to get to that point.

ot3, 2022-11-09:
Thanks, that's a very clear answer. 

At this stage - after activating the wifi manager - are any of the LEDs supposed to be on or flashing?. In my case the LEDs don't show any activity whereas they do with an out-of-the-box unit.