arendst/Tasmota #17027: Sensor delivers four values, only the first is sent to Domoticz

manjh, 2022-11-08:
### PROBLEM DESCRIPTION
I use a SenseAir K33 sensor, which delivers CO2, temperature, humitity and dewpoint. I can see all four values on the console line from the sensor, and I also see them on the main Tasmota page. 
I filled the idx's on the Domoticz config. But only one value is passed to Domoticz (CO2). I tried four separate devices for the values, but also tried to combine some of them. No change.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): SenseAir K33
- [ ] Tasmota binary firmware version number used: _____
- [ ] <html>
<body>

Program Version
| 12.2.0.3(allsensors)
-- | --
2022-11-07T14:33:16
2_7_4_9/2.2.2-dev(38a443e)

</body>
</html>
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Windows browser
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:49:07.173 MQT: stat/tasmota_C953AA/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:49:07.385 MQT: stat/tasmota_C953AA/RESULT = {"Module":{"18":"Generic"}}
19:49:07.640 MQT: stat/tasmota_C953AA/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1632":"SAir Rx"},"GPIO13":{"1600":"SAir Tx"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
n/a

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:50:34.102 MQT: stat/tasmota_C953AA/STATUS = {"Status":{"Module":18,"DeviceName":"SenseAir testbed 2","FriendlyName":["SenseAir testbed 2"],"Topic":"tasmota_C953AA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:50:34.109 MQT: stat/tasmota_C953AA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":" ","RestartReason":"Hardware Watchdog","Uptime":"0T00:31:15","StartupUTC":"2022-11-08T18:19:19","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2022-11-08T14:02:18","SaveCount":45,"SaveAddress":"F7000"}}
19:50:34.113 MQT: stat/tasmota_C953AA/STATUS2 = {"StatusFWR":{"Version":"12.2.0.3(allsensors)","BuildDateTime":"2022-11-07T14:33:16","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"327/699"}}
19:50:34.118 MQT: stat/tasmota_C953AA/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hans3",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:50:34.137 MQT: stat/tasmota_C953AA/STATUS4 = {"StatusMEM":{"ProgramSize":737,"Free":264,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F908787","04146001","FFFFB7CF","8F7417FF","303E0AD3","1159785A","26204FFF","5408F0F3","00000080"],"Drivers":"1,2,3,4,7,8,9,10,12,14,15,20,21,24,28,29,34,43,47,50,60,62","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,28,29,30,31,32,38,40,41,42,44,45,46,50,51,54,55,56,57,58,59,63,64,65,66,67,70,71,72,73,74,75,76,78,79,81,82,85,86,88,89,90,91,92,93"}}
19:50:34.146 MQT: stat/tasmota_C953AA/STATUS5 = {"StatusNET":{"Hostname":"tasmota-C953AA-5034","IPAddress":"10.0.0.35","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:C9:53:AA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:50:34.151 MQT: stat/tasmota_C953AA/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.199","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C953AA","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:50:34.159 MQT: stat/tasmota_C953AA/STATUS7 = {"StatusTIM":{"UTC":"2022-11-08T18:50:34","Local":"2022-11-08T19:50:34","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:48","Sunset":"17:19"}}
19:50:34.168 MQT: stat/tasmota_C953AA/STATUS10 = {"StatusSNS":{"Time":"2022-11-08T19:50:34","Kx0":{"CarbonDioxide":1104,"Temperature":21.5,"Humidity":64.7,"DewPoint":14.6},"TempUnit":"C"}}
19:50:34.174 MQT: stat/tasmota_C953AA/STATUS11 = {"StatusSTS":{"Time":"2022-11-08T19:50:34","Uptime":"0T00:31:15","UptimeSec":1875,"Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Hans3","BSSId":"3C:84:6A:FF:9B:F2","Channel":9,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

19:52:04.132 CMD: weblog 4
19:52:04.139 MQT: stat/tasmota_C953AA/RESULT = {"WebLog":4}
19:52:04.285 DBG: SenseAir response error 2
19:52:04.534 DBG: SenseAir response error 2
19:52:04.842 CFG: Saved to flash at F6, Count 46, Bytes 4096
19:52:05.778 DBG: SenseAir response error 2
19:52:06.023 DBG: SenseAir response error 2
19:52:07.304 DBG: SenseAir response error 2
19:52:07.555 DBG: SenseAir response error 2
19:52:08.812 DBG: SenseAir response error 2
19:52:09.062 DBG: SenseAir response error 2
19:52:10.305 DBG: SenseAir response error 2
19:52:10.555 DBG: SenseAir response error 2
19:52:11.790 DBG: SenseAir response error 2
19:52:12.037 DBG: SenseAir response error 2
19:52:13.291 DBG: SenseAir response error 2
19:52:13.542 DBG: SenseAir response error 2
19:52:14.785 DBG: SenseAir response error 2
19:52:15.027 DBG: SenseAir response error 2
19:52:16.279 DBG: SenseAir response error 2
19:52:16.533 DBG: SenseAir response error 2
19:52:17.811 DBG: SenseAir response error 2
19:52:18.058 DBG: SenseAir response error 2
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Simply connect the sensor.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
All four values to be passed to Domoticz, not just one

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

manjh, 2022-11-08:
Sensor connected to D6/D7. All values look correct.

arendst, 2022-11-10:
The current driver only supports sending the co2 value to domoticz. In this case use a rule to send other values to domoticz too.

See the rule cookbook in tasmota docs how to do this. Look for command dzsend.

Background: if the driver would send temp and hum too it would overwrite other default sensors. That's why I choose not to send them. Using rules is a much more flexible way of communicating with domoticz.

manjh, 2022-11-10:
I understand. That is inherent to the approach: GPIO is start point. 
If the sensor had been the start point, it would have been easier to add properties like GPIO, values, IDX's. (ESP-Easy works that way)
Thanks for the tip on dzsend, I will have a look over the weekend.
Hans Man