arendst/Tasmota #17029: Development release 12.2.0.3 OTA flash problem.

JackTalisker, 2022-11-09:
### PROBLEM DESCRIPTION
I regularly update all my devices with stable releases and from time to time I test the development releases.
I tried to update my Sonoff 4ch pro R3 (roller shutter control configured) from 12.2.0 to 12.2.0.3 using OTA as usual.
For the first time in 4 years this bricked my device and had to reflash it with Serial-to-USB Adapter with successful recover.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Sonoff 4ch Pro R3
- [X] Tasmota binary firmware version number used: ota.tasmota.com/tasmota/tasmota.bin.gz 12.2.0.2 precompiled the 20221107
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmota OTA gui
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`: not applicable
```lua
  Configuration output here: not applicable

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: not applicable

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: not applicable

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here: not applicable

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Launched OTA update via GUI with url http://ota.tasmota.com/tasmota/tasmota.bin.gz

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Reboot of updated device

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-09:
Which version do you used? 12.2.0.2 or 12.2.0.3?

> Tasmota binary firmware version number used: ota.tasmota.com/tasmota/tasmota.bin.gz 12.2.0.2 precompiled the 20221107
 Pre-compiled

Jason2866, 2022-11-09:
If it is a "old" ESP8285 the change to flash mode `dio` is the reason.
The change to `dio` has just been reverted. Flash mode for all ESP8266 is mode `dout` again.

arendst, 2022-11-09:
Fixed in v12.2.0.4

JackTalisker, 2022-11-09:
> If it is a "old" ESP8285 the change to flash mode `dio` is the reason. The change to `dio` has just been reverted. 
Yes I supposed that it was the "dio" instruction. The chip id is ESP8285 indeed.
I tried 12.2.0.3 (I'm going to edit the ssue request above).

> Flash mode for all ESP8266 is mode `dout` again.

Thank you

JackTalisker, 2022-11-09:
> Fixed in v12.2.0.4

Thank you.

closed 2022-11-09 by Jason2866