arendst/Tasmota #17038: Scripting Google Chart "cnth" Problem with Tasmota 12.x

ottelo9, 2022-11-11:
### PROBLEM DESCRIPTION
Scripting (not berry) problem with the Google Chart function "cnth" (cnthX/Y not yet in the documentation). @gemu2015 added this functionality to "xdrv_10_scripter.ino" on [31. March 2022](https://github.com/arendst/Tasmota/commit/d176aeaaf64899d68bbbe4263e34ddc0c1fe29c8) . With this you can create charts with seconds at x-axis. With cstr="cnth"+s(hours*240)+"/240"  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266
- [x] Tasmota binary firmware version number used: 12.2.0, 12.1.1, (11.0.0.6)
  - [ ] Pre-compiled
  - [x] Self-compiled with TasmoCompiler
- [x] Flashing tools used: OTA (and even serial flashing)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
- not relevant -
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
- not relevant -
```
- [x] Provide the output of this command: `Status 0`:
```
- not relevant -
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```
- not relevant -
```

### TO REPRODUCE
Self compiled via [TasmoCompiler](https://github.com/benzino77/tasmocompiler) and these defines to use SML and google chart:
```
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifndef USE_SML_SCRIPT_CMD
#define USE_SML_SCRIPT_CMD
#endif
#ifndef USE_GOOGLE_CHARTS
#define USE_GOOGLE_CHARTS
#endif
#ifndef LARGE_ARRAYS
#define LARGE_ARRAYS
#endif
#ifndef USE_SCRIPT_WEB_DISPLAY
#define USE_SCRIPT_WEB_DISPLAY
#endif
#ifndef USE_EEPROM
#define USE_EEPROM
#endif
#ifdef EEP_SCRIPT_SIZE
#undef EEP_SCRIPT_SIZE
#endif
#define EEP_SCRIPT_SIZE 6200
```

Create a new script:
```
>D 40
hour=0
; 4h power chart
M:s4h=0 960

>S
hour=hours
;put a random number into array for testing every 15s
if (secs%15==0)
then
	s4h=rnd(1000)
endif

; set x axis index to 4 hours a 15 sec therefore /240, no decimal point
hour-=4
if (hour<0)
then
	cstr2="cnth"+s(1.0((24-4+hours)*240)+(mins*4))+"/240"
else
	cstr2="cnth"+s(1.0((hours-4)*240)+(mins*4))+"/240"
endif
hour=hours

>W
; 4h power chart
$<div id="chart1" style="text-align:center;width:400px;height:100%%;padding:0px"></div>
$gc(lt s4h "wr" "Leistung [W]" cstr2)
$var options = {
$chartArea:{left:60,width:'83%%'},
$width:'100%%',
$legend:'none',
$vAxis:{format:'# W',viewWindow:{min:-300}},
$explorer:{actions:['dragToZoom', 'rightClickToReset']},
$title:'Verbrauch 4 Stunden [Watt]'
$};
$gc(e)
```

### EXPECTED BEHAVIOUR
With Tasmota 11.0.0.6 (includes the new cnth) the 4h chart (last 4h) works perfect. See screenshots.

### SCREENSHOTS
Working 4h Google Chart with cstr2="cnth"+s(1.0((hours-4)*240)+(mins*4))+"/240". Full 4 hours and on the right side is the actual time (left = -4h). The 24h chart below has no issues with the x-axis labels. Correct time stamps.
[image]

With Tasmota 12.1 or 12.2 I got this here. On the right side is the actual time-4h. And only ~20 minutes on X-Axis (I know uptime is only 34mins but even after hours I got only 20 minutes on x-axis). And take a look at the 24h chart labels.
[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-24:
@gemu2015  Can you have a look and help?

gemu2015, 2022-11-24:
i am already in contact with @ottelo9 at another forum about this issue.
i think we can close this here

closed 2022-11-24 by Jason2866

MarcoLeoncino, 2023-03-19:
Hi everyone, I am using the following code, based on the example you gave here above.

`>D 
hour=0
temp=0
M:s4h=0 960
cstr2="cnth0/120"

>T
temp=DS18B20#Temperature

>S
hour=hours

if (secs%15==0)
then
s4h=temp
endif

hour-=4
if (hour<0)
then
cstr2="cnth"+s(1.0((24-4+hours)*240)+(mins*4))+"/240"
else
cstr2="cnth"+s(1.0((hours-4)*240)+(mins*4))+"/240"
endif
hour=hours

>W
$<div id="chart1" style="width:360px;height:360px;margin:-15px 0 0 -18px;"></div>
$gc(lt s4h "wr" "Temperature [°C]" cstr2)
$var options = {
$chartArea:{left:30,width:'90%%',right:10},
$width:'90%%',
$legend:'none',
$vAxis:{format:'# °C',viewWindow:{min:0},viewWindow:{max:80}},
$title:'Boiler temperature last 4 hours [°C]'
$};
$gc(e)
#`

and I get a well working google chart. 

I have tried to replace the hours to 12, the minutes to 720, but I cannot get a working 12 h plot. 
Can you give me an hint about how the above code shall be modified to have a 12 hours plot?
Thanks a lot!

gemu2015, 2023-03-19:
you can not keep 15 seconds resolution with 12 hours on an esp8266, so we switch to every minute

```
>D
hour=0
temp=0
M:s4h=0 720
cstr2="cnth0/120"

>T
temp=DS18B20#Temperature
>S
hour=hours
if (secs%60==0)
then
s4h=temp
endif

hour-=12
if (hour<0)
then
cstr2="cnth"+s(1.0((24-12+hours)*60)+mins)+"/60"
else
cstr2="cnth"+s(1.0((hours-12)*60)+mins)+"/60"
endif

>W
$<div id="chart1" style="text-align:center;width:400px;height:100%%;padding:0px"></div>
$gc(lt s4h "wr" "Temperature [°C]" cstr2)
$var options = {
$chartArea:{left:30,width:'90%%',right:10},
$width:'90%%',
$legend:'none',
$vAxis:{format:'# °C',viewWindow:{min:0},viewWindow:{max:80}},
$title:'Boiler temperature last 12 hours [°C]'
$};
$gc(e)
#
```

MarcoLeoncino, 2023-03-20:
Thanks! Every 60 seconds is perfect!