arendst/Tasmota #17039: Pzem 004 v3 tasmota12.2 sonoff basic v1 wont reset EnergyTotal0

FARKIr, 2022-11-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:36:59.762 CMD: Backlog Template
09:36:59.851 MQT: stat/tasmota_2EC9F8/RESULT = {"NAME":"PZEM 141","GPIO":[32,1472,0,1536,0,0,0,0,224,0,1184,0,0,0],"FLAG":0,"BASE":1}
09:39:58.242 CMD: Module
09:39:58.247 MQT: stat/tasmota_2EC9F8/RESULT = {"Module":{"0":"PZEM 141"}}
09:40:20.436 CMD: GPIO 255
09:40:20.446 MQT: stat/tasmota_2EC9F8/RESULT = {"GPIO0":{"32":"Tlačidlo1"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"0":"Žiaden"},"GPIO3":{"1536":"PZEM016 Rx"},"GPIO4":{"0":"Žiaden"},"GPIO5":{"0":"Žiaden"},"GPIO9":{"0":"Žiaden"},"GPIO10":{"0":"Žiaden"},"GPIO12":{"224":"Relé1"},"GPIO13":{"0":"Žiaden"},"GPIO14":{"1184":"DHT11"},"GPIO15":{"0":"Žiaden"},"GPIO16":{"0":"Žiaden"},"GPIO17":{"0":"Žiaden"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
09:44:24.912 CMD: Backlog Rule1
09:44:25.000 MQT: stat/tasmota_2EC9F8/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:44:33.257 CMD: Backlog Rule2
09:44:33.342 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:44:41.276 CMD: Backlog Rule3
09:44:41.367 MQT: stat/tasmota_2EC9F8/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
  09:46:46.078 CMD: Status 0
09:46:46.084 MQT: stat/tasmota_2EC9F8/STATUS = {"Status":{"Module":0,"DeviceName":"PZEM 141","FriendlyName":["PZEM 141"],"Topic":"tasmota_2EC9F8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:46:46.091 MQT: stat/tasmota_2EC9F8/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:05:11","StartupUTC":"2022-11-11T08:41:35","Sleep":50,"CfgHolder":4617,"BootCount":35,"BCResetTime":"2022-11-09T16:03:10","SaveCount":188,"SaveAddress":"F8000"}}
09:46:46.096 MQT: stat/tasmota_2EC9F8/STATUS2 = {"StatusFWR":{"Version":"12.2.0.4(tasmota)","BuildDateTime":"2022-11-09T11:26:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"351/699"}}
09:46:46.100 MQT: stat/tasmota_2EC9F8/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Internet",""],"TelePeriod":20,"Resolution":"558100C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00404080","00006000","0000C000","00000000"]}}
09:46:46.492 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000405","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
09:46:46.707 RSL: STATUS5 = {"StatusNET":{"Hostname":"pzem-01","IPAddress":"192.168.100.141","Gateway":"192.168.100.252","Subnetmask":"255.255.255.0","DNSServer1":"192.168.100.252","DNSServer2":"0.0.0.0","Mac":"2C:3A:E8:2E:C9:F8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:46:47.013 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.112","MqttPort":1883,"MqttClientMask":"PZEM 141","MqttClient":"PZEM 141","MqttUser":"mkvettpict","MqttCount":8,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:46:47.228 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-11-11T08:46:47","Local":"2022-11-11T09:46:47","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:53","Sunset":"17:14"}}
09:46:47.515 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:46:47.734 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-11-11T09:46:47","ENERGY":{"TotalStartTime":"2022-11-11T00:00:00","Total":15.319,"Yesterday":4.549,"Today":3.173,"Power":249,"ApparentPower":277,"ReactivePower":121,"Factor":0.90,"Frequency":50,"Voltage":235,"Current":1.18},"DHT11":{"Temperature":20.0,"Humidity":25.0,"DewPoint":-0.7},"TempUnit":"C"}}
09:46:48.013 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-11-11T09:46:47","Uptime":"0T00:05:12","UptimeSec":312,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":8,"POWER":"ON","Wifi":{"AP":1,"SSId":"Internet","BSSId":"08:40:F3:DB:C2:91","Channel":11,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:04"}}}
09:46:48.234 RSL: STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["40254db0","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025c434","40242e29","40243169","402431ee","402432c8","4022a880","4024da4c","40222210","4022a8c2","4022a990","4022a9d6","4024da4c","40291105","4022ab14","402896e9","4028e4fc","40291105","40201dd6","40291105","4022b5fc","4022b7ac","40235df3","40222210","40251b40","40251ae2","40251b40","401000e1","40222248","4024db85","40229cfe","40229e17"]}}
 

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
09:48:18.149 CMD: WebLog 4
09:48:18.154 MQT: stat/tasmota_2EC9F8/RESULT = {"WebLog":4}
09:48:18.401 DMP: 01 04 14 09 2F 04 86 00 00 09 96 00 00 3B DF 00 00 01 F4 00 5A 00 00 AA 15
09:48:18.403 NRG: EnergyTotal[0] 15.3270 kWh
09:48:19.212 CFG: Uložené do flash pamäte v F7, Počítaj 189, Bytov 4096
09:48:19.401 DMP: 01 04 14 09 2E 04 9E 00 00 09 C1 00 00 3B DF 00 00 01 F4 00 5A 00 00 D6 F9
09:48:19.403 NRG: EnergyTotal[0] 15.3270 kWh
09:48:20.401 DMP: 01 04 14 09 29 04 93 00 00 09 AB 00 00 3B DF 00 00 01 F4 00 5A 00 00 6E B4
09:48:20.403 NRG: EnergyTotal[0] 15.3270 kWh
09:48:20.413 DHT: Pin14 read 0F001A0029
09:48:21.401 DMP: 01 04 14 09 2B 04 83 00 00 09 88 00 00 3B DF 00 00 01 F4 00 5A 00 00 60 6A
09:48:21.403 NRG: EnergyTotal[0] 15.3270 kWh
09:48:22.401 DMP: 01 04 14 09 29 04 99 00 00 09 B8 00 00 3B DF 00 00 01 F4 00 5A 00 00 73 45
09:48:22.403 NRG: EnergyTotal[0] 15.3270 kWh
09:48:23.401 DMP: 01 04 14 09 29 04 99 00 00 09 B8 00 00 3B DF 00 00 01 F4 00 5A 00 00 73 45
09:48:23.404 NRG: EnergyTotal[0] 15.3270 kWh
09:48:24.401 DMP: 01 04 14 09 2C 04 9D 00 00 09 B4 00 00 3B DF 00 00 01 F4 00 5A 00 00 EC 84
09:48:24.403 NRG: EnergyTotal[0] 15.3270 kWh
09:48:24.412 DHT: Pin14 read 0F001A0029
09:48:25.401 DMP: 01 04 14 09 2E 04 A9 00 00 09 E1 00 00 3B E0 00 00 01 F4 00 5A 00 00 1A 56
09:48:25.403 NRG: EnergyTotal[0] 15.3280 kWh
09:48:26.387 WIF: Skúška spojenia...
09:48:26.401 DMP: 01 04 14 09 2D 04 8B 00 00 09 9E 00 00 3B E0 00 00 01 F4 00 5A 00 00 4D 87
09:48:26.403 NRG: EnergyTotal[0] 15.3280 kWh
09:48:27.402 DMP: 01 04 14 09 2F 04 98 00 00 09 B8 00 00 3B E0 00 00 01 F4 00 5A 00 00 BF AC
09:48:27.404 NRG: EnergyTotal[0] 15.3280 kWh
09:48:28.401 DMP: 01 04 14 09 2F 04 98 00 00 09 B8 00 00 3B E0 00 00 01 F4 00 5A 00 00 BF AC
09:48:28.403 NRG: EnergyTotal[0] 15.3280 kWh
09:48:28.412 DHT: Pin14 read 0F001A0029

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-11:
There is no description of the expected behaviour, nor what you did or command you used? We have no Crystal Balls.

FARKIr, 2022-11-11:
can i somehow see history of commands that i send ? 
i dont remember realy, i install a reinstall it more times, and have used more hardware ...

please what haw do wrong ?

fwebcc, 2022-11-15:
The mqtt option on the web page sets the limit of the number of digits for the password. Here, you enter the password to cancel the limit

Jason2866, 2022-11-23:
Closing since powering a connected PZEM 004 from a sonoff basic will never work reliable. The power supply of a sonoff basic is to weak for.

closed 2022-11-23 by Jason2866