arendst/Tasmota #17045: Firmware upload bug

lmagyar, 2022-11-12:
### PROBLEM DESCRIPTION
The `curl http://.../u2?fsz=...` works only if there were a `curl http://.../up` executed before it.

I've tested it on 10+ devices, multiple firmwares, original and custom built, win and linux curl, the only common thing: **if /up is not downloaded previously, /u2 never works, it /up is downloaded previously, /u2 always works**.

I don't know whether it is a bug, or security feature or by design, I just inform you.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff TH
- [x] Tasmota binary firmware version number used: 12.0.2, 12.1.0, 12.2.0
  - [x] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"4":"Sonoff TH"}}
{"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
{"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":121,"Free":390,"Rules":"ON Wifi#Disconnected DO RuleTimer1 900 ENDON ON Wifi#Connected DO RuleTimer1 0 ENDON ON Rules#Timer=1 DO Restart 99 ENDON"}}
{"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
{"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
{"Status":{"Module":4,"DeviceName":"Spare","FriendlyName":["Spare Switch"],"Topic":"spare","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:15:53","StartupUTC":"2022-11-12T10:37:18","Sleep":50,"CfgHolder":4617,"BootCount":56,"BCResetTime":"2021-01-04T17:18:50","SaveCount":186,"SaveAddress":"FA000"}}
{"StatusFWR":{"Version":"12.2.0(sensors)","BuildDateTime":"2022-10-17T19:32:37","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"393/699"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Z2",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008008","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
{"StatusMEM":{"ProgramSize":666,"Free":336,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F8A8587","0405A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","00000001"],"Drivers":"1,2,3,4,5,6,8,9,10,12,14,16,17,24,29,34,62","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98"}}
{"StatusNET":{"Hostname":"spare","IPAddress":"192.168.1.191","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"80.244.99.36","DNSServer2":"80.244.99.37","Mac":"A4:CF:12:D7:CB:00","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
{"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%12X","MqttClient":"DVES_A4CF12D7CB00","MqttUser":"Device","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
{"StatusTIM":{"UTC":"2022-11-12T10:53:11","Local":"2022-11-12T11:53:11","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:54","Sunset":"17:13"}}
{"StatusSNS":{"Time":"2022-11-12T11:53:11","SI7021":{"Temperature":13.5,"Humidity":90.9,"DewPoint":12.0},"TempUnit":"C"}}
{"StatusSTS":{"Time":"2022-11-12T11:53:11","Uptime":"0T00:15:53","UptimeSec":953,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Z2","BSSId":"34:2C:C4:1B:98:80","Channel":1,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
19:38:35.702 CMD: weblog 4
19:38:41.276 CFG: Saved to flash at FB, Count 219, Bytes 4096
19:38:41.278 UPL: File tasmota-minimal.bin.gz
19:38:41.298 UPL: File type 0
19:38:42.518 HTP: Upload done
19:38:44.606 WIF: Checking connection...
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

This is logged on a Sonoff TH with original/prebuilt 12.2.0 tasmota-minimal (it finished too quickly, clearly without uploading the full firmware).

And the html response was: **Upload Failed / Not enough space**

```
$ curl http://192.168.1.191/u2?fsz=497205 -u admin:xxxxxxxx -F u2=@./tasmota-sensors.bin.gz -v > upload_response_sensors.html
*   Trying 192.168.1.191:80...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Connected to 192.168.1.191 (192.168.1.191) port 80 (#0)
* Server auth using Basic with user 'admin'
> POST /u2?fsz=497205 HTTP/1.1
> Host: 192.168.1.191
> Authorization: Basic xxxxxxxxxxxxxxxxxxxxxxxxxx=
> User-Agent: curl/7.78.0
> Accept: */*
> Content-Length: 497417
> Content-Type: multipart/form-data; boundary=------------------------f9909c15d48bc7cf
>
} [65536 bytes data]
 79  485k    0     0   79  384k      0   179k  0:00:02  0:00:02 --:--:--  179k* We are completely uploaded and fine
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: text/html
< Server: Tasmota/12.2.0 (ESP8266EX)
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: -1
< Accept-Ranges: none
< Transfer-Encoding: chunked
< Connection: close
<
100  485k    0     0  100  485k      0   134k  0:00:03  0:00:03 --:--:--  134k{ [2689 bytes data]
100  488k    0  2952  100  485k    760   125k  0:00:03  0:00:03 --:--:--  126k
* Closing connection 0
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-11-12:
When you choose to mess around without having a documented API, you're on your own when it comes to making it work or not.

Probably neither a bug, feature or by design.

closed 2022-11-12 by Jason2866

barbudor, 2022-11-12:
Looks like a totally wrong curl command
This used to work at sometime but haven't used it for quite a while:
```
curl -F "data=@$2" -H "Authorization: Basic xxxxxxxxxxxxxxxxxxxxxx" "http://$1/u2"
``` 
Where $1 is the IP address and $2 is the path to the binary to upload
Replace  `xxxxxxxxxxxxxxxxxx` by the base64 encoding of `admin:password`

lmagyar, 2022-11-12:
@barbudor Thanks, though I have no problem using curl, works perfectly, Tasmota has issue handling it, Tasmota can't upload it into the proper position in flash and runs out of space, it requires to download http://xxx/up before it. Additionally I can crash any Tasmota with a simple GET and a slightly malformed url also, so there seems to be general issues handling web requests, and usually attackers use "nondocumented" API "features"...

sfromis, 2022-11-12:
Tasmota has no issue, as you are not using a documented API. And Tasmota has absolute no claim of even aiming for being resistive to attackers on the internal network.

KizzyCode, 2023-04-23:
@sfromis While you're technically correct, this is quite a bit confusing. A simple GET request should not modify internal state; so while this is not a public API, it's also not really HTTP-like behavior. I can fully understand if you advocate the position: "Undocumented API → undefined behavior", but it's pretty confusing if you are used to HTTP semantics 😅

However I can fully understand if microcontroller-constraints and necessary simplifications lead to unusual behavior. Is there an official or better way to upload new firmware without using this internal API?

Jason2866, 2023-04-23:
@KizzyCode As described in Tasmota documentation  https://tasmota.com

sfromis, 2023-04-23:
What I'm doing when automating the upgrade of Tasmota firmware is to have the new images on an internal web server, and then use the regular `OtaUrl` + `Upgrade` commands to initiate the update, Tasmota will then fetch from that server, including the automatic logic to use the extra step with minimal when relevant.

In the real world, I'd only have an expectation of strict adherence to GET never modifying anything if using a documented RESTful API, not just with any HTTP request. Sure, I know the RFC text, but as HTTP is mainly a transfer protocol, and ....

KizzyCode, 2023-04-23:
@Jason2866 I was unable to find a way to _upload_ the firmware file using the official API – everything seems to revolve around the OTA-URL, which always requires a web server.

@sfromis I also thought about this, but since I'm using an isolated WiFi, in my case it's not trivial to run an additional web server. However I'm aware that this is probably an exotic setup I'm also fine if I'll have to resort to the unofficial API calls.

KizzyCode, 2023-04-24:
> In the real world, I'd only have an expectation of strict adherence to GET never modifying anything if using a documented RESTful API, not just with any HTTP request. Sure, I know the RFC text, but as HTTP is mainly a transfer protocol, and ....

This is not only a problem with REST-semantics; basically everything assumes that GET-requests do not change state. Browsers do silent preloading, apps use GET to fetch resources to create previews or similar, etc. – all this crucially depends on the fact that GET requests do not modify state.

I don't know what `/up` does in background – in this case it's probably not a problem; but in general it's not a good idea to violate this rule.

sfromis, 2023-04-24:
As HTTP is very secondary thing to Tasmota (the "M" in the name name means MQTT), no attempt has been made to make the web server handling optimized or stand up to any scrutiny about being robust or standards-compliant. Thus semantic discussions about "should" or "the right way" are just well outside of the rudimentary things Tasmota implements via HTTP, including GET being used to make a request from the browser, irrespective of what the request is meant to do.

KizzyCode, 2023-04-24:
Fair point. I only use the HTTP part since it's much easier to integrate it into my environment; but I get your point.