arendst/Tasmota #17052: 12.2.0: Mainmenu: "Tuya MCU" and "Gosund SP1 v23" no values displayed

rolsch, 2022-11-13:
### PROBLEM DESCRIPTION
After go to 12.2.0 at the mainmenu no values for "Tuya MCU" and "Gosund SP1 v23" are seen.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:41:08.299 MQT: stat/tasmota_152BD8/STATUS = {"Status":{"Module":55,"DeviceName":"Tasmota_Stall","FriendlyName":["Tasmota_Stall"],"Topic":"tasmota_152BD8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:41:08.332 MQT: stat/tasmota_152BD8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-lite.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:43","StartupUTC":"2022-11-13T10:37:25","Sleep":50,"CfgHolder":4617,"BootCount":59,"BCResetTime":"2021-01-04T19:53:31","SaveCount":3731,"SaveAddress":"F8000"}}
11:41:08.369 MQT: stat/tasmota_152BD8/STATUS2 = {"StatusFWR":{"Version":"12.2.0(lite)","BuildDateTime":"2022-10-17T08:41:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"420/699"}}
11:41:08.387 MQT: stat/tasmota_152BD8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["namoro_de_01","namoro_de_bs_rt"],"TelePeriod":15,"Resolution":"568380C0","SetOption":["00208009","2805C80001000600003C5A00002800000000","00000088","00006000","00004000","00000000"]}}
11:41:08.420 MQT: stat/tasmota_152BD8/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F000782","04108001","00000002","00000000","00000000","00000000","00000000","04000000","00000000"],"Drivers":"1,2,4,9,10,12,16,20,21,62","Sensors":""}}
*****
*****
11:41:08.500 MQT: stat/tasmota_152BD8/STATUS7 = {"StatusTIM":{"UTC":"2022-11-13T10:41:08","Local":"2022-11-13T11:41:08","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:56","Sunset":"17:12"}}
11:41:08.518 MQT: stat/tasmota_152BD8/STATUS10 = {"StatusSNS":{"Time":"2022-11-13T11:41:08"}}
11:41:08.529 MQT: stat/tasmota_152BD8/STATUS11 = {"StatusSTS":{"Time":"2022-11-13T11:41:08","Uptime":"0T00:03:43","UptimeSec":223,"Vcc":3.462,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":2,"SSId":"**********","BSSId":"******","Channel":13,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:17"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

rolsch, 2022-11-13:
12.2.0
[image]

12.1.1
[image]

arendst, 2022-11-13:
See #17002.

closed 2022-11-13 by arendst