arendst/Tasmota #17053: device reboot of ESP32 will set EQ3 BLE Themostat to 30 degree ("on")

michapr, 2022-11-13:
### PROBLEM DESCRIPTION
Using ESP32 with BLE option (sensors) and EQ3 thermostat.
After some time (once per day about) the ESP reboot (software reset) and after start the thermostat is set to 30 degree (max)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32
- [x] Tasmota binary firmware version number used: 12.2.0(bluetooth)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:35:01.850 CMD: Backlog Template; Module; GPIO 255
13:35:01.898 MQT: stat/tasmota_190F70/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:35:02.132 MQT: stat/tasmota_190F70/RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:35:02.388 MQT: stat/tasmota_190F70/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:32:43.130 CMD: Status 0
13:32:43.151 MQT: stat/tasmota_190F70/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_190F70","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:32:43.197 MQT: stat/tasmota_190F70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T03:12:18","StartupUTC":"2022-11-13T09:20:25","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-11-03T16:56:11","SaveCount":21}}
13:32:43.243 MQT: stat/tasmota_190F70/STATUS2 = {"StatusFWR":{"Version":"12.2.0(bluetooth)","BuildDateTime":"2022-10-17T08:39:34","Core":"2_0_5","SDK":"v4.4.3.4","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"450/699"}}
13:32:43.270 MQT: stat/tasmota_190F70/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":1,"SysLog":1,"LogHost":"192.168.178.200","LogPort":514,"SSId":["FRITZ!Box Fon WLAN 7170",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004002","00000000"]}}
13:32:43.313 MQT: stat/tasmota_190F70/STATUS4 = {"StatusMEM":{"ProgramSize":1584,"Free":1295,"Heap":69,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7C7","00148001","000000CF","010013C0","C0000981","00004080","00200000","5400082C","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,85","Sensors":"1,2,3,5,6,52,62,127"}}
13:32:43.365 MQT: stat/tasmota_190F70/STATUS5 = {"StatusNET":{"Hostname":"tasmota-190F70-3952","IPAddress":"192.168.178.107","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"78:E3:6D:19:0F:70","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:32:43.402 MQT: stat/tasmota_190F70/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.92","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_190F70","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:32:43.436 MQT: stat/tasmota_190F70/STATUS7 = {"StatusTIM":{"UTC":"2022-11-13T12:32:43","Local":"2022-11-13T13:32:43","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:56","Sunset":"17:12"}}
13:32:43.465 MQT: stat/tasmota_190F70/STATUS10 = {"StatusSNS":{"Time":"2022-11-13T13:32:43"}}
13:32:43.483 MQT: stat/tasmota_190F70/STATUS11 = {"StatusSTS":{"Time":"2022-11-13T13:32:43","Uptime":"0T03:12:18","UptimeSec":11538,"Heap":71,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7170","BSSId":"DC:15:C8:B7:83:7D","Channel":1,"Mode":"11n","RSSI":100,"Signal":-13,"LinkCount":1,"Downtime":"0T00:00:04"}}}
13:32:43.533 MQT: stat/tasmota_190F70/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"IllegalInstruction","EPC":"40092207","EXCVADDR":"00000000","CallChain":["40092204","40087ce2","40019f36","4001a1f2","4001a282","4019e17d","400469b6","40019d11","40055b4d","40194663","40194cd4"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
no way...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
no reboot - and if reboot, then do not send cmd "settemp" to EQ3 device

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

`{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T03:12:18","StartupUTC":"2022-11-13T09:20:25","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-11-03T16:56:11","SaveCount":21}}`

`{"cmd":"settemp","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","RSSI":-72,"stattime":1668331245,"temp":30.0,"posn":0,"mode":"manual","hassmode":"idle","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":21.0,"night":17.0,"offset":1.0}
`

**(Please, remember to close the issue when the problem has been addressed)**

Ysbrand, 2022-11-13:
HI,

Some questions:
1) Is this the 1st Tasmota release you have been using with the EQ3 TRV that exhibits the issue, or have you seen the same behavior with previous releases?
2) Is the issue only with a single TRV or do you see the same with (possible) others?
3) Can you try to use TAS 12.2.0.1 or higher, as this version is using a new version of ESP32 NimBLE (1.4.1) as the previous version had some issues which also introduced issues with the EQ3 TRV's (not the same issues as you are seeing, but migth be related in a way).
4) What is the firmware version of the EQ3 TRV?

(I'm using 12.2.0.2 with a total of 9 TRV's and I have not seen the issue you are describing, And yes, on a regular basis ESP32 reboots)

michapr, 2022-11-14:
1. - yes, this is the first and only release I have tried. Have used before HomeAssistant with direct connection but because of the "bleak issue" the control is not stable more and I moved to Tasmota32
2.  it is a single TRV, all others are Z-Wave based. Was looking for alternate TRV for places near the hub...
3. have updated to 12.2.0.4 - same issue, see log below
4. firmware is 1.20  (number displayed after starting TRV)

I can reproduce the issue rebooting the ESP32 via Web-UI
11:58:41
`{"Info3":{"RestartReason":"Software reset CPU","BootCount":13}}`

11:58:47
`{"cmd":"settemp","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","stattime":1668423527,"temp":19.0,"posn":0,"mode":"auto","hassmode":"auto","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}`
11:58.50
`{"cmd":"manual","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","RSSI":-76,"stattime":1668423530,"temp":19.0,"posn":0,"mode":"manual","hassmode":"idle","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}`
11:58:53
`{"cmd":"settemp","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","RSSI":-76,"stattime":1668423532,"temp":30.0,"posn":0,"mode":"manual","hassmode":"idle","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}`
11:59:00
`{"cmd":"poll","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","RSSI":-76,"stattime":1668423539,"temp":30.0,"posn":0,"mode":"manual","hassmode":"idle","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}`

fwebcc, 2022-11-15:
The mqtt option on the web page sets the limit of the number of digits for the password. Here, you enter the password to cancel the limit

Ysbrand, 2022-11-15:
Not sure what is happening, your log is showing a "manual" and a "settemp" command send to the trv, where are these commands coming from? 
I've restarted my device multiple times, even with an exception (28) and fter initializing my variables it goes straight into polling mode, no settemp or manual commands whatsoever.
Where are these commands coming from? Do you have something in the autoexec.bat or is mqtt sending these commands?

michapr, 2022-11-15:
@fwebcc  - not sure what you want to say me about the password digits - related to this issue... ;)

@Ysbrand I have an autoecex.bat - there is the only option `MI32Option6 1` inside.
I already have renamed the command topics in Home Assistant configuration.yaml to be sure that the commands are not coming from there... but no changes. I thought that because of the JSON structure (and content) it is coming from tasmota - or not?

But will check it more today in the evening - if you say that it isn't coming from tasmota script there must be another source for it... Thanks for this hint. Will tell you later.

michapr, 2022-11-15:
additional info:

I have these commands only in MQTT `topic stat/EQ3/001A2208BFD6`
In `cmnd/tasmota_190F70/EQ3/001A2208BFD6/settemp`  I do not see the settemp to 30 degree, So I do not think that the TRV can be set via MQTT, right? Or is there other way I could check it?

Ysbrand, 2022-11-15:
Yes, you can definitely set the temperature with settemp through mqtt:
`/usr/bin/mosquitto_pub  -h 192.168.0.199 -p 1883 -t cmnd/eq3_devices/EQ3/001A2216A476/settemp -m "20"`

michapr, 2022-11-15:
> Yes, you can definitely set the temperature with settemp through mqtt:
/usr/bin/mosquitto_pub -h 192.168.0.199 -p 1883 -t cmnd/eq3_devices/EQ3/001A2216A476/settemp -m "20"

I know - but in this case I would see it in MQTT log/trace. But there is nothing... After I restart the ESP32 the temperature will go to 30degree but no command is in trace - only in the topic "stat/..." as described.

all debug level already set to 4 - result after reboot:

```
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 mqtt: cmnd/tasmota_190F70/EQ3/001A2208BFD6/settemp:19.0
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3: mqtt->cmdstr 001A2208BFD6 settemp 19.0
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 cmd index: 1
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6: Op queued len now 1
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6:Op dequeued len now 0
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 queued
Nov 15 16:45:56 tasmota-190F70-3952 ESP-MQT: stat/tasmota_190F70/EQ3/001A2208BFD6 = {"result":"queued"}
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 mqtt: cmnd/tasmota_190F70/EQ3/001A2208BFD6/mode:heat
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3: mqtt->cmdstr 001A2208BFD6 mode heat
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 cmd index: 1
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6: Op queued len now 1
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6: Op queued len now 2
Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 queued
Nov 15 16:45:56 tasmota-190F70-3952 ESP-MQT: stat/tasmota_190F70/EQ3/001A2208BFD6 = {"result":"queued"}
Nov 15 16:45:59 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6: trv operation failed - retrying -11
Nov 15 16:46:02 tasmota-190F70-3952 ESP-MQT: stat/EQ3/001A2208BFD6 = {"cmd":"settemp","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","stattime":1668527160,"temp":19.0,"posn":100,"mode":"manual","hassmode":"heat","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}
Nov 15 16:46:02 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6:Op dequeued len now 1
Nov 15 16:46:08 tasmota-190F70-3952 ESP-MQT: stat/EQ3/001A2208BFD6 = {"cmd":"manual","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","stattime":1668527166,"temp":19.0,"posn":100,"mode":"manual","hassmode":"heat","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}
Nov 15 16:46:08 tasmota-190F70-3952 ESP-EQ3 001A2208BFD6:Op dequeued len now 0
Nov 15 16:46:12 tasmota-190F70-3952 ESP-EQ3 poll cycle starting
Nov 15 16:46:13 tasmota-190F70-3952 ESP-MQT: stat/EQ3/001A2208BFD6 = {"cmd":"settemp","result":"ok","MAC":"001A2208BFD6","tas":"tasmota-190F70-3952","stattime":1668527171,"temp":30.0,"posn":0,"mode":"manual","hassmode":"idle","boost":"inactive","dst":"set","window":"closed","state":"unlocked","battery":"GOOD","holidayend":"00-00-00 00:00","windowtemp":12.0,"windowdur":15,"day":19.5,"night":17.0,"offset":1.0}
```
But from where it can come?

Ysbrand, 2022-11-15:
Here is your problem:
`Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 mqtt: cmnd/tasmota_190F70/EQ3/001A2208BFD6/mode:heat`
This will put the trv to 30C

F

michapr, 2022-11-15:
@Ysbrand 

> Here is your problem:
> `Nov 15 16:45:56 tasmota-190F70-3952 ESP-EQ3 mqtt: cmnd/tasmota_190F70/EQ3/001A2208BFD6/mode:heat`
> This will put the trv to 30C

Sure. But it is not clear from where this is coming. Will search more....

Jason2866, 2022-11-24:
@michapr Any news? Found the source of the cmd for heating?

michapr, 2022-11-24:
> @michapr Any news? Found the source of the cmd for heating?

No,... not in this time. Already checked different things, but.... Have solved it in this time in HA with script checking the state.
When found will update the issue here.

Jason2866, 2022-11-30:
Closing. @michapr Please reopen if it is a Tasmota bug.

closed 2022-11-30 by Jason2866