arendst/Tasmota #17067: ModuleNotFoundError: No module named 'zopfli'

joba-1, 2022-11-14:
### PROBLEM DESCRIPTION
build of tasmota-rangeextender (from platformio_tasmota_cenv_sample.ini) fails

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): none, build problem
- [ ] Tasmota binary firmware version number used: none yet, build problem (latest development)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: n/a
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
git clone https://github.com/arendst/Tasmota
cd Tasmota
cp platformio_tasmota_cenv_sample.ini platformio_tasmota_cenv.ini
pio run -t build -e tasmota-rangeextender

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
builds fine

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
```
Processing tasmota-rangeextender (platform: https://github.com/tasmota/platform-espressif8266/releases/download/v2.7.4/platform-espressif8266-2.7.4.zip; framework: arduino; board: esp8266_1M)
--------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8266_1M.html
PLATFORM: Espressif 8266 (2.7.4+20221113) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS
HARDWARE: ESP8266 80MHz, 80KB RAM, 972.00KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.4+9 
 - tool-esptoolpy @ 1.40301.221017 (4.3.1) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
Converting tasmota.ino
WARNING: You are using pip version 20.0.2; however, version 21.3.1 is available.
You should consider upgrading via the '/home/joachim/.platformio/penv/bin/python -m pip install --upgrade pip' command.
Installing Python dependencies
ERROR: Could not find a version that satisfies the requirement zopfli>=0.2.1 (from versions: 0.0.0, 0.0.1, 0.0.2, 0.0.3, 0.0.4, 0.1.0.post1, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.6, 0.1.7, 0.1.8, 0.1.9b1, 0.1.9)
ERROR: No matching distribution found for zopfli>=0.2.1
WARNING: You are using pip version 20.0.2; however, version 21.3.1 is available.
You should consider upgrading via the '/home/joachim/.platformio/penv/bin/python -m pip install --upgrade pip' command.
*** Error 1

```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-14:
Nice one. Which Operating System are you using?

joba-1, 2022-11-14:
openSUSE Leap 15.3

Jason2866, 2022-11-14:
Can you try `pip install zopfli `

Jason2866, 2022-11-14:
The python version and the packages are kinda outdated. Can you upgrade to python v3.9 ?

joba-1, 2022-11-14:
I'd like to keep the default system python version (what zypper up provides) because distro packages might depend on it, but as I said, with conda I can use any version. Trying...

joba-1, 2022-11-14:
System python and conda python are both not used. PlatformIO uses its own venv. Forgot about that. 
And it seems I'm on par with pio versions:

```
tasmota-issue-17067) pio upgrade 
You're up-to-date!
PlatformIO 6.1.5 is currently the newest version available.

```

Jason2866, 2022-11-14:
You may run (in near future) in general problems with python v3.6 and platformio.
A standard Platformio Installation installs a pyenv for Platformio which is independent from the used in the system and installs everything needed in this pyenv.
The way `zopfli` is installed like platformio installs all its pypi needed packages.

Jason2866, 2022-11-14:
Can you delete the whole folder `.platformio` ?  The pyenv will never be updated.
Only new installed / generated when deleted.

joba-1, 2022-11-14:
I understand platformio uses its own python environment (clever move), but is the environment still based on system python instead of also providing their own? Can I somehow convince it to use a conda environment provided python as basis?
I guess I could do pip install platformio in a conda environment, but then: how do I convince ms code to use that in its pio extension? Maybe worth a try to start that in the conda env as well.

joba-1, 2022-11-14:
I'll try delete .platformio

Jason2866, 2022-11-14:
Yes, the pyenv is generated from the used in system. BUT you can change in Platformio settings to use a version you specify (which is on your system).
This settings can be found in the Plugins section of VSC -> Platformio

joba-1, 2022-11-14:
I'll dig for that if needed.

`pio run -t upload -e tasmota-rangeextender` while in conda env with python 3.9 worked :)
 
As an aside: Do you know the pio target for compile only, no upload? tried compile and build, didn't work.

Currently checking inside vs code...

Jason2866, 2022-11-14:
`pio run -e tasmota-rangextender`

joba-1, 2022-11-14:
lol - too easy

Jason2866, 2022-11-14:
BTW. There is no way out for using at least Python v3.7 since a actual `zopfli` requires Python 3.7 or greater.

joba-1, 2022-11-14:
You mean this vs code pio extension setting?
"platformio-ide.useBuiltinPython": false
I'll try...

joba-1, 2022-11-14:
platformio-ide.useBuiltinPython": false didn't work, same outcome in vs code

```
WARNING: You are using pip version 20.0.2; however, version 21.3.1 is available.
You should consider upgrading via the '/home/joachim/.platformio/penv/bin/python -m pip install --upgrade pip' command.
Installing Python dependencies
ERROR: Could not find a version that satisfies the requirement zopfli>=0.2.1 (from versions: 0.0.0, 0.0.1, 0.0.2, 0.0.3, 0.0.4, 0.1.0.post1, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.6, 0.1.7, 0.1.8, 0.1.9b1, 0.1.9)
ERROR: No matching distribution found for zopfli>=0.2.1

```
I'll check what happens if I start vs code within the conda environment...

Jason2866, 2022-11-14:
Mhh, i have used the setting `platformio-ide.useBuiltinPython": false` and it did work.
Have you provided the path to the version to use? It is to set "somewhere" in this pio extension. It is a while since i used (was on my Windows PC). Not using Windows anymore...

joba-1, 2022-11-14:
removing ~/.platformio again, conda activate {env-with-py3.9}, pio run... works as before.
Now start code from commandline (e.g. with the conda env activated) it also builds:

```
...
============================================================= [SUCCESS] Took 39.57 seconds =============================================================

Environment            Status    Duration
---------------------  --------  ------------
tasmota-rangeextender  SUCCESS   00:00:39.565
============================================================== 1 succeeded in 00:00:39.565 ==============================================================

Terminal will be reused by tasks, press any key to close it.

```

good enough for me. Thank you!

btw. actual == tatsächlich, current == aktuell (hope you don't mind the hint)

joba-1, 2022-11-14:
final test done:
reverted the setting platformio-ide.useBuiltinPython": false, removed .platformio, started code in conda env -> also works.
Now start code outside conda environment gives the error again.
So the key is always start code with python>3.7 in current environment (by using conda in my case)

Jason2866, 2022-11-14:
So i think iissue can be closed. Solution is to use Python >= v3.7

closed 2022-11-14 by Jason2866

sfromis, 2022-11-15:
Did you try building a new docker image? It is plausible that dependencies need to be evaluated/fulfilled again.

barbudor, 2022-11-16:
Yes I 'rmi' the image and rebuilt it from Dockerfile

blakadder, 2022-11-16:
added `pip install zopfli` to the container, seems its working now

barbudor, 2022-11-16:
Thanks a lot @blakadder 
I didn't had time to investigate further
I think Jason's script tries to do some but fails because of those write access errors

blakadder, 2022-11-16:
I notced that too but I just did the Occam's razor solution