arendst/Tasmota #17093: BL0937 power usage is always positive

SerialVelocity, 2022-11-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using tasmota, the power readings are always positive even when electricity is being fed in the opposite direction.

It looks like it was fixed with #15265 but then reverted due to #15374 in commits https://github.com/arendst/Tasmota/commit/f85e0dd8059e8ea08a7efd7381d196859029ea8b and https://github.com/arendst/Tasmota/commit/772e2bddae38a193c3e4561eda191a4eb4efb67a.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Athom PG04-UK16A
- [X] Tasmota binary firmware version number used: 12.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:24:16.741 CMD: Backlog Template; Module; GPIO 255
22:24:16.766 MQT: stat/ben/garage/inverter/RESULT = {"NAME":"Athom PG04-UK16A","GPIO":[0,0,0,32,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":18}
22:24:16.994 MQT: stat/ben/garage/inverter/RESULT = {"Module":{"0":"Athom PG04-UK16A"}}
22:24:17.202 MQT: stat/ben/garage/inverter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:25:10.785 CMD: Status 0
22:25:10.792 MQT: stat/ben/garage/inverter/STATUS = {"Status":{"Module":0,"DeviceName":"SmartPlug_GarageInverter","FriendlyName":["Garage Inverter"],"Topic":"inverter","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:25:10.798 MQT: stat/ben/garage/inverter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:15:53","StartupUTC":"2022-11-15T19:09:17","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2022-11-15T19:21:09","SaveCount":54,"SaveAddress":"F6000"}}
22:25:10.804 MQT: stat/ben/garage/inverter/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-17T08:39:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"449/699"}}
22:25:10.809 MQT: stat/ben/garage/inverter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ComeFlyWithMe-Devices",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:25:10.822 MQT: stat/ben/garage/inverter/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
22:25:10.828 MQT: stat/ben/garage/inverter/STATUS5 = {"StatusNET":{"Hostname":"inverter-4441","IPAddress":"192.168.18.187","Gateway":"192.168.16.1","Subnetmask":"255.255.252.0","DNSServer1":"192.168.16.1","DNSServer2":"0.0.0.0","Mac":"4C:EB:D6:10:11:59","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:25:10.832 MQT: stat/ben/garage/inverter/STATUS6 = {"StatusMQT":{"MqttHost":"vernemq.household.svc.chillichef.internal","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_101159","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:25:10.839 MQT: stat/ben/garage/inverter/STATUS7 = {"StatusTIM":{"UTC":"2022-11-15T21:25:10","Local":"2022-11-15T22:25:10","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"07:59","Sunset":"17:09"}}
22:25:10.842 MQT: stat/ben/garage/inverter/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:25:10.853 MQT: stat/ben/garage/inverter/STATUS10 = {"StatusSNS":{"Time":"2022-11-15T22:25:10","ENERGY":{"TotalStartTime":"2022-11-15T19:21:09","Total":0.352,"Yesterday":0.000,"Today":0.352,"Power":401,"ApparentPower":408,"ReactivePower":77,"Factor":0.98,"Voltage":253,"Current":1.616}}}
22:25:10.861 MQT: stat/ben/garage/inverter/STATUS11 = {"StatusSTS":{"Time":"2022-11-15T22:25:10","Uptime":"0T02:15:53","UptimeSec":8153,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"ComeFlyWithMe-Devices","BSSId":"3C:84:6A:F5:9A:9A","Channel":6,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":2,"Downtime":"0T00:00:08"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
22:26:40.201 HLW: EPC 376, CFlen 2658 usec
22:26:41.201 HLW: EPC 375, CFlen 2676 usec
22:26:42.201 HLW: EPC 375, CFlen 2672 usec
22:26:43.201 HLW: EPC 376, CFlen 2636 usec
22:26:44.201 HLW: EPC 376, CFlen 2636 usec
22:26:45.201 HLW: EPC 376, CFlen 2627 usec
22:26:46.201 HLW: EPC 378, CFlen 2666 usec
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1- Plug an inverter into the socket
2- See positive values

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
1- Plug an inverter into the socket
2- See negative values

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-11-16:
Let me think of an easy way to let users configure for negative values....

arendst, 2022-11-16:
The BL0937 uses a different driver than you suggest. The commits mentionends were for the BL09xx driver. You are using the HLW8012 driver which was never able to detect negative values.

closed 2022-11-16 by arendst