arendst/Tasmota #17106: Can not set modbusbaud to 115200

mrRobot62, 2022-11-16:
### PROBLEM DESCRIPTION
If I set `modbusbaudrate 115200` response is always `stat/modbus/RESULT = {"ModbusBaudrate":38400}`

If i restart my esp32, default baud rate is set to 9600

Below log output from my tasmota device:
```
22:54:47.607 CMD: modbusbaudrate 115200
22:54:47.637 MQT: stat/modbus/RESULT = {"ModbusBaudrate":38400}
22:56:15.861 MQT: tele/modbus/STATE = {"Time":"2022-11-16T22:56:15","Uptime":"0T00:15:10","UptimeSec":910,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"TP-LINK_7692","BSSId":"70:4F:57:45:76:92","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:59:01.141 CMD: modbusbaudrate 57600
22:59:01.166 MQT: stat/modbus/RESULT = {"ModbusBaudrate":57600}
22:59:33.384 CMD: modbusbaudrate 115200
22:59:33.410 MQT: stat/modbus/RESULT = {"ModbusBaudrate":38400}
```
I tried this with several other Hausrates, seems 57600 is max. Unfortunately my EPEVER XTRA 4415N is set to 115200

Any idea what I can do?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): esp32-wroom_____
- [ ] Tasmota binary firmware version number used: tasmota32.bin (latest)_____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: chrome online flashing tool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-11-16:
@jeroenSt
There's a problem with `Settings->modbus_sbaudrate`. As a `uint8_t` it's not working for baudrate above 76500.
The easiest would be to use a uint16_t but that would mean a breaking change to move the value else where.
Or split in hi and low and split/combine as needed until we can refactor the Settings

Or is there a way to add an upgrade rule @arendst?

https://github.com/arendst/Tasmota/blob/c304319f5d44714edc41d49d1adafc7aa2fdd9cf/tasmota/include/tasmota_types.h#L833

https://github.com/arendst/Tasmota/blob/c304319f5d44714edc41d49d1adafc7aa2fdd9cf/tasmota/tasmota_xdrv_driver/xdrv_63_modbus_bridge.ino#L193-L198

barbudor, 2022-11-16:
@mrRobot62 Can you please advise what devices you are using that is needing 115200 ?
That doesn't seem much common to have such bit rate in Modbus

mrRobot62, 2022-11-17:
@barbudor thanks for your response and your explanation.

Out from a doc "B-Series MODBUS Specification EPEVER Corporation, 2.3" I read the information
```
The B-Series controller supports RTU mode only. 16bit MODBUS addresses (per the modbus.org spec) The serial communication parameters are
BPS:115200 baud 
Parity: None
Data bits: 8
Stop bits:1
Flow control: None
All addresses listed are for the request PDU.
```
My Model: EPEVER XTRA 4415N

I asked EPVER to send me a specification about their RS485 interface and what registers... are used, unfortunately the sup
port was not very helpful. They didn't send me anything. Searching via Google I found attached spec.
[EPEVER_ControllerProtocolV2.3.pdf](https://github.com/arendst/Tasmota/files/10028241/EPEVER_ControllerProtocolV2.3.pdf)

Hope my understanding this spec is ok :-/

barbudor, 2022-11-17:
Unless @jeroenst can provide a fox quickly I can look at it this weekend as I am too much busy with real life work

If you can't wait and are able to compile a change yourself, you can hardcode the value 115200 in xdrv_63_modbus_bridge.ino line 198
```
int result = modbusBridgeModbus->Begin(115200, ConvertSerialConfig(Settings->modbus_sconfig)); 
```
That should do the trick

arendst, 2022-11-17:
I'll change it to uint16_t like other baudrates now. Hold on.

mrRobot62, 2022-11-17:
@barbudor  & @arendst : wow - that's awesome from you !

Thanks a lot !!!!

@arendst : I saw your commit - sorry for this stupid question, were can I download the bin file with this bugfix - uhhhh blame on me ;-)

joba-1, 2022-11-17:
Now, you made me smile :) Where did you get the version with the problem? Birthday present on an USB stick?

But then, I didn't find it by the docs, too. My first try github -> releases -> release notes only talks about URLs for releases. You would have to know for development versions you have to delete the /release part. Or find it elswhere in the extensive tasmota docs.

TL;DR: http://ota.tasmota.com/tasmota/  Have fun!

arendst, 2022-11-17:
Or just read the README at https://github.com/arendst/Tasmota#readme and find the link below the heading Development....

mrRobot62, 2022-11-17:
short update (and than I move to the discussion group), because maybe it's an mistake on my side or something else, so my hope is on the discussion board it's a better place for this issue

1. flashed my esp32 with version 12.2.0.5 - yeah
2. with `modbusbaudrate 115200` it's possible to change to 115200
```
20:00:54.564 CMD: modbusbaudrate
20:00:54.579 MQT: stat/solar/RESULT = {"ModbusBaudrate":115200}
```
3. with `modbusserialconfig 702`I set "8N1"
```
20:10:10.994 CMD: modbusserialconfig 702
20:10:11.005 MQT: stat/solar/RESULT = {"ModbusSerialConfig":"8N1"}
```
4. with `modbussend...`I tried to read EPEVER register on 0x3100
```
20:12:19.651 CMD: modbussend {"deviceAddress":1, "functionCode":2, "startAddress":12544, "type":"uint8","count":4}
20:12:19.663 MQT: stat/solar/RESULT = {"ModbusSend":"Done"}
```
nothing more :-/

mrRobot62, 2022-11-17:
See - discussion - problem solved - problem was 30cm in front of the screen - omg - 
See [Discussion](https://github.com/arendst/Tasmota/discussions/17123)

closed 2022-11-17 by mrRobot62