arendst/Tasmota #17107: Shelly EM Calibration, CurrentSet and PowerSet doesn´t working

Starfoxfs, 2022-11-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have bought a new Shelly EM with a 50Amps Clamp on P1+ and P1-.
Then I have fresh flashed Tasmota 12.2.0 from http://ota.tasmota.com/tasmota/release/ 
The Tasmota.bin.gz

The Template I set is from Blakadder for the Shelly EM:
`{"NAME":"Shelly EM","GPIO":[0,0,0,0,0,0,0,0,640,3457,608,224,8832,1],"FLAG":0,"BASE":18}`

Now I want to Calibrate it with a 60W Lamp.
VoltageSet works fine and will be verified in Console and shown in Mainmenu
CurrentSet also will be verified in Console but there are no changes in Mainmenu
PowerSet the same as CurrentSet

So my Problem is i can´t calibrate this thing, what do i wrong ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Shelly EM
- [x ] Tasmota binary firmware version number used: on the Release Site core 2.7.4.9 Version 12.2.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

09:42:13.423 MQT: stat/tasmota_7456F9/STATUS = {"Status":{"Module":0,"DeviceName":"ShellyEM","FriendlyName":["ShellyEM"],"Topic":"tasmota_7456F9","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:42:13.429 MQT: stat/tasmota_7456F9/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:10:37","StartupUTC":"2022-11-17T08:31:36","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-11-16T14:02:20","SaveCount":26,"SaveAddress":"FA000"}}
09:42:13.433 MQT: stat/tasmota_7456F9/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-17T08:39:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"357/699"}}
09:42:13.438 MQT: stat/tasmota_7456F9/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WlanWii",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
09:42:13.449 MQT: stat/tasmota_7456F9/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1620C2","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
09:42:13.457 MQT: stat/tasmota_7456F9/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7456F9-5881","IPAddress":"xxx","Gateway":"xxx","Subnetmask":"xxx","DNSServer1":"xxx","DNSServer2":"0.0.0.0","Mac":"xxx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:42:13.463 MQT: stat/tasmota_7456F9/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7456F9","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:42:13.469 MQT: stat/tasmota_7456F9/STATUS7 = {"StatusTIM":{"UTC":"2022-11-17T08:42:13","Local":"2022-11-17T09:42:13","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:02","Sunset":"17:07"}}
09:42:13.473 MQT: stat/tasmota_7456F9/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:42:13.482 MQT: stat/tasmota_7456F9/STATUS10 = {"StatusSNS":{"Time":"2022-11-17T09:42:13","ENERGY":{"TotalStartTime":"2022-11-16T14:02:20","Total":0.076,"Yesterday":0.000,"Today":0.076,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]}}}
09:42:13.488 MQT: stat/tasmota_7456F9/STATUS11 = {"StatusSTS":{"Time":"2022-11-17T09:42:13","Uptime":"0T00:10:37","UptimeSec":637,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WlanWii","BSSId":"xxx","Channel":6,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
09:52:29.410 ADE: ACCMODE 0x2D3C00, VRMS 6236480, Period 4475, IRMS 1773, 1773, WATT 3, -2, VA 658, 658, VAR 7, 7
09:52:30.410 ADE: ACCMODE 0x2D3C00, VRMS 6236581, Period 4475, IRMS 1773, 1773, WATT 20, 14, VA 658, 658, VAR -21, -15
09:52:31.410 ADE: ACCMODE 0x2D3C00, VRMS 6241275, Period 4475, IRMS 1773, 1773, WATT 66, -17, VA 659, 659, VAR -16, -2
09:52:31.992 HTP: Main Menu
09:52:32.386 ADE: ACCMODE 0x2D3C00, VRMS 6239728, Period 4475, IRMS 1773, 1773, WATT -60, -30, VA 658, 658, VAR -69, -6
09:52:33.322 HTP: Console
09:52:33.410 ADE: ACCMODE 0x2D3C00, VRMS 6238231, Period 4475, IRMS 1773, 1773, WATT 40, 9, VA 658, 658, VAR 22, 23
09:52:34.410 ADE: ACCMODE 0x2D3C00, VRMS 6238539, Period 4474, IRMS 1773, 1773, WATT -71, 2, VA 658, 658, VAR -79, -9
09:52:35.410 ADE: ACCMODE 0x2D3C00, VRMS 6239545, Period 4474, IRMS 1773, 1773, WATT -27, 14, VA 658, 658, VAR 15, 17
09:52:36.410 ADE: ACCMODE 0x2D3C00, VRMS 6238097, Period 4474, IRMS 1773, 1773, WATT -26, 7, VA 658, 658, VAR -43, -8
[... 32 lines cut ...]
09:53:06.410 ADE: ACCMODE 0x2D3C00, VRMS 6236567, Period 4473, IRMS 1773, 1773, WATT -27, 5, VA 658, 658, VAR 25, -7
09:53:07.410 ADE: ACCMODE 0x2D3C00, VRMS 6234156, Period 4473, IRMS 1773, 1773, WATT 17, 0, VA 658, 658, VAR -7, -16
09:53:08.410 ADE: ACCMODE 0x2D3C00, VRMS 6230873, Period 4474, IRMS 1773, 1773, WATT -21, -2, VA 657, 658, VAR -7, -14
09:53:09.410 ADE: ACCMODE 0x2D3C00, VRMS 6235104, Period 4473, IRMS 1773, 1773, WATT -37, -7, VA 658, 658, VAR -7, -17
09:53:10.402 ADE: ACCMODE 0x2D3C00, VRMS 6233921, Period 4474, IRMS 1773, 1773, WATT -38, 14, VA 658, 658, VAR -35, -9
09:53:11.410 ADE: ACCMODE 0x2D3C00, VRMS 6238757, Period 4473, IRMS 1773, 1773, WATT -54, 2, VA 658, 658, VAR -36, 2
09:53:12.411 ADE: ACCMODE 0x2D3C00, VRMS 6238330, Period 4473, IRMS 1773, 1773, WATT -55, -14, VA 658, 658, VAR -3, 18

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Shelly EM with newest Tasmota.bin.gz and try to calibrate the Amp Clamp.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Calibration works 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-11-17:
Just checked mine with v12.2.0 and see no problem.
```
12:33:55.327 CMD: currentcal
12:33:55.329 SRC: WebConsole from 192.168.2.1
12:33:55.332 CMD: Grp 0, Cmd 'CURRENTCAL', Idx 1, Len 0, Pld -99, Data ''
12:33:55.338 MQT: stat/shellyem/RESULT = {"CurrentCal":8900}
12:34:01.194 CMD: teleperiod
12:34:01.196 SRC: WebConsole from 192.168.2.1
12:34:01.199 CMD: Grp 0, Cmd 'TELEPERIOD', Idx 1, Len 0, Pld -99, Data ''
12:34:01.213 MQT: stat/shellyem/RESULT = {"TelePeriod":300}
12:34:01.438 MQT: tele/shellyem/STATE = {"Time":"2022-11-17T12:34:01","Uptime":"0T00:03:07","UptimeSec":187,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-26,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:34:01.453 MQT: tele/shellyem/SENSOR = {"Time":"2022-11-17T12:34:01","ENERGY":{"TotalStartTime":"2022-09-30T16:29:30","Total":[-0.036,0.190],"Yesterday":[0.000,0.000],"Today":[-0.036,0.190],"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[0.107,0.000],"Period":[-0.84,0.84],"Power":[-58.70,58.14],"ApparentPower":[58.72,58.45],"ReactivePower":[0.00,0.00],"Factor":[-1.00,0.99],"Frequency":49.989,"Voltage":219.767,"Current":[0.262,0.260]}}
12:34:08.633 CMD: currentset 200
12:34:08.636 SRC: WebConsole from 192.168.2.1
12:34:08.638 CMD: Grp 0, Cmd 'CURRENTSET', Idx 1, Len 3, Pld 200, Data '200'
12:34:08.645 MQT: stat/shellyem/RESULT = {"CurrentSetCal":11600}
12:34:09.587 CFG: Saved to flash at F5, Count 336, Bytes 4096
12:34:14.665 CMD: teleperiod
12:34:14.667 SRC: WebConsole from 192.168.2.1
12:34:14.670 CMD: Grp 0, Cmd 'TELEPERIOD', Idx 1, Len 0, Pld -99, Data ''
12:34:14.675 MQT: stat/shellyem/RESULT = {"TelePeriod":300}
12:34:15.420 MQT: tele/shellyem/STATE = {"Time":"2022-11-17T12:34:15","Uptime":"0T00:03:21","UptimeSec":201,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-26,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:34:15.434 MQT: tele/shellyem/SENSOR = {"Time":"2022-11-17T12:34:15","ENERGY":{"TotalStartTime":"2022-09-30T16:29:30","Total":[-0.036,0.191],"Yesterday":[0.000,0.000],"Today":[-0.036,0.191],"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[0.107,0.000],"Period":[-0.23,0.23],"Power":[-58.68,58.12],"ApparentPower":[58.93,58.34],"ReactivePower":[0.00,0.00],"Factor":[-1.00,1.00],"Frequency":49.989,"Voltage":220.327,"Current":[0.201,0.199]}}
12:34:25.700 CMD: currentcal 8900
12:34:25.702 SRC: WebConsole from 192.168.2.1
12:34:25.705 CMD: Grp 0, Cmd 'CURRENTCAL', Idx 1, Len 4, Pld 8900, Data '8900'
12:34:25.712 MQT: stat/shellyem/RESULT = {"CurrentCal":8900}
12:34:26.576 CFG: Saved to flash at F4, Count 337, Bytes 4096
12:34:35.702 CMD: teleperiod
12:34:35.704 SRC: WebConsole from 192.168.2.1
12:34:35.706 CMD: Grp 0, Cmd 'TELEPERIOD', Idx 1, Len 0, Pld -99, Data ''
12:34:35.713 MQT: stat/shellyem/RESULT = {"TelePeriod":300}
12:34:36.431 MQT: tele/shellyem/STATE = {"Time":"2022-11-17T12:34:36","Uptime":"0T00:03:42","UptimeSec":222,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-26,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:34:36.446 MQT: tele/shellyem/SENSOR = {"Time":"2022-11-17T12:34:36","ENERGY":{"TotalStartTime":"2022-09-30T16:29:30","Total":[-0.036,0.191],"Yesterday":[0.000,0.000],"Today":[-0.036,0.191],"TodaySumImport":0.001,"TodaySumExport":0.001,"ExportActive":[0.107,0.000],"Period":[-0.34,0.34],"Power":[-58.69,58.15],"ApparentPower":[58.91,58.44],"ReactivePower":[0.00,0.00],"Factor":[-1.00,1.00],"Frequency":49.989,"Voltage":220.043,"Current":[0.262,0.260]}}
```
As can be seen above command ``CurrentCal`` returns the current calibration value. Setting it to 200mA (``CurrentSet 200``) changes the output of teleperiod (and GUI). Setting it back to previous calibration value of 8900 changes the output teleperiod once more to the original value of 260mA

You might want to upgrade to latest dev release which allows independant calibration for both channels using commands ``currentset1`` and ``currentset2``.

Starfoxfs, 2022-11-17:
Hi,
ok thanks i will try the DEV Version.

In my case CurrentSet will be accepted by Console and the Value will be changed in Console but i can´t see changes on the GUI (MainMenu).

CurrentCal works, but i thinked this is for "Fine Tuning" and not for general "rough" Calibration

Starfoxfs, 2022-11-17:
Yeah, thanks with Development it works

`PowerSet2 <value>` and `CurrentSet2 <value>`

One thing, my ADE7953 IRQ is configured to 1 and i have to use `PowerSet2 <value>` and `CurrentSet2 <value>` for right values in GUI, is there a twist or is that the right way ?

arendst, 2022-11-17:
Yes. If your IRQ is set to 1 you do NOT use the template as documented in the beginning of this issue.

For the Shelly EM the IRQ should be set to 2. This will swap the channels and make commands ``PowerSet1`` act on channel 1 and ``PowerSet2`` act on channel 2.

Starfoxfs, 2022-11-17:
Ok big thanks for help

closed 2022-11-19 by arendst