arendst/Tasmota #17111: ESP32-D0WD-V3+ Tasmota 12.2.0 No start.

Arunias, 2022-11-17:
## PROBLEM DESCRIPTION
In the ESP32-WROOM-32D module, through the programmer, I installed Tasmota 12.2.0. The firmware did not start. Then I installed the old firmware Tasmota 10.0.0. The module started working.In the module information, the flash size was defined as 8192kb, 4M flash 25Q32 is installed.After trying to update via the web interface on Tasmota 12.2.0, the module stops working. In the log at startup 00:00:07.000 QPC: Reset

00:00:05.777 WIF: Connected
00:00:06.974 HTP: Web server active on tasmota-46AFFC-4092 with IP address 192.168.1.124
00:00:06.985 QPC: Reset
09:52:35.585 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"10.0.0(tasmota)","FallbackTopic":"cmnd/DVES_46AFFC_fb/","GroupTopic":"cmnd/tasmotas/"}}
09:52:35.592 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-46AFFC-4092","IPAddress":"192.168.1.124"}}
09:52:35.603 RSL: INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
09:52:39.915 RSL: STATE = {"Time":"2022-11-17T09:52:39","Uptime":"0T00:00:13","UptimeSec":13,"Heap":141,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":42,"MqttCount":0,"Wifi":{"AP":1,"SSId":"5g","BSSId":"D4:6E:0E:A8:AF:CF","Channel":6,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:52:39.939 RSL: SENSOR = {"Time":"2022-11-17T09:52:39","ESP32":{"Temperature":53.3},"TempUnit":"C"}
09:53:53.190 UPP: Multicast (re)joined
09:54:18.144 UPL: File tasmota32 (1).bin
09:55:07.273 UPP: Multicast (re)joined
09:55:07.574 UPL: Successful 1501984 bytes
09:55:08.977 APP: Restarting
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.002 HDW: ESP32-D0WD-V3 
00:00:00.005 CFG: Use defaults
00:00:00.046 QPC: Reset
00:00:00.075 BRY: Exception> 'internal_error' - Error calling spi_flash_read(0x8000, 4096)
00:00:00.076 stack traceback:
00:00:00.077   <native>: in native function
00:00:00.089   <unknown source>: in function load
00:00:00.090   <unknown source>: in function init
00:00:00.092   <unknown source>: in function main
00:00:00.099 BRY: Berry initialized, RAM used=3709 bytes
00:00:00.117 Project tasmota - Tasmota Version 12.2.0(tasmota)-2_0_5(2022-10-17T08:39:59)
00:00:00.274 WIF: WifiManager active for 3 minutes
00:00:00.380 WIF: Retry set Mode...
00:00:00.482 WIF: Retry set Mode...
00:00:01.078 HTP: Web server active on tasmota-46AFFC-4092 with IP address 0.0.0.0
00:00:06.969 QPC: Reset

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP32-WROOM-32D   ESP32-D0WD-V3  FLASH 25Q32
- [x ] Tasmota binary firmware version number used: Tasmota Version 12.2.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: flash_download_tool_3.8.5.exe  ,WEB
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-11-17:
Please used the recommended flashing toolssuch as the web flasher at https://github.io/tasmota/install
No support is provided when using other tools

Arunias, 2022-11-17:
> Please used the recommended flashing toolssuch as the web flasher at https://github.io/tasmota/install No support is provided when using other tools

Your link is not working.

 I found the problem. This is the cheapest module on aliexpress, I bought a batch of them. My ESP32 module uses flash 25Q32ET . I replaced the flash with a proven one. The module worked. I think the problem is that Tasmota is not compatible with flash 25Q32ET .