arendst/Tasmota #17125: ESP32-WROOM-32D_DevKitC_v4

 keeps crashing

AleXSR700, 2022-11-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Good morning,
I am trying to find the cause for my frequent reboots (roughly every 6-12h).

Error message:
```

01:22:39.973 SRC: Berry
01:22:39.974 CMD: Grp 0, Cmd 'BLEALIAS', Idx 1, Len 0, Pld -99, Data ''
01:22:39.988 MQT: stat/tasmota_D2EE70/RESULT = {"BLEAlias":{"001A2208C9FA":"eq3_dining_room","001A2208B97D":"eq3_living_room","001A2214ADC8":"eq3_master_bedroom","001A2214A6DF":"eq3_kitchen","001A2214ADBF":"eq3_master_bathroom","001A2208CFD3":"eq3_walk_in","001A2214A73A":"eq3_guest_toilet","001A2214B86E":"eq3_office","A4C138F28078":"xiaomi_living_room","A4C138E28B41":"xiaomi_master_bathroom","A4C13876598C":"xiaomi_master_bedroom","A4C13880F246":"xiaomi_walk_in","A4C1380D5C0C":"xGuru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4001a917  PS      : 0x00060034  A0      : 0x80055fe4  A1      : 0x3ffc264c  
A2      : 0x0014a55a  A3      : 0x002f0007  A4      : 0x00060024  A5      : 0x00000001  
A6      : 0x3ffc0fc4  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x0000002f  
A10     : 0x00000000  A11     : 0x00060024  A12     : 0x00060024  A13     : 0x3ffb2fdc  
A14     : 0x3ffb5ba0  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0014a55e  LBEG    : 0x4009337c  LEND    : 0x40093392  LCOUNT  : 0xffffffff  

Backtrace: 0x4001a914:0x3ffc264c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
```
Can anybody make something of the error message?
Is there some way I can log to get more details on the error (already logging via FTDI serial connection with `SerialLog 4`

I also tried resetting but it had no impact. 

Thank you
Alex

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): esp32
- [x ] Tasmota binary firmware version number used: 12.2.0.1-bluetooth
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-18:
Try latest development version. I make my bet the issues is caused from Bluettooth part.

arendst, 2022-11-18:
Does it always stop at the same situation (long BLEAlias message)? If not pls provide another logging info too.

You might want to add heap monitoring with command ``SetOption130 1``.

AleXSR700, 2022-11-20:
So, I built the latest dev 2 days ago and so far no crash. I will continue to monitor and if it starts crashing (regularly) again, I will add heap monitoring and report back.

Thank you both for your suggestions!