arendst/Tasmota #17141: Moes zigbee thermostat shows wrong humidity. Should be 43% but is 4%

stefanbode, 2022-11-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Release 12.1.1. any help to investigate welcome

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2022-11-19:
12:00:59.828 MQT: hm/status/ESP_E2FC64/SENSOR = {"ZbInfo":{"Raumsensor":{"Device":"0x4B66","Name":"Raumsensor","IEEEAddr":"0xA4C138AD9A90E472","ModelId":"TS0201","Manufacturer":"_TZ3000_ywagc4rj","Endpoints":[1],"Config":["T01"],"Temperature":23.2,"Humidity":4.3,"Reachable":true,"BatteryPercentage":73,"BatteryLastSeenEpoch":1668854472,"LastSeen":1187,"LastSeenEpoch":1668854472,"LinkQuality":32}}}
12:00:59.831 MQT: hm/status/ESP_E2FC64/RESULT = {"ZbInfo":"Done"}

Looks like there is 10x shift in numbers.

s-hadinger, 2022-11-19:
Could you please capture the same reading with `weblog 3` ?

stefanbode, 2022-11-19:
Doing the Zbinfo make no change on the weblog 3. I saw there is something like a multiplier you can define in the value. As far as I understand you need a .zb file in the filesystem. As mentioned before it looks like the current Multiplier is 1 and need to be 10.

s-hadinger, 2022-11-19:
Yes indeed. But I wanted to check the low level value. I.e. is it a bug in Tasmota or a bug in the device. Weblog 3 was meant to capture the logs when the device publishes Humidity

stefanbode, 2022-11-19:
Some more info during transmitting data. 

18:41:04.823 ZIG: ZbZCLRawReceived: {"0x4B66":{"0402/0000":2316,"Endpoint":1,"LinkQuality":16}}
18:41:04.830 ZIG: {"ZbEZSPReceived":"34000009"}
18:41:04.916 ZIG: {"ZbEZSPReceived":"3F0000664B0401020401010000000009010000"}
18:41:04.938 ZIG: {"ZbEZSPReceived":"45000004010504010100010000FA48AE664BFFFF08080C0A000021A60102"}
18:41:04.940 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0405","srcaddr":"0x4B66","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":14,"securityuse":0,"seqnumber":250,"fc":"0x08","frametype":0,"direction":1,"disableresp":0,"manuf":"0x0000","transact":12,"cmdid":"0x0A","payload":"000021A601"}}
18:41:04.941 ZigbeeZCLSend device: 0x4B66, endpoint:1, cluster:0x0405, cmd:0x0B, send:"0A00"
18:41:04.943 ZIG: ZbEZSPSend 340000664B040105040101000000000C0105100C0B0A00
18:41:04.944 ZIG: ZbZCLRawReceived: {"0x4B66":{"0405/0000":422,"Endpoint":1,"LinkQuality":14}}
18:41:04.951 ZIG: {"ZbEZSPReceived":"3400000A"}
18:41:05.039 ZIG: {"ZbEZSPReceived":"3F0000664B040105040101000000000A010000"}
18:41:05.341 MQT: hm/status/ESP_E2FC64/SENSOR = {"ZbReceived":{"Raumsensor":{"Device":"0x4B66","Name":"Raumsensor","Temperature":23.16,"Humidity":4.22,"BatteryPercentage":73,"Endpoint":1,"LinkQuality":14}}}

s-hadinger, 2022-11-20:
Thanks, this confirms that the device does not follow the ZCL standard. Humidity is supposed to be reported in 1/100 not in 1/10.

This plugin file should do the trick (name it with extension .zb):

```
#Z2Tv1
: TS0201,_TZ3000_ywagc4rj
0405/0000=0405/0000,mul:10
```

closed 2022-11-20 by stefanbode

s-hadinger, 2022-11-20:
Documentation for zigbee plugins is now available https://tasmota.github.io/docs/Zigbee/#advanced-topic-zigbee-device-plugin