arendst/Tasmota #17155: INA3221 and INA226 are reported as INA219

nexxoc, 2022-11-21:
### PROBLEM DESCRIPTION
Both current sensors INA226 and INA3221 are reported as INA219 in the latest Tasmota release. I compiled the latest .bin using VSC, and removed the lines for the INA219 by commenting "//  #define USE_INA219" in the my_user_config.h. 

The screenshot shows an INA3221 connected via I2C bus 0x40 ("16:01:24.550 MQT: stat/tasmota_DCE620/RESULT = {"I2CScan":"Device(s) found at 0x40"}" from i2cscan, but it is reported as an INA219:
```
"INA219":{"Id":40,"Voltage":0,"Current":0.000,"Power":0}
```

This is also the case when I connect an INA226.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32
- [ ] Tasmota binary firmware version number used: 12.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VS Code
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:03:28.325 MQT: stat/tasmota_DCE620/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
16:03:34.808 MQT: stat/tasmota_DCE620/RESULT = {"Module":{"1":"ESP32-DevKit"}}
16:03:37.908 MQT: stat/tasmota_DCE620/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1408":"MHZ Tx"},"GPIO2":{"0":"None"},"GPIO3":{"1440":"MHZ Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"1312":"DS18x20"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"4864":"ADC Range1"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
16:04:25.512 MQT: stat/tasmota_DCE620/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_DCE620","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:04:25.551 MQT: stat/tasmota_DCE620/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:05:44","StartupUTC":"2022-11-21T14:58:41","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2022-11-21T15:25:41","SaveCount":9}}
16:04:25.599 MQT: stat/tasmota_DCE620/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022.11.21 15:23:45","Core":"2_0_5","SDK":"v4.4.3.4","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"442/699"}}
16:04:25.624 MQT: stat/tasmota_DCE620/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Palisades 2,4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
16:04:25.676 MQT: stat/tasmota_DCE620/STATUS4 = {"StatusMEM":{"ProgramSize":1443,"Free":1856,"Heap":140,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000407","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D434482D","00000001"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,100,127"}}
16:04:25.756 MQT: stat/tasmota_DCE620/STATUS5 = {"StatusNET":{"Hostname":"tasmota-DCE620-1568","IPAddress":"192.168.0.61","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"24:62:AB:DC:E6:20","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:04:25.808 MQT: stat/tasmota_DCE620/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DCE620","MqttUser":"bheimmitte","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:04:25.845 MQT: stat/tasmota_DCE620/STATUS7 = {"StatusTIM":{"UTC":"2022-11-21T15:04:25","Local":"2022-11-21T16:04:25","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:08","Sunset":"17:03"}}
16:04:25.892 MQT: stat/tasmota_DCE620/STATUS10 = {"StatusSNS":{"Time":"2022-11-21T16:04:25","ANALOG":{"Range1":49},"DS18B20":{"Id":"3C01B556AAB6","Temperature":20.7},"INA219":{"Id":40,"Voltage":0,"Current":0.000,"Power":0},"MHZ19B":{"Model":"","CarbonDioxide":0,"Temperature":0.0},"TempUnit":"C"}}
16:04:25.927 MQT: stat/tasmota_DCE620/STATUS11 = {"StatusSTS":{"Time":"2022-11-21T16:04:25","Uptime":"0T00:05:44","UptimeSec":344,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":8365,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"Palisades 2,4","BSSId":"2C:91:AB:8E:D7:9E","Channel":1,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Flash an ESP32 with the latest Tasmota build, with INA3221 enabled, and connect the device to I2C.

### EXPECTED BEHAVIOUR
The device should report as INA3221, and not as INA219

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-21:
Have you disabled the `INA219` driver?

nexxoc, 2022-11-21:
By commenting `// #define USE_INA219` in the `my_user_config.h`, the INA219 driver shouldn't be included in the .bin anyways, but when I set `I2cDriver14 0` in the console, the whole device disappears.

closed 2022-11-21 by nexxoc