arendst/Tasmota #17169: Sonoff basic with PZEM-004 configuration loosing WiFi connection after powercycle

HS1979, 2022-11-23:
### PROBLEM DESCRIPTION

Hi,
I have completed power meter with use of Sonoff basic R1 and PZEM-004 using instructions from here https://tasmota.github.io/docs/PZEM-0XX/ . So actions made are only adding additional 1k resistor and connecting wires between PZEM-004 and sonoff. 
Tasmota config for sonoff is like this:
[image]

Everything work fine until power is lost. After powering up, Sonoff cannot connect to WiFi anymore. Sometimes in very short periods (second or two) I can see on my WiFi access point that it is connected for very shor time (1-5 seconds).

Tried with different access points, with PZEM-004 connected or not., with multiple Sonoff Basic R1 devices, always same behaviour. 

After moment when I configure Generic(18) template as on picture, Sonoff should not be powered off as it can not connect to WiFi anymore. If Sonoff is not powered off, everything works OK for days, no any issues with WiFi connection.

To me this looks like some bug in Tasmota or missconfiguration that I don't know how to resolve. Tried with 10.0 and latest 12.2 Tasmota version,

Does anybody have some hint or solution for this?

Thanks. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Basic R1
- [ ] Tasmota binary firmware version number used: 12.2, 10.0
  - [x] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:55:51.843 MQT: stat/sonoff_18B3F2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:55:52.070 MQT: stat/sonoff_18B3F2/RESULT = {"Module":{"18":"Generic"}}
11:55:52.275 MQT: stat/sonoff_18B3F2/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"0":"None"},"GPIO3":{"1504":"PZEM004 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:57:11.752 MQT: stat/sonoff_18B3F2/STATUS = {"Status":{"Module":18,"DeviceName":"BasicR1+PowL3","FriendlyName":["BasicR1+PowL3"],"Topic":"sonoff_18B3F2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:57:11.759 MQT: stat/sonoff_18B3F2/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"3T12:45:11","StartupUTC":"2022-11-19T22:12:00","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2022-11-19T23:09:20","SaveCount":16,"SaveAddress":"F4000"}}
11:57:11.765 MQT: stat/sonoff_18B3F2/STATUS2 = {"StatusFWR":{"Version":"10.0.0.4(tasmota)","BuildDateTime":"2022-11-19T18:30:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"351/699"}}
11:57:11.771 MQT: stat/sonoff_18B3F2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000500060000005A00000000000000","00000080","00006000","00004000"]}}
11:57:11.780 MQT: stat/sonoff_18B3F2/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","80400487","04008001","00000E8E","010013C0","00000800","00004000","00000000","00002020"],"Drivers":"1,2,3,4,12,16,29","Sensors":"3,6,8,9,91"}}
11:57:11.785 MQT: stat/sonoff_18B3F2/STATUS5 = {"StatusNET":{"Hostname":"sonoff-18B3F2-5106","IPAddress":"192.168.x.x","Gateway":"192.168.x.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.x.1","DNSServer2":"0.0.0.0","Mac":"80:7D:3A:18:B3:F2","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:57:11.789 MQT: stat/sonoff_18B3F2/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_18B3F2","MqttUser":"xxxx","MqttCount":12,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:57:11.793 MQT: stat/sonoff_18B3F2/STATUS7 = {"StatusTIM":{"UTC":"2022-11-23T10:57:11","Local":"2022-11-23T11:57:11","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00"}}
11:57:11.796 MQT: stat/sonoff_18B3F2/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:57:11.804 MQT: stat/sonoff_18B3F2/STATUS10 = {"StatusSNS":{"Time":"2022-11-23T11:57:11","ENERGY":{"TotalStartTime":"2022-11-19T23:09:20","Total":8.347,"Yesterday":1.979,"Today":0.958,"Power":18,"ApparentPower":51,"ReactivePower":47,"Factor":0.36,"Voltage":241,"Current":0.210}}}
11:57:11.811 MQT: stat/sonoff_18B3F2/STATUS11 = {"StatusSTS":{"Time":"2022-11-23T11:57:11","Uptime":"3T12:45:11","UptimeSec":305111,"Vcc":3.423,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":12,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"02:00:00:AB:00:00","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":8,"Downtime":"0T00:00:48"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Configure template for use with PZEM-004 and power cycle device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

After power cycle, device should connect to WiFi without problems.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-23:
Sonoff Basic R1 power supply is not able to deliver enough current to power itself and a connected PZEM. It really does not count if config gets lost since this setup will not work reliable in real life

closed 2022-11-23 by Jason2866

HS1979, 2022-11-23:
I'm sorry but your answer is wrong. Issue is present even if PZEM is not connected to Sonoff. Moreover, PZEM is not supplyed by Sonoff, PZEM is supplyed with its own power supply. How do you explain that everything works fine for days if Sonoff is not powered off after configuration change?

barbudor, 2022-11-23:
I understand your problem as your Sonoff is loosing it's configuration on power lost.

Is that a correct understanding ?

What happens when loosing configuration ? 
- Does it present the default TASMOTA Access Point so you can connect to the device on 192.168.4.1 and reconfigure ? 
- In that case, is the template still here or did the template totally disapear ?

Does the problem occurs only if PZEM GPIO are configured ?
Or does it happens also if the PZEM GPIOs are not configured ?

HS1979, 2022-11-23:
Here are steps how to reproduce issue (I made it many times), you can do it without connecting device to mains, USB connection via 3,3V adapter would work, like when you flashing new firmware with Tasmotizer :
1. reset device to defaults, so device acts as access point with ssid <tasmota-somenumbers>
2. connect to this SSID with pc and configure access to local WiFi
3. after step 2. device sucessfully connects to WiFi. You can power off device and power on again, it will connect to WiFi without problems
4. go to Configuration > Configure module > select Generic (18), click Save
5. device restarts, disconnects and connects to WiFi very quickly. In this state, if you power off device and power it back, it connect to WiFi without problems.
6. go again to Configuration >  Configure module > select Generic (18) and select exactly this options:

[image]
7. Device soft resets, disconnects and connects to WiFi very quickly, and everything looks OK. If PZEM is now connected, everything works. BUT...if now you power of device (PZEM connected or not), device does not connect to WiFi anymore.
Also there is no default WiFi SSID "Tasmota-<somenumbers>" so device is not reset configuration, but have problems with connecting to WiFi. In this state, sometimes I saw on my WiFi AP that device actually connects to AP but disconnects imidiately. Sometimes (mostly) there is no any logs on AP so device is in some unknown state (LED blinking like as it is not connected to WiFi).

To repair device from this state, there is only reset procedure (6x quickly power on/power off) or reflashing via USB.

barbudor, 2022-11-23:
Why are you using Generic (18) instead of Sonoff Basic (1) ?

I just did the same test as Jason, using the proper template with Sonoff Basic (1)
Power cycled the device 3 times and it always come back on Wifi

You may have done something wrong with your cabling

HS1979, 2022-11-24:
HI,
I was just following guide founded on internet how to integrate PZEM and Sonoff Basic and there was used Generic(18) template.
Actually yes, when I tried with Sonof Basic (1) template, then all works fine. So it is obviously thath Generic(18) template is not compatible with Sonoff Basic.

barbudor, 2022-11-24:
There might be a reason that Theo included a this predefined template

HS1979, 2022-11-24:
Yes, but I thought that General templates are there to customize device. What if I want to my relay or led acting in reverse way? Or completely disable relay and LED output and connect some other sensors on that sonoff pins. Than I should use General templates, right?

Jason2866, 2022-11-24:
Use the module/template and change the settings based on that.

sfromis, 2022-11-24:
There is no `Generic(18)` template. In Configure Module, you can select `Generic (18)`, meaning that you use no template, but has to configure all relevant pins in Configure Module; thus you can replicate the pin configs coming from Sonoff Basic in a generic module config, in addition to whatever other gpio configs you want.