arendst/Tasmota #17178: Energy measurement: Suggestion to introduce a separate "ImportActive" set of counters plus sum

hagenbuch, 2022-11-24:
### PROBLEM DESCRIPTION
First off: My Tasmota is only version 12.1.1 but the online compiler could not get newer sourcecode and as far I've read the changelog this topic had not been addressed betwenn 12.1 and now (12.2)

From a Shelly 3 EM, I am able to read the counter values, like so

    SENSOR = {"Time":"2022-11-24T13:04:03","ENERGY":{"TotalStartTime":"2022-11-22T13:42:58","Total":0.00627,"Yesterday":-0.00006,"Today":0.00634,"ExportActive":[0.02465,0.00001,0.00001],"Period":[0.0,0.0,0.0],"Power":[-0.2,0.0,0.0],"ApparentPower":[3.2,0.0,0.0],"ReactivePower":[3.2,0.0,0.0],"Factor":[-0.05,0.00,0.00],"Frequency":[49.96,48.85,136.83],"Voltage":[236.9,0.1,0.1],"Current":[0.013,0.014,0.013],"CurrentNeutral":0.011}}

(don't worry about odd frequencies, i connected only one phase).

ExportActive gives three values as you see here but I can access variables ExportActive[0] to [3], that is four values. Is the [0] the sum of the three? That would be great, actually!

However, I discovered that the "Total" energy counter runs backwards when energy is exported. This is not a bad behaviour per se as there are many old "Ferraris" mechanical energy counters that do exactly this.

However, I would suggest to introduce one more set of counters, like ImportActive that does not run backwards. Ideally, it should hold the sum of the three in the index [0]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly EM 3
- [x] Tasmota binary firmware version number used: 12.1.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

13:10:22.451 CMD: Backlog Template; Module; GPIO 255;
13:10:22.467 RSL: RESULT = {"NAME":"Shelly 3EM","GPIO":[1,1,288,1,32,8065,0,0,640,8064,608,224,8096,0],"FLAG":0,"BASE":18}
13:10:22.700 RSL: RESULT = {"Module":{"0":"Shelly 3EM"}}
13:10:22.947 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"8065":"ADE7880 IRQ2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"8064":"ADE7880 IRQ1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"224":"Relay1"},"GPIO16":{"8096":"Reset"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

13:11:17.061 CMD: Backlog Rule1; Rule2; Rule3
13:11:17.077 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":200,"Free":311,"Rules":"ON tele-ENERGY#ExportActive[0]>0 DO var1 %value% ENDON ON tele-ENERGY#Total>0 DO WebSend [example.com] /?id=mydeviceid&time=%utctime%&uptime=%uptime%&bezug=%value%&lieferung=%var1% ENDON"}}
13:11:17.310 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:11:17.561 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

13:12:08.231 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"myname","FriendlyName":["myname"],"Topic":"tasmota_74AB20","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:12:08.243 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T03:13:15","StartupUTC":"2022-11-24T08:58:53","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-11-22T13:42:58","SaveCount":83,"SaveAddress":"F9000"}}
13:12:08.254 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.1(TasmoCompiler-esp8266generic)","BuildDateTime":"2022.11.16 19:24:21","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"353/699"}}
13:12:08.264 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dellekom",""],"TelePeriod":60,"Resolution":"568B90C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
13:12:08.278 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F804587","04260001","00000007","00000000","C000E981","00000004","00001000","54800000","00000000"],"Drivers":"1,2,3,4,6,8,9,10,12,18,19,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3"}}
13:12:08.290 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-74AB20-2848","IPAddress":"192.168.13.40","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.13.1","DNSServer2":"0.0.0.0","Mac":"34:94:54:74:AB:20","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:12:08.299 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_74AB20","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:12:08.311 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-11-24T12:12:08","Local":"2022-11-24T13:12:08","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:12","Sunset":"17:00"}}
13:12:08.320 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:12:08.336 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-11-24T13:12:08","ENERGY":{"TotalStartTime":"2022-11-22T13:42:58","Total":0.00627,"Yesterday":-0.00006,"Today":0.00634,"ExportActive":[0.02465,0.00001,0.00001],"Power":[0.1,0.0,-0.0],"ApparentPower":[3.1,0.0,0.0],"ReactivePower":[3.1,0.0,0.0],"Factor":[0.02,0.00,0.00],"Frequency":[49.98,48.20,32.08],"Voltage":[236.1,0.1,0.1],"Current":[0.013,0.013,0.013],"CurrentNeutral":0.012}}}
13:12:08.349 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-11-24T13:12:08","Uptime":"0T03:13:15","UptimeSec":11595,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","BSSId":"74:42:7F:31:xx:xx","Channel":11,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

hagenbuch, 2022-11-24:
Many thanks for your quick reply!

- Your suggestion would work but to be honest, I would prefer two (three) counters (Total, Import, Export) because the readings might confuse a lot of people just as myself, and: Professional counters summarize in two data fields, OBIS code 1.8.0 and 2.8.0 - so I think it would be nice to have something comparable. 

- The "Total" currently only holds the sum of the three phases. I would like to have Total[1], Total[2], Total[3] for the sums of each phase and Total[0] for the sum of the three phases. ExportActive[0] seems to work as variable but is not mentioned in the docs and does not appear in the JSON apparently.

I know, we ask for feature after feature :)  but I am not good enough to compile a pull request myself yet...

arendst, 2022-11-24:
> The "Total" currently only holds the sum of the three phases.

That's exactly why I suggested to use SO129.....