arendst/Tasmota #17183: Sonoff ZBBridge Pro - cc.check() report error

pploume, 2022-11-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
SonOff Zigbee Pro flashed with Tasmota ([tasmota32-zbbrdgpro.factory.bin), 
FS resized
4 files uploaded : 
cc2652_flasher.be
intelhex.be
sonoff_zb_pro_flasher.be
SonoffZBPro_coord_20220219.hex

when running the berry script, cc.check() report an error :
value error - invalid size for line : 20000000003A0120D1F8010039FB021039FB021039FB021039FB021039FB021039FB

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try flahsing with latest (2022-11-15) files available on github...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
seems to be a mistake from me, don't understand what I missed

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-11-24:
I can't reproduce. How did you download the files? It looks like the CR/LF was changed

pploume, 2022-11-25:
Github
Raw + ctrl C
VisualStudioCode + ctrl V

basic way...

Yes, it sounds like crlf prob.
I,ll check again

Jason2866, 2022-11-25:
The file you have uploaded is modified and will not work.

Jason2866, 2022-11-25:
Why is fs resized before using the files which are already in place?

pploume, 2022-11-25:
@Jason2866 
_Why is fs resized before using the files which are already in place?_
sorry, I dont understand what you mean ...

Jason2866, 2022-11-25:
You wrote
> A clear and concise description of what the problem is.
SonOff Zigbee Pro flashed with Tasmota ([tasmota32-zbbrdgpro.factory.bin),
FS resized
4 files uploaded :
cc2652_flasher.be
intelhex.be
sonoff_zb_pro_flasher.be
SonoffZBPro_coord_20220219.hex

There is FS resized. Do not resize, because this will erase all the files already in place you need for flashing the zigbee mcu

pploume, 2022-11-25:
@sfromis 
That's what I didn't understood ...

Trying to reflash the [tasmota32-zbbrdgpro.factory.bin] with esp-flasher,
error message is : 
Using 'COM3' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:D7:EB:10:C1:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

:(

s-hadinger, 2022-11-25:
Try the web installer instead. I don't know how you download files from Github but the way you do it is visibly broken.

https://tasmota.github.io/install/

sfromis, 2022-11-25:
ESP-Flasher does not support .factory images. The web installer is best to use, but you could also do it with esptool.py, flashing the single-file image at location 0x000000

Jason2866, 2022-11-25:
Maybe i have some day the spare time and the motivation to add the possibility to use *.factory images with ESP_Flasher

pploume, 2022-11-25:
@all 
Thanks for advices ... and the best of all : 
[tasmota32-zbbrdgpro.factory.bin] contains all needed files
A detail I didn't understood ...

Tasmota PyFlasher has uploaded correctly the .bin file

Writing while updatating as ZB Coordinator ... hope all will be OK

thanks again to everybody

closed 2022-11-25 by Jason2866