arendst/Tasmota #17185: Tasmota for ESP32 keeps the setup IP (192.168.4.1) in the running configuration

meyergru, 2022-11-25:
### PROBLEM DESCRIPTION
When using "SetOption53 1", one can see the configured IP on the main screen. With Tasmota32, there are two IP addresses, one of which is 192.168.4.1, the setup IP. I assume that this is a residue from the setup mode as a WLAN AP. If the address is used otherwise, it would lead to all kinds of problems.

This does not happen in Tasmota for ESP8266.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Pro Bridge
- [x] Tasmota binary firmware version number used: 12.2.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:50:46.044 MQT: stat/zigbeepro/RESULT = {"NAME":"TCP ZBBridge Pro","GPIO":[0,0,576,0,480,0,0,0,0,1,1,5792,0,0,0,5472,0,320,5793,5504,0,640,608,32,0,0,0,0,0,1,0,0,0,0,0,0],"FLAG":0,"BASE":1}
08:50:46.271 MQT: stat/zigbeepro/RESULT = {"Module":{"0":"TCP ZBBridge Pro"}}
08:50:46.538 MQT: stat/zigbeepro/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5792":"Zigbee Rst1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"5472":"TCP Tx"},"GPIO20":{"0":"None"},"GPIO21":{"320":"Led_i1"},"GPIO22":{"5793":"Zigbee Rst2"},"GPIO23":{"5504":"TCP Rx"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"608":"I2C SCL1"},"GPIO27":{"32":"Button1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:51:56.118 MQT: stat/zigbeepro/STATUS = {"Status":{"Module":0,"DeviceName":"zigbeepro","FriendlyName":["zigbeepro"],"Topic":"zigbeepro","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:51:56.131 MQT: stat/zigbeepro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"34T21:19:57","StartupUTC":"2022-10-21T10:31:59","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2022-10-07T20:21:07","SaveCount":93}}
08:51:56.141 MQT: stat/zigbeepro/STATUS2 = {"StatusFWR":{"Version":"12.2.0.1(zbbrdgpro)","BuildDateTime":"2022-10-19T11:48:47","Core":"2_0_5","SDK":"v4.4.3.5-17-gd772747b2","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"408/699"}}
08:51:56.150 MQT: stat/zigbeepro/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"syslog.mgsoft","LogPort":5140,"SSId":["IdIoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]}}
08:51:56.173 MQT: stat/zigbeepro/STATUS4 = {"StatusMEM":{"ProgramSize":1401,"Free":454,"Heap":139,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F8A87CF","00548001","000000CF","010013C0","C0000981","40024080","00200000","5400082C","00000000"],"Drivers":"1,2,3,4,5,8,9,10,12,16,20,21,23,24,27,29,35,38,41,50,52,59,62,82","Sensors":"1,2,3,5,6,127"}}
08:51:56.188 MQT: stat/zigbeepro/STATUS5 = {"StatusNET":{"Hostname":"zigbeepro","IPAddress":"192.168.7.30","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.7.1","DNSServer2":"0.0.0.0","Mac":"0C:DC:7E:32:BE:9C","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:51:56.204 MQT: stat/zigbeepro/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32BE9C","MqttUser":"DVES_USER","MqttCount":11,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:51:56.214 MQT: stat/zigbeepro/STATUS7 = {"StatusTIM":{"UTC":"2022-11-25T07:51:56","Local":"2022-11-25T08:51:56","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:14","Sunset":"16:59"}}
08:51:56.234 MQT: stat/zigbeepro/STATUS10 = {"StatusSNS":{"Time":"2022-11-25T08:51:56"}}
08:51:56.251 MQT: stat/zigbeepro/STATUS11 = {"StatusSTS":{"Time":"2022-11-25T08:51:56","Uptime":"34T21:19:57","UptimeSec":3014397,"Heap":143,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":11,"Berry":{"HeapUsed":3,"Objects":48},"Wifi":{"AP":1,"SSId":"IdIoT","BSSId":"6A:22:32:29:CF:3A","Channel":1,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":6,"Downtime":"0T00:04:33"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Just use the device.
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I would expect the 192.168.4.1 IP only to be assigned when in setup mode, not during normal operation.
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-11-25:
Please try latest development version.

meyergru, 2022-11-25:
Same result with the latest 12.2.0.5.

dkadioglu, 2022-12-08:
For me the same with a build from two days ago.

dkadioglu, 2023-01-02:
Before this gets closed automatically, just a little reminder, that this is still happening.

sfromis, 2023-01-02:
Has anyone posted any info about even having a problem? You see an IP, but does not seem to have had any trouble....

With Tasmota on ESP32 it is entirely normal that the AP mode IP address is displayed along with the active one, even if only used when actually in AP mode.