arendst/Tasmota #17200: ME81H Thermostat + Tasmota

RobertBerger, 2022-11-27:
I probably should not report it here, but i theory the ME81H Termostat should be supported by Tasmota[1]

[1] https://templates.blakadder.com/ME81H.html

The chip soldered in my device is a Model: WBR3 

Is this supposed to be flashed in circuit, or do I need to unsolder/replace it with another chip?

I will contact:

- If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4) as well.

closed 2022-11-27 by close-issue-app[bot]

barbudor, 2022-11-27:
> I probably should not report it here

Issues are only to report qualified bugs
You should have open a Discussion

Nevertheless WBR3 does not include a ESP processor and won't support tasmota
You need to replace the module.
Apparently ESP12-E/ESP12-F should be fine:
https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy
https://tasmota.github.io/docs/Pinouts/#esp-12eesp-12f

A good opportunity to replace by an ESP32c3 based module such as the ESP12-C3-F : https://templates.blakadder.com/ESP-C3-12F.html

sfromis, 2022-11-27:
In this case, you could report the device change using the edit link on each template page.

g6094199, 2023-12-06:
[image]

for reference

g6094199, 2023-12-23:
> I probably should not report it here, but i theory the ME81H Termostat should be supported by Tasmota[1]
>
> [1] https://templates.blakadder.com/ME81H.html
>
> The chip soldered in my device is a Model: [...]

the WBR3 chip needs to be replaced to a ESP based one to be able to run tasmota.
or you flash OpenBK7231T_App  to the wbr3 modul which is a tasmota like firmware replacement for non esp based boards