arendst/Tasmota #17211: Tasmota client garbled code

cn-superman, 2022-11-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Still me; -), Last time I submitted a question,
I didn't express myself too clearly.
This time, I carefully checked and compared.
As shown below:
1. In the code, I send the code to TASMOTA and send it to another serial port.
2. All hardware is welded.
3. The level voltage is 3.3V (5V has also been tried)
The result is still irregular garbled code.
It can be seen from the screenshot that the data output to the serial port is in a normal format,
TAS received garbled code.
Some rules I found:
    The shorter the data sent, the lower the probability of garbled code.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ✓] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ✓] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ✓] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ✓] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ]✓ Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ✓] Tasmota binary firmware version number used:  9.5.0.1  12.2.0
  - [ ✓] Pre-compiled
  - [ ✓] Self-compiled
- [ ✓] Flashing tools used: _esptool ota
- [ ✓] Provide the output of command: `Backlog Template; Module; GPIO 255`:
-       {"NAME":"McuDuino","GPIO":[32,0,4224,0,4192,4160,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
```lua
  Configuration output here:
![error](https://user-images.githubusercontent.com/45441185/204463363-58bcd0ec-2abc-470c-803c-9512a2ca0080.png)

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

  arduino use tasmotaclient     sendtel

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

cn-superman, 2022-11-30:
20221130: supplement
If the compiled version is 8.4.0, all faults disappear.

cn-superman, 2022-12-04:
I'm sorry, but I'm not familiar with the use of GITHUB.
I try my best to provide detailed information for reference.
I hope to get your reply and thank you for your hard work.

barbudor, 2022-12-04:
There is not many people using TasmotaClient and having experience with it
I can try to have a look later today but no warranty. I'm not even sure to still have a working arduino environment to test

cn-superman, 2022-12-04:
> There is not many people using TasmotaClient and having experience with it I can try to have a look later today but no warranty. I'm not even sure to still have a working arduino environment to test

Thank you for your reply. I just did some tests.
The IO used for the failure is: CLIENT TX: GPIO4 CLIENT RX: GPIO5
I set TX and RX to the hard serial port of 8266: GPIO1 GPIO3,
The probability of finding garbled code decreases
Conclusion: TX and RX using custom IO will irregularly generate garbled code at a fast transmission speed (actually not fast, about 500ms or even slower), which has completely ruled out the problem of sending end and connection.

arendst, 2022-12-05:
I ran the default apps available for TasmotaClient without a problem.

I suspect you are battering tasmota from the client with too much data. A much better approach is to use the client/server paradigm where you let tasmota retreive the data from the client when it is ready for it (like the examples do). 

The difference you see with lower tasmota versions is likely caused by additional sensor data requests introduced in latest versions. The goal is to use these requests to gather info from the client and process it as teleperiod data or regular rule processing data.

cn-superman, 2022-12-05:
> I ran the default apps available for TasmotaClient without a problem.
>
> I suspect you are battering tasmota from the client with too much data. A much better approach is to use the client/server [...]

Thanks for your reply.
That's not the truth.
I once suspected that it was caused by the high frequency of the data sent.
I use the MQTT tool to send data manually with a maximum interval of several seconds.
The length of each transmission string is about 60 characters.
I have tested version by version from 8.1.0 to 9.1.0.
I found that the tasmotaClient seems to have been refactored from the version up to 9.1.0 (excluding 9.1.0).
I tested the phenomenon:
1. For versions below 9.1.0, it is better to use custom IO, but it is not good to use GPIO1 GPIO3.
2. For versions above 9.1.0 (excluding 9.1.0), it is better to use a hard serial port (io1 io3). Using custom IO is prone to garbled code.
3. The occurrence of garbled code is random, independent of the transmission frequency, but related to the length of the transmission data.
English is not my native language. I hope Google can translate it clearly. Thank you.

cn-superman, 2022-12-07:
> As some changes were made to TasmotaSerial since v12.2 pls try latest dev release from http://ota.tasmota.com/tasmota
> 
> The preffered "higher" speed comms between TasmotaClient and the Uno/Mini is 57600 when not using HardwareSerial (= GPIO1/3 on ESP8266). Any lower baudrate is highly appreciated to lower serial interrupts and data corruption. Speeds over 50000 are being recognised by TasmotSerial (SoftwarSerial) as very_high_speed and will use a method to read as much data in one action into the default 100 byte data buffer.

Thanks for your reply.
I will continue to test.

closed 2022-12-07 by cn-superman