arendst/Tasmota #17236: tasmota xiaomi phone ap webserver connect error

cheng5230, 2022-12-01:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x48
load:0x403ce000,len:0x64c
load:0x403d0000,len:0x215c
SHA-256 comparison failed:
Calculated: 469c045d559266a8ca425d808dbb2c8e79976d6f6c0d095621fa0d16175a9536
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000

00:00:00.001-242/50 HDW: ESP32-C3FH4 
00:00:00.012-240/50 UFS: FlashFS mounted with 308 kB free
00:00:00.030 CFG: Loaded from File, Count 8
00:00:00.035 QPC: Count 2
00:00:00.036 CFG: CR 384/699, Busy 0
00:00:00.038 ROT: Mode 1
00:00:00.043 BRY: GC from 3598 to 3398 bytes, objects freed 4/19 (in 0 ms) - slots from 31/61 to 23/61
00:00:00.073 CFG: No '*.autoconf' file found
00:00:00.076 BRY: GC from 5298 to 4215 bytes, objects freed 6/38 (in 1 ms) - slots from 56/76 to 35/76
00:00:00.077 BRY: Berry initialized, RAM used=4215 bytes
00:00:00.094 BRY: No 'preinit.be'
00:00:00.096 SRC: Restart
00:00:00.098 Project tasmota - Tasmota Version 12.0.2.4(tasmota)-2_0_4_1(2022-11-30T16:42:06)
00:00:00.119 BRY: No 'autoexec.be'
00:00:00.203 WIF: Checking connection...
00:00:00.204 WIF: WifiManager active for 3 minutes
00:00:00.370 WIF: Wifimanager set AccessPoint
00:00:00.877 HTP: Blank Device - Initial Configuration
00:00:00.879 HTP: Web server active on tasmota-A01914-6420 with IP address 192.168.4.1
00:00:06.915 QPC: Reset
00:00:08.918 APP: Boot Count 6
00:00:09.850 CFG: Saved, Count 9, Bytes 4096
00:01:49.834 HTP: Redirected to captive portal
00:06:13.573 HTP: Redirected to captive portal
00:06:13.672 HTP: Configure WiFi
00:06:13.702 WIF: Scanning...
00:06:20.954 WIF: Scan done
00:06:29.745 HTP: Configure WiFi
00:06:29.762 WIF: Scanning...
00:06:36.614 WIF: Scan done
00:06:40.482 HTP: Redirected to captive portal
00:06:40.682 HTP: Redirected to captive portal
00:06:40.830 HTP: Configure WiFi
00:06:40.872 WIF: Scanning...
00:06:48.117 WIF: Scan done
00:06:53.090 HTP: Configure WiFi
00:06:53.121 WIF: Scanning...
00:07:00.169 WIF: Scan done

xiaomi phoe  webserver connect error
 
[image]

barbudor, 2022-12-01:
Your Xiaomi is automatically trying to connect to http://connect.rom.miui.com through Tasmota AP
No way this is going to work through Tasmota. It is meant to access a connection portal on real AP that have access to the Internet.

Ignore the error and directly open http://192.168.4.1

cheng5230, 2022-12-02:
hi barbudor
browser directly open http://192.168.4.1/
**connect success no show control ip** 
[image]