arendst/Tasmota #17238: Unable to Flash Beca BAC-6000

mojtabavi, 2022-12-01:
I'm Trying to flash beca BAC-6000 thermostat using CH340G device
But the desired esp module is not placed in the program mode and sends incomprehensible characters on the serial
ESP Model: 
TYWE3S
Connection:
esp-gnd --> gnd
esp-3v3 --> 3v3
esp-gpio0 --> gnd
esp-rx --> tx
esp-tx --> rx
[image]
[image]
I'm really confused, I'd really appreciate it if someone could help

mojtabavi, 2022-12-01:
https://user-images.githubusercontent.com/63472358/205036010-e8d3612c-2c84-4e5e-8e38-842dd2eebb89.mp4

julred, 2022-12-01:
Correct me if I'm wrong - but this is not a tasmota related issue

mojtabavi, 2022-12-01:
> Correct me if I'm wrong - but this is not a tasmota related issue

I know, but i'm trying to flash tasmota, And I'm trying to get help

barbudor, 2022-12-01:
It is not a Tasmota Issue so indeed it should have been into Discussions instead of Issue

However @mojtabavi You don't show how you have tried to put the ESP in flashing mode
Are you 100% to have connected GPIO0 to GND ?
How ? Directly with a wire ? On the TYWES3 module ?
The connection must be done BEFORE applying power
I would recommand soldering a wire directly on the TYWES3 between GPIO and GND

mojtabavi, 2022-12-01:
yes I soldering a wire directly on the TYWES3 between GPIO and GND

mojtabavi, 2022-12-01:
> It is not a Tasmota Issue so indeed it should have been into Discussions instead of Issue
>
> However @mojtabavi You don't show how you have tried to put the ESP in flashing mode Are you 100% to have [...]

yes I soldering a wire directly on the TYWES3 between GPIO and GND

barbudor, 2022-12-01:
Could be a deffective device but can't tell more
No photos of your actual wiring and setup no it could be anything

joba-1, 2022-12-01:
Do you use baud rate 74880 during boot? Might make incomprehensible characters readable and give a hint on boot mode.

sfromis, 2022-12-01:
The PCB has two big chips, one of which may be using serial to communicate with the ESP8266. To flash, you may very well have to disable that chip to get the serial lines free for flashing. If you manage to get the ESP8266 flashed, next step would be to find out how you can control the device via the other MCU. It may be a Tuya MCU, which is normally supported by Tasmota. Or it may be something else.

sfromis, 2022-12-01:
Then find out how you can disable it during flashing. Many of those chips have some sort of enable or reset pin.