arendst/Tasmota #17240: Light off after wifi reconection

mr-varga, 2022-12-02:
Dear all,
I have some sonoff basic with tasmota and I've been having a problem for some time. 

I switch the device ON, after a random time the device lost the wifi connection for a short time, perform the reconection and after reconection the state of device become OFF. 
This is very painful because the devices switch on the light in my house.

To GPIO14 is connected external switch.

The devices is connected to mosquito mqtt in Home Assistant. 

The wifi network is UniFi access point.

- [ ] This is the output of: `Status 0`:
```lua
03:13:21.777 CMD: status 0
03:13:21.784 MQT: stat/lusala1/STATUS = {"Status":{"Module":1,"DeviceName":"lusala1","FriendlyName":["lusala1"],"Topic":"lusala1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"lusala1","SwitchMode":[5,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
03:13:21.790 MQT: stat/lusala1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T03:22:23","StartupUTC":"2022-12-01T22:50:58","Sleep":50,"CfgHolder":4617,"BootCount":60,"BCResetTime":"2021-08-02T10:51:52","SaveCount":2094,"SaveAddress":"F8000"}}
03:13:21.795 MQT: stat/lusala1/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"401/699"}}
03:13:21.799 MQT: stat/lusala1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mr-Varga_Admin","P9_Zsolt"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["40028029","0505C80001000600003C5A0A002800000000","00002280","00006000","00004000","00000000"]}}
03:13:21.810 MQT: stat/lusala1/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
03:13:21.816 MQT: stat/lusala1/STATUS5 = {"StatusNET":{"Hostname":"lusala1","IPAddress":"192.168.2.31","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:93:56:0B","Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
03:13:21.820 MQT: stat/lusala1/STATUS6 = {"StatusMQT":{"MqttHost":"mrvarga.duckdns.org","MqttPort":1883,"MqttClientMask":"lusala1","MqttClient":"lusala1","MqttUser":"pi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:13:21.826 MQT: stat/lusala1/STATUS7 = {"StatusTIM":{"UTC":"2022-12-02T02:13:21","Local":"2022-12-02T03:13:21","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"08:23","Sunset":"16:55"}}
03:13:21.831 MQT: stat/lusala1/STATUS10 = {"StatusSNS":{"Time":"2022-12-02T03:13:21","Switch1":"ON"}}
03:13:21.837 MQT: stat/lusala1/STATUS11 = {"StatusSTS":{"Time":"2022-12-02T03:13:21","Uptime":"0T03:22:23","UptimeSec":12143,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

- [ ] This is the output of: `Backlog Template; Module; GPIO 255`:
```lua
03:22:12.510 MQT: stat/lusala1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
03:22:12.720 MQT: stat/lusala1/RESULT = {"Module":{"1":"Sonoff Basic"}}
03:22:12.975 MQT: stat/lusala1/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"160":"Switch1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

- [ ] Console log:
```lua
00:00:00 CFG: Loaded from flash at FB, Count 1738
00:00:00 QPC: Count 1
00:00:00 Project tasmota lusala2 Version 9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:04 WIF: Connecting to AP1 Mr-Varga_Admin Channel 6 BSSId 1E:E8:29:9D:7D:09 in mode 11n as lusala2...
00:00:05 WIF: Connected
00:00:06 HTP: Web server active on lusala2 with IP address 192.168.2.32
00:00:06 QPC: Reset
00:51:48 MQT: Attempting connection...
00:51:48 MQT: Connected
00:51:48 MQT: tele/lusala2/LWT = Online (retained)
00:51:48 MQT: cmnd/lusala2/POWER = 
00:51:48 MQT: tele/lusala2/INFO1 = {"Module":"Sonoff Basic","Version":"9.2.0(tasmota)","FallbackTopic":"cmnd/lusala2_fb/","GroupTopic":"cmnd/tasmotas/"}
00:51:48 MQT: tele/lusala2/INFO2 = {"WebServerMode":"Admin","Hostname":"lusala2","IPAddress":"192.168.2.32"}
00:51:48 MQT: tele/lusala2/INFO3 = {"RestartReason":"Software/System restart"}
00:51:48 MQT: stat/lusala2/RESULT = {"POWER":"OFF"}
00:51:48 MQT: stat/lusala2/POWER = OFF (retained)
00:51:48 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T00:51:48","Uptime":"0T00:00:08","UptimeSec":8,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":77,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:51:48 MQT: stat/lusala2/RESULT = {"POWER":"OFF"}
00:51:48 MQT: stat/lusala2/POWER = OFF (retained)
00:51:49 MQT: stat/lusala2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:09","StartupUTC":"2022-12-01T23:51:40","Sleep":50,"CfgHolder":4617,"BootCount":49,"BCResetTime":"2020-04-05T18:07:11","SaveCount":1738,"SaveAddress":"FB000"}}
00:51:49 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T00:51:49","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":42,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:51:49 MQT: stat/lusala2/RESULT = {"Time":"2022-12-02T00:51:49","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":42,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:51:49 MQT: stat/lusala2/STATUS11 = {"StatusSTS":{"Time":"2022-12-02T00:51:49","Uptime":"0T00:00:09","UptimeSec":9,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":45,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}}
00:51:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T00:51:52","Uptime":"0T00:00:12","UptimeSec":12,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:51:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T00:51:52","Switch1":"ON"}
00:52:23 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T00:52:23","Uptime":"0T00:00:43","UptimeSec":43,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:52:23 MQT: stat/lusala2/RESULT = {"POWER":"ON"}
00:52:23 MQT: stat/lusala2/POWER = ON (retained)
00:56:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T00:56:52","Uptime":"0T00:05:12","UptimeSec":312,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"1E:E8:29:9D:7D:09","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:56:52 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T00:56:52","Switch1":"ON"}
01:01:39 APP: Serial logging disabled
01:01:52 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:01:52","Uptime":"0T00:10:12","UptimeSec":612,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSI [... cut ...]
01:36:51 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:36:51","Switch1":"ON"}
01:41:51 MQT: tele/lusala2/STATE = {"Time":"2022-12-02T01:41:51","Uptime":"0T00:50:11","UptimeSec":3011,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:09"}}
01:41:51 MQT: tele/lusala2/SENSOR = {"Time":"2022-12-02T01:41:51","Switch1":"ON"}
```

There is someone with this problem who can help me to solve it?

barbudor, 2022-12-02:
You haven't set 'Weblog 4' as requested to provide the logs.
Please do as requested and publish new logs

Jason2866, 2022-12-02:
It seems you have retained mqtt messages. Delete the retained messages and your ghost switching will be gone.

mr-varga, 2022-12-02:
> You haven't set 'Weblog 4' as requested to provide the logs. Please do as requested and publish new logs

Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F6, Count 2104
00:00:00.056 QPC: Count 1
00:00:00.062 ROT: Mode 1
00:00:00.063 SRC: Restart
00:00:00.065 Project tasmota - lusala1 Version 12.1.1(tasmota)-2_7_4_9(2022-08-25T11:33:55)
00:00:00.218 WIF: Checking connection...
00:00:00.219 WIF: Attempting connection...
00:00:00.556 WIF: Connecting to AP1 Mr-Varga_Admin Channel 13 BSSId 7E:8A:20:B7:F9:97 in mode 11n as lusala1...
00:00:01.754 WIF: Checking connection...
00:00:01.754 WIF: Connected
00:00:02.008 HTP: Web server active on lusala1 with IP address 192.168.2.31
00:00:02.405 WIF: Sending Gratuitous ARP
00:00:02.407 NTP: Sync time...
00:00:04.406 RTC: UTC 2022-12-02T12:06:09, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
13:06:09.001 RTC: Synced by NTP
13:06:09.010 HTP: Console
13:06:10.024 MQT: Attempting connection...
13:06:10.082 MQT: Connected
13:06:10.086 MQT: tele/lusala1/LWT = Online (retained)
13:06:10.087 MQT: cmnd/lusala1/POWER = 
13:06:10.089 MQT: Subscribe to cmnd/lusala1/#
13:06:10.090 MQT: Subscribe to cmnd/tasmotas/#
13:06:10.091 MQT: Subscribe to cmnd/lusala1_fb/#
13:06:10.095 MQT: tele/lusala1/INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"12.1.1(tasmota)","FallbackTopic":"cmnd/lusala1_fb/","GroupTopic":"cmnd/tasmotas/"}}
13:06:10.101 MQT: tele/lusala1/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"lusala1","IPAddress":"192.168.2.31"}}
13:06:10.112 MQT: tele/lusala1/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":62}}
13:06:10.121 MQT: stat/lusala1/RESULT = {"POWER":"OFF"}
13:06:10.125 MQT: stat/lusala1/POWER = OFF (retained)
13:06:10.334 SRC: MQTT
13:06:10.336 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'OFF'
13:06:10.343 MQT: tele/lusala1/STATE = {"Time":"2022-12-02T13:06:10","Uptime":"0T00:00:05","UptimeSec":5,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":118,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:06:10.367 MQT: stat/lusala1/RESULT = {"POWER":"OFF"}
13:06:10.370 MQT: stat/lusala1/POWER = OFF (retained)
13:06:11.234 SRC: MQTT
13:06:11.237 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 1, Data '1'
13:06:11.242 MQT: stat/lusala1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:06","StartupUTC":"2022-12-02T12:06:05","Sleep":50,"CfgHolder":4617,"BootCount":61,"BCResetTime":"2021-08-02T10:51:52","SaveCount":2104,"SaveAddress":"F6000"}}
13:06:11.284 SRC: MQTT
13:06:11.287 CMD: Grp 0, Cmd 'STATE', Idx 1, Len 0, Pld -99, Data ''
13:06:11.292 MQT: tele/lusala1/STATE = {"Time":"2022-12-02T13:06:11","Uptime":"0T00:00:06","UptimeSec":6,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":57,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:06:11.313 MQT: stat/lusala1/RESULT = {"Time":"2022-12-02T13:06:11","Uptime":"0T00:00:06","UptimeSec":6,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":57,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:06:11.348 SRC: MQTT
13:06:11.351 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 2, Pld 11, Data '11'
13:06:11.357 MQT: stat/lusala1/STATUS11 = {"StatusSTS":{"Time":"2022-12-02T13:06:11","Uptime":"0T00:00:06","UptimeSec":6,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":61,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:06:12.399 QPC: Reset
13:06:14.391 APP: Boot Count 62
13:06:14.398 MQT: tele/lusala1/STATE = {"Time":"2022-12-02T13:06:14","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:06:14.427 MQT: tele/lusala1/SENSOR = {"Time":"2022-12-02T13:06:14","Switch1":"ON"}
13:06:14.681 CFG: Saved to flash at F5, Count 2105, Bytes 4096
13:06:19.416 MQT: tasmota/discovery/84F3EB93560B/config = {"ip":"192.168.2.31","dn":"lusala1","fn":["lusala1",null,null,null,null,null,null,null],"hn":"lusala1","mac":"84F3EB93560B","md":"Sonoff Basic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.1.1","t":"lusala1","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":1,"20":0,"30":1,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
13:06:19.428 MQT: tasmota/discovery/84F3EB93560B/sensors = {"sn":{"Time":"2022-12-02T13:06:19","Switch1":"ON"},"ver":1} (retained)
[... cut ...]
13:21:14.391 MQT: tele/lusala1/STATE = {"Time":"2022-12-02T13:21:14","Uptime":"0T00:15:09","UptimeSec":909,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Mr-Varga_Admin","BSSId":"7E:8A:20:B7:F9:97","Channel":13,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":2,"Downtime":"0T00:00:09"}}
13:21:14.397 MQT: tele/lusala1/SENSOR = {"Time":"2022-12-02T13:21:14","Switch1":"ON"}
13:21:32.016 WIF: Checking connection...
13:21:51.996 WIF: Checking connection...
```

barbudor, 2022-12-02:
13:16:12.433 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'OFF'

Confirms that the power off command comes from MQTT
As suggested by Jason it's most certainly a retained power command
See https://tasmota.github.io/docs/MQTT/#clearing-retained-messages

mr-varga, 2022-12-02:
> 13:16:12.433 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'OFF'
> 
> Confirms that the power off command comes from MQTT As suggested by Jason it's most certainly a retained power command See https://tasmota.github.io/docs/MQTT/#clearing-retained-messages

Thanks I try the procedure and for now seems is working.

closed 2022-12-02 by mr-varga