arendst/Tasmota #17266: Serial Bridge send additional space to serial port

avp-avp, 2022-12-04:
### PROBLEM DESCRIPTION

SSerialSend3 command send additional space to serial port. Other (S)SerialSend(3) have a same problem.

14:39:44.265 CMD: SSerialSend3 page loading_page\xFF\xFF\xFF
14:39:44.270 CMD: Grp 0, Cmd 'SSERIALSEND', Idx 3, Len 30, Pld -99, Data ' page loading_page\xFF\xFF\xFF'

For example it's broke communication with nextion display. Problem is in bug fix https://github.com/arendst/Tasmota/issues/16723#issuecomment-1336397605

In oldest versions leading spaces was cleaned 
00:00:00.112 Project tasmota - Bathroom Version 12.0.2(tasmota)-2_7_4_9(2022-06-20T12:41:03)
14:28:19.117 CMD: Grp 0, Cmd 'SSERIALSEND', Idx 3, Len 28, Pld -99, Data 'page temperature\xFF\xFF\xFF'

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues) - I found issue that cause this problem
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ -] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): https://templates.blakadder.com/lytko_101.html
- [ x] Tasmota binary firmware version number used: _____
  - [ x] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:22:33.378 CMD: Backlog Template; Module; GPIO 255
15:22:33.414 MQT: stat/tasmota_E1B5EC/RESULT = {"NAME":"Lytko 101","GPIO":[1,7584,1312,1,1792,1824,0,0,1,1,1,224,1,4736],"FLAG":0,"BASE":18}
15:22:33.641 MQT: stat/tasmota_E1B5EC/RESULT = {"Module":{"0":"Lytko 101"}}
15:22:33.852 MQT: stat/tasmota_E1B5EC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"7584":"Heartbeat"},"GPIO2":{"1312":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"1792":"SerBr Tx"},"GPIO5":{"1824":"SerBr Rx"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"225":"Relay2"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:23:07.360 CMD: Backlog Rule1; Rule2; Rule3
15:23:07.390 MQT: stat/tasmota_E1B5EC/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":458,"Free":53,"Rules":"ON System#Boot DO Backlog ThermostatModeSet %mem2%; SensorInputSet 1;temptargetset %mem1%;SSerialSend3 page temperature\\xFF\\xFF\\xFF ENDON     \tON Tele-ANALOG#Temperature DO var1=%value% ENDON \tON SSerialReceived#Data DO publish lytko/serial %value% ENDON \tON Time#Initialized DO Backlog Mem1=Mem1+0;Mem2=Mem2+0 ENDON \tON SSerialReceived#Data=1AFFFFFF DO Backlog SSerialSend3 page temperature\\xFF\\xFF\\xFF;Mem1=Mem1+0;Mem2=Mem2+0;Var1=Var1+0;Var2=Var2+0 ENDON\t"}}
15:23:07.613 RUL: Rule2 ON System#Init DO SSerialSend3 page loading_page\xFF\xFF\xFF 	ON Power1#state DO var2=%value% ENDON 	ON var2#state=0.000 DO SSerialSend3 arrow_temp.val=0\xFF\xFF\xFF ENDON 	ON var2#state=1.000 DO SSerialSend3 arrow_temp.val=1\xFF\xFF\xFF ENDON 	ON SSerialReceived#Data=65010E01FFFFFF DO Backlog Mem1=Mem1-0.5; ENDON 	ON SSerialReceived#Data=65010D01FFFFFF DO Backlog Mem1=Mem1+0.5; ENDON 	ON mem1#state DO Backlog temptargetset %value%;SSerialSend3 degree.txt="%value%"\xFF\xFF\xFF ENDON 	ON var1#state>0 DO
15:23:07.618 RUL: Rule2 +SSerialSend3 degree_big.txt="%value%"\xFF\xFF\xFF ENDON 	ON SSerialReceived#Data=65011101FFFFFF DO Mem2=1 ENDON 	ON SSerialReceived#Data=65011401FFFFFF DO Mem2=0 ENDON	 	ON mem2#state=1.000 DO Backlog ThermostatModeSet 1;SSerialSend3 H_active.val=1\xFF\xFF\xFF ENDON 	ON mem2#state=0.000 DO Backlog ThermostatModeSet 0;SSerialSend3 H_active.val=0\xFF\xFF\xFF ENDON 	ON SSerialReceived#Data=6608FFFFFF DO Backlog Mem1=Mem1+0;Mem2=Mem2+0;Var1=Var1+0;Var2=Var2+0 ENDON 	ON SSerialReceived#Data=6601FFFFFF DO
15:23:07.622 RUL: Rule2 +Backlog Mem1=Mem1+0;Mem2=Mem2+0;Var1=Var1+0;Var2=Var2+0 ENDON ENDON
15:23:07.630 MQT: stat/tasmota_E1B5EC/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":1080,"Free":116,"Rules":"ON System#Init DO SSerialSend3 page loading_page\\xFF\\xFF\\xFF \tON Power1#state DO var2=%value% ENDON \tON var2#state=0.000 DO SSerialSend3 arrow_temp.val=0\\xFF\\xFF\\xFF ENDON \tON var2#state=1.000 DO SSerialSend3 arrow_temp.val=1\\xFF\\xFF\\xFF ENDON \tON SSerialReceived#Data=65010E01FFFFFF DO Backlog Mem1=Mem1-0.5; ENDON \tON SSerialReceived#Data=65010D01FFFFFF DO Backlog Mem1=Mem1+0.5; ENDON \tON mem1#state DO Backlog temptargetset %value%;SSerialSend3 degree.txt=\"%value%\"\\xFF\\xFF\\xFF ENDON \tON var1#state>0 DO SSerialSend3 degree_big.txt=\"%value%\"\\xFF\\xFF\\xFF ENDON \tON SSerialReceived#Data=65011101FFFFFF DO Mem2=1 ENDON \tON SSerialReceived#Data=65011401FFFFFF DO Mem2=0 ENDON\t \tON mem2#state=1.000 DO Backlog ThermostatModeSet 1;SSerialSend3 H_active.val=1\\xFF\\xFF\\xFF ENDON \tON mem2#state=0.000 DO Backlog ThermostatModeSet 0;SSerialSend3 H_active.val=0\\xFF\\xFF\\xFF ENDON \tON SSerialReceived#Data=6608FFFFFF DO Backlog Mem1=Mem1+0;Mem2=Mem2+0;Var1=Var1+0;Var2=Var2+0 ENDON \tON SSerialReceived#Data=6601FFFFFF DO Backlog Mem1=Mem1+0;Mem2=Mem2+0;Var1=Var1+0;Var2=Var2+0 ENDON ENDON"}}
15:23:07.838 MQT: stat/tasmota_E1B5EC/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":39,"Free":472,"Rules":"ON TempTargetSet1#state DO Var3=1 ENDON"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:23:28.601 CMD: Status 0
15:23:28.608 MQT: stat/tasmota_E1B5EC/STATUS = {"Status":{"Module":0,"DeviceName":"Bathroom","FriendlyName":["Tasmota",""],"Topic":"tasmota_E1B5EC","ButtonTopic":"0","Power":2,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:23:28.669 MQT: stat/tasmota_E1B5EC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release-12.2.0/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:45:38","StartupUTC":"2022-12-04T11:37:50","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2022-08-31T18:24:02","SaveCount":31,"SaveAddress":"F8000"}}
15:23:28.727 MQT: stat/tasmota_E1B5EC/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-17T08:39:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"392/699"}}
15:23:28.782 MQT: stat/tasmota_E1B5EC/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZoooT",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
15:23:28.844 MQT: stat/tasmota_E1B5EC/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
15:23:28.901 MQT: stat/tasmota_E1B5EC/STATUS5 = {"StatusNET":{"Hostname":"tasmota-E1B5EC-5612","IPAddress":"192.168.118.41","Gateway":"192.168.118.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.118.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:E1:B5:EC","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:23:28.956 MQT: stat/tasmota_E1B5EC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.122.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E1B5EC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:23:29.001 MQT: stat/tasmota_E1B5EC/STATUS7 = {"StatusTIM":{"UTC":"2022-12-04T12:23:28","Local":"2022-12-04T15:23:28","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+03:00","Sunri [... cut ...]
4. See Data ' page loading_page\xFF\xFF\xFF'. That's mean that leading space sended to serial port

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
4. I expect Data 'page loading_page\xFF\xFF\xFF'. That's mean that NO leading space sended to serial port

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-12-04:
It does not send an additional space since this commit, it does not remove anymore leading spaces. Thats the intended wanted behaviour. Yes thats an breaking change.

closed 2022-12-04 by Jason2866

avp-avp, 2022-12-04:
When you send command from web interface it have leading space 

SerialSend something

before SerialSend AND something you should add space and this space will be sended to serial port!

arendst, 2022-12-04:
See https://github.com/arendst/Tasmota/issues/16723#issuecomment-1336416448