arendst/Tasmota #17271: Tasmota on ESP32 crashes after MQ analog interface is set

zokl, 2022-12-04:
### PROBLEM DESCRIPTION
I have a problem with setting gpio interface to type ADC MQ on my TTGO ESP32 (http://www.lilygo.cn/prod_view.aspx?Id=1126) with the latest Tasmota. If I set MQ to analog gpio without connected input, everything works if I connect an input to the ADC MQ analog ping esp measure. 

The problem is if the analog input is connected to the ping before/during the boot process. Tasmota produces an error log shown in the following output, and after esp boot, the configuration of all gpio is in default.

This behavior is observed only in ADC MQ type of analog input if I use ADC Input boot process without a problem and everything works. 

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WDQ6-V3 
00:00:00.021 UFS: FlashFS mounted with 304 kB free
00:00:00.038 CFG: Loaded from File, Count 323
00:00:00.040 FRC: Some settings have been reset (3)
00:00:00.047 SPI: Software using GPIO18(CLK) and GPIO19(MOSI)
00:00:00.114 BRY: Berry initialized, RAM used=4041 bytes
00:00:02.598 DSP: ST7789
00:00:02.614 Project tasmota - Pokoj2 Version 12.2.0.6(tasmota)-2_0_5_3(2022-12-02T13:14:11)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40003230: 20c0edb8 b1049800 20c0f52a
Core  1 register dump:
PC      : 0x40003236  PS      : 0x00060630  A0      : 0x8015ea84  A1      : 0x3ffb2730  
A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x3ffc4ad8  A5      : 0x3ffc4cd8  
A6      : 0x3ffc3308  A7      : 0x3ffb2380  A8      : 0x80003236  A9      : 0x3ffb2710  
A10     : 0x00000000  A11     : 0x0000002d  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40086e39  LEND    : 0x40086e41  LCOUNT  : 0x00000027  

Backtrace: 0x40003233:0x3ffb2730 0x4015ea81:0x3ffb2750 0x400e76d7:0x3ffb2820

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WDQ6-V3 
00:00:00.021 UFS: FlashFS mounted with 304 kB free
00:00:00.038 CFG: Loaded from File, Count 323
00:00:00.040 FRC: Some settings have been reset (4)
00:00:00.047 SPI: Software using GPIO18(CLK) and GPIO19(MOSI)
00:00:02.526 DSP: ST7789
00:00:02.535 Project tasmota - Pokoj2 Version 12.2.0.6(tasmota)-2_0_5_3(2022-12-02T13:14:11)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40003230: 20c0edb8 b1049800 20c0f52a
Core  1 register dump:
PC      : 0x40003236  PS      : 0x00060c30  A0      : 0x8015ea84  A1      : 0x3ffb2730  
A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x3ffc4ad8  A5      : 0x3ffc4cd8  
A6      : 0x3ffc3308  A7      : 0x3ffb2380  A8      : 0x80003236  A9      : 0x3ffb2710  
A10     : 0x00000000  A11     : 0x0000002d  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40086e39  LEND    : 0x40086e41  LCOUNT  : 0x00000027  

Backtrace: 0x40003233:0x3ffb2730 0x4015ea81:0x3ffb2750 0x400e76d7:0x3ffb2820

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

00:00:00.002 HDW: ESP32-D0WDQ6-V3 
00:00:00.021 UFS: FlashFS mounted with 304 kB free
00:00:00.038 CFG: Loaded from File, Count 323
00:00:00.040 FRC: Some settings have been reset (5)
00:00:00.047 SPI: Software using GPIO18(CLK) and GPIO19(MOSI)
00:00:02.526 DSP: ST7789
00:00:02.535 Project tasmota - Pokoj2 Version 12.2.0.6(tasmota)-2_0_5_3(2022-12-02T13:14:11)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40003230: 20c0edb8 b1049800 20c0f52a
Core  1 register dump:
PC      : 0x40003236  PS      : 0x00060c30  A0      : 0x8015ea84  A1      : 0x3ffb2730  
A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x3ffc4ad8  A5      : 0x3ffc4cd8  
A6      : 0x3ffc3308  A7      : 0x3ffb2380  A8      : 0x80003236  A9      : 0x3ffb2710  
A10     : 0x00000000  A11     : 0x0000002d  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40086e39  LEND    : 0x40086e41  LCOUNT  : 0x00000027  

Backtrace: 0x40003233:0x3ffb2730 0x4015ea81:0x3ffb2750 0x400e76d7:0x3ffb2820

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4

```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): _____
- [X ] Tasmota binary firmware version number used: 12.2.0.6 (2_0_5_3/v4.4.3)
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X ] Flashing tools used: VSC
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

Working configuration with ADC Input

{"NAME":"TemesvarTTGO","GPIO":[32,0,1216,0,992,0,1,1,0,0,0,0,6624,1,896,864,0,256,257,1024,0,0,0,0,0,0,0,0,0,0,0,33,0,0,0,4704],"FLAG":0,"BASE":1}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

zokl, 2022-12-09:
I have a new finding. The problem occurs only when the analog input VCC max. If the voltage is < 3.3V at start up, everything is fine.

barbudor, 2022-12-09:
An ADC input should never ever go above VCC on ESP32 and above 1V on esp8266

zokl, 2022-12-09:
> An ADC input should never ever go above VCC on ESP32 and above 1V on esp8266

Yes, it is true but VCC on the analog pin clears after a cold reboot of the configuration.

barbudor, 2022-12-09:
I'm not sure to understand what you mean
Any voltage above VCC on the ESP32 can not only damage the ESP but also put it into a bad state and show bad behavior
Insure that there is never any over voltage

Jason2866, 2022-12-11:
Closing, since issue is caused by using device out of specifications.

zokl, 2022-12-11:
> Closing, since issue is caused by using device out of specifications.

There's a bug in the software. If Analog Input mode is set, it shows 4095 and everything works. The input is VCC 3 V. If I switch to MQ it crashes after reboot.

I would expect it to boot up normally and show high ppm values. This is not happening.

Translated with DeepL (https://www.deepl.com/app/?utm_source=ios&utm_medium=app&utm_campaign=share-translation

zokl, 2022-12-11:
> Closing, since issue is caused by using device out of specifications.

There's a bug in the software. If Analog Input mode is set, it shows 4095 and everything works. The input is VCC 3 V. If I switch to MQ it crashes after reboot.

I would expect it to boot up normally and show high ppm values. This is not happening.

zokl, 2022-12-12:
> @zokl Please try latest development version.

Thank you. It's working properly now.

[image]

closed 2022-12-12 by zokl