arendst/Tasmota #17272: SCD41 sensor no longer works in 12.2.xx

s-weise, 2022-12-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi,

the CO² sensor SCD41 no longer works under the Tasmota 12.2.0.0 (and 12.2.0.6). In the version 12.1.0.0 the sensor works normal. I use the Adafruit QT Py ESP32-C3 board with the SCD40, SGP30 and VINDRIKTNING sensor. A WS2812 LED is also used. I looked into the Tasmota libs for sensor control and i2c control but i can't find a change the explains this behavior. 

Hope my informations can help you.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Adafruit QT Py ESP32-C3
- [x] Tasmota binary firmware version number used: 12.2.0.6
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PlatformIO integrated flashing tool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:35:20.577 MQT: stat/VINDRIKTNING/RESULT = {"NAME":"ESP32C3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
21:35:20.947 MQT: stat/VINDRIKTNING/RESULT = {"Module":{"1":"ESP32C3"}}
21:35:21.159 MQT: stat/VINDRIKTNING/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"1376":"WS28121"},"GPIO4":{"8096":"VINDRIKTNING"},"GPIO5":{"640":"I2C SDA1"},"GPIO6":{"608":"I2C SCL1"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:36:21.504 MQT: stat/VINDRIKTNING/STATUS = {"Status":{"Module":1,"DeviceName":"VINDRIKTNING+","FriendlyName":["Tasmota"],"Topic":"VINDRIKTNING","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:36:21.513 MQT: stat/VINDRIKTNING/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T01:19:55","StartupUTC":"2022-12-04T19:16:26","Sleep":250,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2022-08-28T15:38:57","SaveCount":86}}
21:36:21.519 MQT: stat/VINDRIKTNING/STATUS2 = {"StatusFWR":{"Version":"12.2.0.6(tasmota)","BuildDateTime":"2022-12-04T20:15:00","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-C3 rev.3","CR":"421/699"}}
21:36:21.527 MQT: stat/VINDRIKTNING/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:36:21.539 MQT: stat/VINDRIKTNING/STATUS4 = {"StatusMEM":{"ProgramSize":1521,"Free":1358,"Heap":169,"StackLowMark":5,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DOUT","Features":["00000809","97DAD7CF","001DA001","B7F7BFCF","05DA9BC0","E0360DC7","000840D2","20200000","D43C682D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,91,92,95,98,!127"}}
21:36:21.544 MQT: stat/VINDRIKTNING/STATUS5 = {"StatusNET":{"Hostname":"VINDRIKTNING-6360","IPAddress":"192.168.1.146","Gateway":"192.168.1.10","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.20","DNSServer2":"0.0.0.0","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:36:21.551 MQT: stat/VINDRIKTNING/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EA38D8","MqttUser":"vindriktning","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:36:21.555 MQT: stat/VINDRIKTNING/STATUS7 = {"StatusTIM":{"UTC":"2022-12-04T20:36:21","Local":"2022-12-04T21:36:21","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"08:17","Sunset":"16:01"}}
21:36:21.563 MQT: stat/VINDRIKTNING/STATUS10 = {"StatusSNS":{"Time":"2022-12-04T21:36:21","SGP30":{"eCO2":400,"TVOC":12},"VINDRIKTNING":{"PM2.5":27}}}
21:36:21.570 MQT: stat/VINDRIKTNING/STATUS11 = {"StatusSTS":{"Time":"2022-12-04T21:36:21","Uptime":"0T01:19:55","UptimeSec":4795,"Heap":164,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":44},"POWER":"ON","Dimmer":20,"Color":"003300","HSBColor":"120,100,20","Channel":[0,20,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"XXXXX","Channel":11,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

00:00:00.265 Script: nv=12, tv=0, vns=62, vmem=330, smem=8192
00:00:00.400 SHT: Sensor did not ACK command
00:00:00.446 I2C: SGP30 found at 0x58
00:00:01.406 I2C: Error 5 at 0x49
00:00:01.420 SCD40 force-stop error: 0
00:00:01.970 SCD40 reinit error: 0
00:00:01.992 SCD40 serial nr 0xXXXX 0xXXXX 0xXXXX
00:00:01.993 I2C: SCD40 found at 0x62
00:00:01.993 SCD40 found, measurements started.
00:00:05.018 SCD40: no data available.
00:00:05.762 SCD40: no data available.
00:00:07.203 SCD40: Update: ReadMeasurement error: 0x500, counter: 4
00:00:08.095 SCD40: (rd) in error state: 2, good: 0, no data: 3, sc30 reset: 0, i2c reset: 0
00:00:08.218 SCD40: not answering, sending soft reset, counter: 5
00:00:08.342 SCD40: resetting got error: 0x200
21:38:12.883 SCD40: (rst) in error state: 3, good: 0, no data: 3, sc30 reset: 1, i2c reset: 0
21:38:12.883 SCD40: clearing i2c bus
21:38:13.295 SCD40: no data available.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Connect SCD41 with QT Py ESP32-C3.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Normal read out from the SCD41 sensor

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-12-05:
FWIW, I'm using SCD040 (not SCD041, but same driver) on Tasmota 12.2, no problems.

Jason2866, 2022-12-05:
Could be a i2c bus problem. Does the sensor work if you remove the other i2c sensors?

s-weise, 2022-12-05:
Maybe. For testing I desoldered the pull-up resistors from the sensor board and now the sensor works when nothing else is connected. Does Tasmota activates the internal pull-up resistors on the I2C pins? It's strange that it worked on version 12.1.0.0.

Jason2866, 2022-12-05:
i2c is a bus. For a correct bus setup the last device (seen from wiring) should only be terminated with pull up resistors. Lower resistors are more resilent against "noise".
The weak pull ups in esp does not count nor do any harm. Having pull ups on e every connected i2c device do most of the time work. But if power supply is weak you may run in trouble here.