arendst/Tasmota #17281: Periodically can't connect to MQTT broker after upgrade to 12.x

SteveDinn, 2022-12-05:
### PROBLEM DESCRIPTION
Upgraded most of my devices to v12.2.0, and strangely only in a few cases, devices have become "unavailable" in Home Assistant for a few minutes at a time before coming back. I can tell that these devices are able to reconnect to wi-fi immediately because I can still connect to the web console right after they become "unavailable". The problem is that they can't seem to reconnect to the MQTT broker. Nothing has happened to the MQTT broker because the plethora of other devices (TASMOTA included) in my home network are still using it the whole time.

The one thing I am doing that others may not be is using a DNS name instead of an IP address for my MQTT broker. It is provided by my local DNS server (Pi-Hole), which is up the entire time as well. I've verified that I can lookup the DNS name of the MQTT broker while TASMOTA is having this problem.

Here's what TASMOTA's console messages look like when I connect to the web console (I have redacted personal information):
```lua
15:13:51.914 WIF: Connecting to AP1 WiFi_SSID Channel 4 BSSId [WiFi_BSSID] in mode 11n as MonitorLight...
15:13:51.919 MQT: Attempting connection...
15:13:52.065 MQT: Connect failed to mqtt.example.com:1883, rc -5. Retry in 10 sec
15:14:05.832 MQT: Attempting connection...
15:14:06.046 MQT: Connect failed to mqtt.example.com:1883, rc -2. Retry in 20 sec
15:14:26.833 MQT: Attempting connection...
15:14:26.847 MQT: Connect failed to mqtt.example.com:1883, rc -2. Retry in 30 sec
15:14:57.822 MQT: Attempting connection...
15:14:57.839 MQT: Connect failed to mqtt.example.com:1883, rc -2. Retry in 40 sec
15:15:01.184 RSL: STATE = {"Time":"2022-12-02T15:15:01","Uptime":"31T01:05:55","UptimeSec":2682355,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":44,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WiFi_SSID","BSSId":"[WiFi_BSSID]","Channel":4,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":39,"Downtime":"0T00:02:22"}}
15:15:38.839 MQT: Attempting connection...
15:15:38.850 MQT: Connect failed to mqtt.example.com:1883, rc -2. Retry in 50 sec
15:16:30.113 MQT: Attempting connection...
15:16:30.228 MQT: Connected
15:16:30.232 MQT: tele/MonitorLight/LWT = Online (retained)
15:16:30.235 MQT: cmnd/MonitorLight/POWER = 
```
And as shown, it eventually does reconnect, but it takes a few minutes.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic and Sonoff S20
- [x] Tasmota binary firmware version number used: v12.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA upgrade
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:41:49.515 CMD: Backlog Template; Module; GPIO 255
15:41:49.517 SRC: WebConsole from 10.1.1.116
15:41:49.519 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
15:41:49.588 SRC: Backlog
15:41:49.591 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
15:41:49.597 MQT: stat/MonitorLight/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:41:49.801 SRC: Backlog
15:41:49.804 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
15:41:49.809 MQT: stat/MonitorLight/RESULT = {"Module":{"1":"Sonoff Basic"}}
15:41:50.052 SRC: Backlog
15:41:50.055 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
15:41:50.067 MQT: stat/MonitorLight/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
15:30:36.503 MQT: stat/MonitorLight/STATUS = {"Status":{"Module":1,"DeviceName":"Monitor Light","FriendlyName":["Monitor Light"],"Topic":"MonitorLight","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:30:36.511 MQT: stat/MonitorLight/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://tasmota.example.com/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"31T01:21:30","StartupUTC":"2022-11-01T13:09:06","Sleep":50,"CfgHolder":4617,"BootCount":2949,"BCResetTime":"2020-04-17T18:36:47","SaveCount":4550,"SaveAddress":"F6000"}}
15:30:36.515 MQT: stat/MonitorLight/STATUS2 = {"StatusFWR":{"Version":"12.2.0(tasmota)","BuildDateTime":"2022-10-17T08:39:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"424/699"}}
15:30:36.521 MQT: stat/MonitorLight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi_SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["40008029","0405C80001000600003C5A0A002800000000","000080C0","00006000","00004000","00000000"]}}
15:30:36.533 MQT: stat/MonitorLight/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
15:30:36.538 MQT: stat/MonitorLight/STATUS5 = {"StatusNET":{"Hostname":"MonitorLight","IPAddress":"10.1.1.168","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.1.1.4","DNSServer2":"1.1.1.1","Mac":"60:01:94:66:B2:52","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
15:30:36.542 MQT: stat/MonitorLight/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.example.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_66B252","MqttUser":"homeassistant","MqttCount":45,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:30:36.548 MQT: stat/MonitorLight/STATUS7 = {"StatusTIM":{"UTC":"2022-12-02T14:30:36","Local":"2022-12-02T15:30:36","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:23","Sunset":"16:55"}}
15:30:36.553 MQT: stat/MonitorLight/STATUS10 = {"StatusSNS":{"Time":"2022-12-02T15:30:36"}}
15:30:36.559 MQT: stat/MonitorLight/STATUS11 = {"StatusSTS":{"Time":"2022-12-02T15:30:36","Uptime":"31T01:21:30","UptimeSec":2683290,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":45,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WiFi_SSID","BSSId":"[WiFi_BSSID]","Channel":4,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":39,"Downtime":"0T00:02:22"}}}
[... 67 lines cut ...]
14:46:20.963 WIF: Checking connection...
14:46:40.949 WIF: Checking connection...
14:46:48.189 MQT: Attempting connection...
14:46:48.270 MQT: Connected
14:46:48.275 MQT: tele/MonitorLight/LWT = Online (retained)
14:46:48.278 MQT: cmnd/MonitorLight/POWER = 
14:46:48.280 MQT: Subscribe to cmnd/MonitorLight/#
14:46:48.283 MQT: Subscribe to cmnd/tasmotas/#
14:46:48.285 MQT: Subscribe to cmnd/tasmota-bin/#
14:46:48.287 MQT: Subscribe to cmnd/DVES_66B252_fb/#
14:46:49.001 MQT: tasmota/discovery/60019466B252/config = {"ip":"10.1.1.168","dn":"Monitor Light","fn":["Monitor Light",null,null,null,null,null,null,null],"hn":"MonitorLight","mac":"60019466B252","md":"Sonoff Basic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.2.0","t":"MonitorLight","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1 [... cut ...] aveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
```

### TO REPRODUCE
There are no steps to reproduce this other than to have a device with v12.2.0 and have a MQTT broker configured via its DNS name.

### EXPECTED BEHAVIOUR
Wi-fi drops happen, but TASMOTA should be able to resolve the name of, and reconnect to, the MQTT broker.

### ADDITIONAL CONTEXT
n/a

barbudor, 2022-12-05:
> "MqttHost":"mqtt.example.com"

Are you using a broker on the Internet ?
Or is that a local one with some DNS trickery ?

Scrambling the requested information is not helping us to understand your setup

Everything starts by a DNS problem:
> 14:41:58.624 MQT: Connect failed to mqtt.example.com:1883, rc -5. Retry in 10 sec

So I would assume something is wrong there. Your MQTT broker may be up but not the DNS

SteveDinn, 2022-12-05:
Should have been explicitly clear before:
* I have pi-hole running on the same network as the tasmota devices.
* My DHCP server points all clients to the pi-hole instance as the DNS server.
* The name "mqtt.mydomain.com" resolves to a local IP via pi-hole (in fact, the MQTT broker and pi-hole are running on the same raspberry pi).

MaceMoneta, 2022-12-06:
DNS2 is set to Cloudflare (1.1.1.1). If you remove that, does the problem still occur?